ayufan/camera-streamer #3: Support for Raspicam v2 camera?

RobeeeJay, 2022-07-12:
Does this support the v2 camera as well? I tried to get it working but when I view a stream I have the error "Can't queue buffer" :(

ayufan, 2022-07-12:
@RobeeeJay It should, but I don't have this camera, so I would need a few details to confirm this:

```
v4l2-ctl --list-devices
v4l2-ctl -d /dev/video0 -L
v4l2-ctl -d /dev/video0 --list-formats
v4l2-ctl -d /dev/v4l-subdev0 -L
v4l2-ctl -d /dev/v4l-subdev1 -L
libcamera-still --list-devices
libcamera-still
```

RobeeeJay, 2022-07-12:
That was a super fast response, thank you! Here are the details:

<details><summary>v4l2-ctl --list-devices</summary>

```
bcm2835-codec-decode (platform:bcm2835-codec):
	/dev/video10
	/dev/video11
	/dev/video12
	/dev/video18
	/dev/video31
	/dev/media0

bcm2835-isp (platform:bcm2835-isp):
	/dev/video13
	/dev/video14
	/dev/video15
	/dev/video16
	/dev/video20
	/dev/video21
	/dev/video22
	/dev/video23
	/dev/media1
	/dev/media2

unicam (platform:fe801000.csi):
	/dev/video0
	/dev/video1
	/dev/media3
```

</details>

<details><summary>v4l2-ctl -d /dev/video0 -L</summary>

```
nothing
```

</details>

<details><summary>v4l2-ctl -d /dev/video0 --list-formats</summary>

```
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'YUYV' (YUYV 4:2:2)
	[1]: 'UYVY' (UYVY 4:2:2)
	[2]: 'YVYU' (YVYU 4:2:2)
	[3]: 'VYUY' (VYUY 4:2:2)
	[4]: 'RGBP' (16-bit RGB 5-6-5)
	[5]: 'RGBR' (16-bit RGB 5-6-5 BE)
	[6]: 'RGBO' (16-bit A/XRGB 1-5-5-5)
	[7]: 'RGBQ' (16-bit A/XRGB 1-5-5-5 BE)
	[8]: 'RGB3' (24-bit RGB 8-8-8)
	[9]: 'BGR3' (24-bit BGR 8-8-8)
	[10]: 'RGB4' (32-bit A/XRGB 8-8-8-8)
	[11]: 'BA81' (8-bit Bayer BGBG/GRGR)
	[12]: 'GBRG' (8-bit Bayer GBGB/RGRG)
	[13]: 'GRBG' (8-bit Bayer GRGR/BGBG)
	[14]: 'RGGB' (8-bit Bayer RGRG/GBGB)
	[15]: 'pBAA' (10-bit Bayer BGBG/GRGR Packed)
	[16]: 'BG10' (10-bit Bayer BGBG/GRGR)
	[17]: 'pGAA' (10-bit Bayer GBGB/RGRG Packed)
	[18]: 'GB10' (10-bit Bayer GBGB/RGRG)
	[19]: 'pgAA' (10-bit Bayer GRGR/BGBG Packed)
	[20]: 'BA10' (10-bit Bayer GRGR/BGBG)
	[21]: 'pRAA' (10-bit Bayer RGRG/GBGB Packed)
	[22]: 'RG10' (10-bit Bayer RGRG/GBGB)
[... 82 lines cut ...]
```
[0:04:15.618607925] [996]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3544-22656360
[0:04:15.684433156] [997]  WARN RPI raspberrypi.cpp:1241 Mismatch between Unicam and CamHelper for embedded data usage!
[0:04:15.685109968] [997]  INFO RPI raspberrypi.cpp:1356 Registered camera /base/soc/i2c0mux/i2c@1/imx219@10 to Unicam device /dev/media3 and ISP device /dev/media1
Available cameras
-----------------
0 : imx219 [3280x2464] (/base/soc/i2c0mux/i2c@1/imx219@10)
    Modes: 'SRGGB10_CSI2P' : 640x480 1640x1232 1920x1080 3280x2464 
           'SRGGB8' : 640x480 1640x1232 1920x1080 3280x2464 
```

</details>

<details><summary>libcamera-still</summary>

```
Preview window unavailable
[0:04:59.391685919] [1008]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3544-22656360
[0:04:59.461613543] [1009]  WARN RPI raspberrypi.cpp:1241 Mismatch between Unicam and CamHelper for embedded data usage!
[0:04:59.462922573] [1009]  INFO RPI raspberrypi.cpp:1356 Registered camera /base/soc/i2c0mux/i2c@1/imx219@10 to Unicam device /dev/media3 and ISP device /dev/media1
[0:04:59.464260696] [1008]  INFO Camera camera.cpp:1029 configuring streams: (0) 1640x1232-YUV420
[0:04:59.464898183] [1009]  INFO RPI raspberrypi.cpp:760 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected sensor format: 1640x1232-SBGGR10_1X10 - Selected unicam format: 1640x1232-pBAA
```

</details>

RobeeeJay, 2022-07-12:
I've tried these:

```
/usr/local/bin/camera-streamer   -camera-path=/dev/video0   -camera-format=YUYV   -camera-width=1920 -camera-height=1080   -camera-fps=20

/usr/local/bin/camera-streamer   -camera-path=/dev/video0   -camera-format=YUYV   -camera-width=1640 -camera-height=1232   -camera-fps=20

/usr/local/bin/camera-streamer
```

And they all produce the same output...

```
device/v4l2/device.c: CAMERA: Device path=/dev/video0 fd=14 opened
device/buffer_list.c: CAMERA:capture: Using: 1920x1080/YUYV, bytesperline=3840
device/v4l2/device.c: H264: Device path=/dev/video11 fd=19 opened
device/buffer_list.c: H264:output:mplane: Using: 1920x1056/YUYV, bytesperline=3840
device/buffer_list.c: H264:capture:mplane: Using: 1920x1056/H264, bytesperline=0
device/v4l2/device.c: JPEG: Device path=/dev/video31 fd=23 opened
device/buffer_list.c: JPEG:output:mplane: Using: 1920x1056/YUYV, bytesperline=3840
device/buffer_list.c: JPEG:capture:mplane: Using: 1920x1056/JPEG, bytesperline=0
device/v4l2/device.c: CAMERA: ioctl(ret=-1): Can't set FPS
device/v4l2/device_options.c: CAMERA: The 'AfTrigger=1' was failed to find.
device/v4l2/device_options.c: JPEG: Configuring option compressionquality (009d0903) = 80
device/v4l2/device_options.c: H264: Configuring option repeatsequenceheader (009909e2) = 1
device/v4l2/device_options.c: H264: Configuring option videobitratemode (009909ce) = 0
device/v4l2/device_options.c: H264: Configuring option videobitrate (009909cf) = 10000000
device/v4l2/device_options.c: H264: Configuring option repeatsequenceheader (009909e2) = 5000000
device/v4l2/device_options.c: H264: Configuring option h264iframeperiod (00990a66) = 30
device/v4l2/device_options.c: H264: Configuring option h264level (00990a67) = 11
device/v4l2/device_options.c: H264: Configuring option h264profile (00990a6b) = 4
device/v4l2/device_options.c: H264: Configuring option h264minimumqpvalue (00990a61) = 16
device/v4l2/device_options.c: H264: Configuring option h264maximumqpvalue (00990a62) = 32
device/links.c: ?: Link 0: CAMERA:capture[1920x1080/YUYV/3] => [H264:output:mplane[1920x1056/YUYV/3], JPEG:output:mplane[1920x1056/YUYV/3]]
device/links.c: ?: Link 1: H264:capture:mplane[1920x1056/H264/3] => [H264-CAPTURE]
device/links.c: ?: Link 2: JPEG:capture:mplane[1920x1056/JPEG/3] => [JPEG-CAPTURE]
device/buffer_list.c: CAMERA:capture: Streaming started... Was 0 of 3 enqueud
device/buffer_list.c: H264:output:mplane: Streaming started... Was 0 of 3 enqueud
device/buffer_list.c: JPEG:output:mplane: Streaming started... Was 0 of 3 enqueud
device/buffer_list.c: H264:capture:mplane: Streaming started... Was 0 of 3 enqueud
device/buffer_list.c: JPEG:capture:mplane: Streaming started... Was 0 of 3 enqueud
http/http.c: HTTP8080/0: Client connected 192.168.1.103.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1): Can't queue buffer.
http/http.c: HTTP8080/0: Client disconnected 192.168.1.103.
```

ayufan, 2022-07-12:
Can you try using via `libcamera` which is the desired approach for ISP-required cameras?

```bash
/usr/local/bin/camera-streamer -camera-path=/base/soc/i2c0mux/i2c@1/imx219@10 -camera-type=libcamera -camera-format=YUYV -camera-width=3280 -camera-height=2464 -camera-fps=30 -camera-nbufs=2 -camera-high_res_factor=2 -camera-low_res_factor=4
```

RobeeeJay, 2022-07-12:
That works wonderfully! Thank you so much for your help! I feel like people should be using this instead of mjpeg-streamer now, the cpu is so much lower because it uses HW.

ayufan, 2022-07-12:
I think this requires docs to ease different setups, but feel free spread the word :)

RobeeeJay, 2022-07-12:
Hmm I actually think it's the camera-path in the service file that meant it didn't work, so I'm not sure if I should add a new service or fix the existing one. It looks like the IMX519 is for the Arducam and not the Pi v2.1.

ayufan, 2022-07-12:
> It looks like the IMX519 is for the Arducam and not the Pi v2.1.

Yes, the current setting is for Arducam 16MP. Your `camera-path` and sensor resolution is different. I guess the best is to have a single `service` per a supported camera exact model, that can be adapted as far as resolution goes.