ayufan/camera-streamer #5: Camera Device not Found

gallitim, 2022-07-18:
Hello,
I have successfully compiled the software however when I try to use it I get this error log in the journal:

░░ Subject: A start job for unit camera-streamer-arducam-16MP.service has finished successfully
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ A start job for unit camera-streamer-arducam-16MP.service has finished successfully.
░░ 
░░ The job identifier is 2581.
Jul 18 09:02:55 raspberrypi camera-streamer[2416]: [0:02:38.646971896] [2416]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3700-f30ad033
Jul 18 09:02:55 raspberrypi camera-streamer[2416]: [0:02:38.682950964] [2427]  WARN RPI raspberrypi.cpp:1252 Mismatch between Unicam and CamHelper for embedded data usage!
Jul 18 09:02:55 raspberrypi camera-streamer[2416]: [0:02:38.683611493] [2427]  INFO RPI raspberrypi.cpp:1368 Registered camera /base/soc/i2c0mux/i2c@1/imx219@10 to Unicam device /dev/media2 and ISP device /dev/media0
Jul 18 09:02:55 raspberrypi camera-streamer[2416]: device/libcamera/device.cc: CAMERA: Available cameras (1)
Jul 18 09:02:55 raspberrypi camera-streamer[2416]: device/libcamera/device.cc: CAMERA: - /base/soc/i2c0mux/i2c@1/imx219@10
Jul 18 09:02:55 raspberrypi camera-streamer[2416]: device/libcamera/device.cc: CAMERA: Camera `/base/soc/i2c0mux/i2c@1/imx519@1a` was not found.
Jul 18 09:02:55 raspberrypi camera-streamer[2416]: device/device.c: CAMERA: Can't open device: /base/soc/i2c0mux/i2c@1/imx519@1a
Jul 18 09:02:55 raspberrypi systemd[1]: camera-streamer-arducam-16MP.service: Main process exited, code=exited, status=255/EXCEPTION
░░ Subject: Unit process exited
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ An ExecStart= process belonging to unit camera-streamer-arducam-16MP.service has exited.
░░ 
░░ The process' exit code is 'exited' and its exit status is 255.
Jul 18 09:02:55 raspberrypi systemd[1]: camera-streamer-arducam-16MP.service: Failed with result 'exit-code'.
░░ Subject: Unit failed
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ The unit camera-streamer-arducam-16MP.service has entered the 'failed' state with result 'exit-code'.

It seems like its looking for an imx519 sensor when an imx219 is plugged in instead, is there a way to change this such that the software looks for an imx219 sensor instead? Thanks in advance for the help.

ayufan, 2022-07-18:
> It seems like its looking for an imx519 sensor when an imx219 is plugged in instead, is there a way to change this such that the software looks for an imx219 sensor instead? Thanks in advance for the help.

Of course. Edit service or try to use https://github.com/ayufan-research/camera-streamer/blob/master/service/camera-streamer-pi219-8MP.service.

ayufan, 2022-07-18:
> Thanks for the help! Now that it works, do I edit this service file to
> change parameters like fps and resolution? Or is there another method to do
> this? I am just looking for ways to smoothen out the stream to about 720p
> with a smooth fps.

Yes. Always edit systemd service

closed 2022-07-18 by gallitim

ayufan, 2022-07-18:
How does it work?

gallitim, 2022-07-19:
It works great! Thanks a lot for the help