ayufan/camera-streamer #7: USB Camera Service File Help

gallitim, 2022-08-09:
Hello,
I want to continue to use this software but on a usb camera instead, but when I try to start the usb camera service file, I end up with this message:

pi@raspberrypi:~ $ sudo systemctl start /camera-streamer/service/camera-streamer-usb-cam
Failed to start camera\x2dstreamer-service-camera\x2dstreamer\x2dusb\x2dcam.mount: Unit camera\x2dstreamer-service-camera\x2dstreamer\x2dusb\x2dcam.mount not found.

Am I doing something wrong here? Or is the way to setup a usb camera with this a bit different? Any help is appreciated, as well as any general tips on how to use the streamer with a usb camera.

closed 2022-08-09 by gallitim

ayufan, 2022-08-10:
Did you solve this?

gallitim, 2022-08-10:
yeah, just made a mistake