ayufan/camera-streamer #10: Unsupported Camera Format

Tiim12312, 2022-08-28:
Hi,
I get this when trying to run any of the scripts to start. I'm using the Arducam 16MP.
[image]

This is the output of checking the formats available.

[image]

What is the issue here?
Thanks!

ayufan, 2022-08-28:
The best is to use preconfigured service: https://github.com/ayufan-research/camera-streamer#use-a-preconfigured-systemd-services. There's one specifically for arducam. You might modify it to suite your needs.

Tiim12312, 2022-08-28:
Thank you for the fast response. This is the output I get when running it that way.
[image]

I have updated the kernel, this is the kernel version.

[image]

ayufan, 2022-08-28:
Can you show your `/boot/config.txt`?

Tiim12312, 2022-08-28:
[image]

I added this line

ayufan, 2022-08-28:
OK. I believe you need to change it to `dtoverlay=imx519` or `dtoverlay=imx519,media-controller=1`.

Tiim12312, 2022-08-28:
I figured out my issue, I did two things. First I didn't realize I needed to install the arducam 16MP drivers from their site.
https://www.arducam.com/docs/cameras-for-raspberry-pi/raspberry-pi-libcamera-guide/how-to-use-arducam-16mp-camera-on-rapberry-pi/

Second I hadn't placed the dtoverlay line under the [all] section of the /boot/config.txt. Not sure which of those things fixed it but seems to be working now! Thank you!!

ayufan, 2022-08-28:
If you don't install 16MP drivers you will not have auto-focus.

The second change is truly required to enable the sensor.

closed 2022-08-28 by ayufan