ayufan/camera-streamer #20: ISP Can't queue buffer.

Trackhe, 2022-10-26:
Hey, when i try to run the raspi v2 cam with that script i get that error: ```device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1): Can't queue buffer.```

script:
```
#!/bin/bash

SCRIPT_DIR=$( cd -- "$( dirname -- "${BASH_SOURCE[0]}" )" &> /dev/null && pwd )
cd "$SCRIPT_DIR"

CAMERA_PATH=/dev/v4l/by-path/platform-fe800000.csi-video-index0

if [[ "$1" == "dump" ]]; then
  shift
  libcamera-still "$@"
  echo
  v4l2-ctl -d /dev/v4l-subdev0 \
    -C exposure -C vertical_blanking -C analogue_gain -C digital_gain | \
    sed -e 's/ //g' -e 's/:/=/g' -e 's/^/-camera-options=/g' -e 's/$/ \\/g'
  v4l2-ctl -d /dev/video13 -C red_balance -C colour_correction_matrix \
    -C black_level -C green_equalisation -C gamma -C denoise -C sharpen \
    -C defective_pixel_correction -C colour_denoise | \
    sed -e 's/ //g' -e 's/:/=/g' -e 's/^/-camera-isp.options=/g' -e 's/$/ \\/g'
  exit 0
fi

set -xeo pipefail
make -j$(nproc)
$GDB ./camera-streamer -camera-path=/dev/v4l/by-path/platform-fe800000.csi-video-index0 \
   -camera-format=YUYV \
#  -camera-options=vertical_blanking=728 \
#  -camera-options=exposure=2444 \
#  -camera-options=analogue_gain=600 \
#  -camera-options=digital_gain=256 \
#  -camera-isp.options=digital_gain=2015 \
#  -camera-isp.options=red_balance=1852 \
#  -camera-isp.options=blue_balance=2146 \
#  -camera-isp.options=colour_correction_matrix=1,0,0,0,146,4,0,0,232,3,0,0,211,255,255,255,232,3,0,0,132,255,255,255,232,3,0,0,91,255,255,255,232,3,0,0,34>
#  -camera-isp.options=black_level=1,0,0,0,0,16,0,16,0,16,38,134 \
#  -camera-isp.options=green_equalisation=1,0,0,0,89,2,0,0,47,0,0,0,232,3,0,0 \
#  -camera-isp.options=gamma=1,0,0,0,0,0,0,4,0,8,0,12,0,16,0,20,0,24,0,28,0,32,0,36,0,40,0,44,0,48,0,52,0,56,0,60,0,64,0,72,0,80,0,88,0,96,0,104,0,112,0,12>
#  -camera-isp.options=denoise=1,0,0,0,0,0,0,0,240,30,0,0,232,3,0,0,238,2,0,0,232,3,0,0 \
#  -camera-isp.options=sharpen=1,0,0,0,208,7,0,0,232,3,0,0,244,1,0,0,232,3,0,0,244,1,0,0,232,3,0,0 \
#  -camera-isp.options=defective_pixel_correction=1,0,0,0,1,0,0,0 \
#  -camera-isp.options=colour_denoise=0,0,0,0,127,0,0,0 \
  "$@"
```

Trackhe, 2022-10-26:
dont worry, my bad. Thank you for your work, maybe you can write the http option in the readme too, that you provide her : https://github.com/ayufan-research/camera-streamer/blob/f08421a796436d14d07894ebd023566e7760a597/cmd/camera-streamer/opts.c because i needed to search in code for that. And i need 2 because of 2 cams. Or is there a option to make 2 streams with one execution?

closed 2022-10-26 by Trackhe

Breezebuilder, 2022-11-13:
@Trackhe Could you please provide details on what the cause of the problem was?

Trackhe, 2022-11-13:
I had to use 
```
[all] # or [pi4] or whatever else
dtoverlay=vc4-kms-v3d,cma-256
dtoverlay=vc4-kms-dsi-7inch
dtoverlay=imx219
```
The Display was the problem for me, and now only a problem with the kernel 5.19 is a problem for me because of not working touch.