ayufan/camera-streamer #21: Input/Output error

Trackhe, 2022-10-27:
Yesterday the services was working with that command, but today. This error, i dont understand why.

```
/usr/local/bin/camera-streamer   -camera-path=/base/soc/i2c0mux/i2c@0/imx219@10   -camera-type=libcamera   -camera-format=YUYV   -camera-nbufs=2   -http=8081   -camera-width=1640   -camera-height=1232   -camera-options=rotation=90   -camera-fps=15 -log-verbose
device/v4l2/device_list.c: bcm2835-isp: Device (/dev/video23) does not support capture (skipping)
device/v4l2/device_list.c: bcm2835-isp: Device (/dev/video20) does not support capture (skipping)
device/v4l2/device_list.c: bcm2835-isp: Device (/dev/video16) does not support capture (skipping)
device/v4l2/device_list.c: bcm2835-isp: Device (/dev/video13) does not support capture (skipping)
[0:04:42.543821854] [2083]  INFO Camera camera_manager.cpp:293 libcamera v0.0.1+21-7c855784
[0:04:42.581297734] [2099]  WARN RPI raspberrypi.cpp:1297 Mismatch between Unicam and CamHelper for embedded data usage!
[0:04:42.582301378] [2099]  INFO RPI raspberrypi.cpp:1414 Registered camera /base/soc/i2c0mux/i2c@0/imx219@10 to Unicam device /dev/media2 and ISP device /dev/media4
[0:04:42.591641765] [2099]  WARN RPI raspberrypi.cpp:1297 Mismatch between Unicam and CamHelper for embedded data usage!
[0:04:42.592775333] [2099]  INFO RPI raspberrypi.cpp:1414 Registered camera /base/soc/i2c0mux/i2c@1/imx219@10 to Unicam device /dev/media3 and ISP device /dev/media5
device/libcamera/device.cc: CAMERA: Device path=/base/soc/i2c0mux/i2c@0/imx219@10 opened
[0:04:42.593671999] [2083]  INFO Camera camera.cpp:1026 configuring streams: (0) 1640x1232-YUYV
[0:04:42.594084341] [2099]  INFO RPI raspberrypi.cpp:800 Sensor: /base/soc/i2c0mux/i2c@0/imx219@10 - Selected sensor format: 1640x1232-SBGGR10_1X10 - Selected unicam format: 1640x1232-pBAA
device/buffer_list.c: CAMERA:capture: Using: 1640x1232/YUYV, bytesperline=3328
device/v4l2/device.c: H264: Device path=/dev/video11 fd=42 opened
device/v4l2/device_media.c: H264: Opened '/dev/media1' (fd=43)
device/v4l2/device_media.c: H264: Link '../../devices/platform/soc/fe00b840.mailbox/bcm2835-codec/video4linux/video10' does not contain '/v4l-subdev'
device/v4l2/device_options.c: H264: The 'codeccontrols' is read-only
device/v4l2/device_options.c: H264: Available control: videobitratemode (009909ce, type=3)
device/v4l2/device_options.c: H264: Available control: videobitrate (009909cf, type=1)
device/v4l2/device_options.c: H264: Available control: sequenceheadermode (009909d8, type=3)
device/v4l2/device_options.c: H264: Available control: repeatsequenceheader (009909e2, type=2)
device/v4l2/device_options.c: H264: Available control: forcekeyframe (009909e5, type=4)
device/v4l2/device_options.c: H264: Available control: h264minimumqpvalue (00990a61, type=1)
device/v4l2/device_options.c: H264: Available control: h264maximumqpvalue (00990a62, type=1)
device/v4l2/device_options.c: H264: Available control: h264iframeperiod (00990a66, type=1)
device/v4l2/device_options.c: H264: Available control: h264level (00990a67, type=3)
device/v4l2/device_options.c: H264: Available control: h264profile (00990a6b, type=3)
device/v4l2/buffer_list.c: H264:output:mplane: Adapting size to 32x32 block: 1640x1232 vs 1632x1216
device/buffer_list.c: H264:output:mplane: Using: 1632x1216/YUYV, bytesperline=3328
device/v4l2/buffer_list.c: H264:capture:mplane: Adapting size to 32x32 block: 1632x1216 vs 1632x1216
device/buffer_list.c: H264:capture:mplane: Using: 1632x1216/H264, bytesperline=0
device/v4l2/device.c: JPEG: Device path=/dev/video31 fd=45 opened
device/v4l2/device_media.c: JPEG: Opened '/dev/media1' (fd=46)
device/v4l2/device_media.c: JPEG: Link '../../devices/platform/soc/fe00b840.mailbox/bcm2835-codec/video4linux/video10' does not contain '/v4l-subdev'
device/v4l2/device_options.c: JPEG: The 'jpegcompressioncontrols' is read-only
device/v4l2/device_options.c: JPEG: Available control: compressionquality (009d0903, type=1)
device/v4l2/buffer_list.c: JPEG:output:mplane: Adapting size to 32x32 block: 1640x1232 vs 1632x1216
device/buffer_list.c: JPEG:output:mplane: Using: 1632x1216/YUYV, bytesperline=3328
device/v4l2/buffer_list.c: JPEG:capture:mplane: Adapting size to 32x32 block: 1632x1216 vs 1632x1216
device/buffer_list.c: JPEG:capture:mplane: Using: 1632x1216/JPEG, bytesperline=0
device/v4l2/device_options.c: JPEG: Configuring option compressionquality (009d0903) = 80
device/v4l2/device_options.c: H264: Configuring option repeatsequenceheader (009909e2) = 1
device/v4l2/device_options.c: H264: Configuring option videobitratemode (009909ce) = 0
device/v4l2/device_options.c: H264: Configuring option videobitrate (009909cf) = 2000000
device/v4l2/device_options.c: H264: Configuring option repeatsequenceheader (009909e2) = 5000000
device/v4l2/device_options.c: H264: Configuring option h264iframeperiod (00990a66) = 30
device/v4l2/device_options.c: H264: Configuring option h264level (00990a67) = 11
device/v4l2/device_options.c: H264: Configuring option h264profile (00990a6b) = 4
device/v4l2/device_options.c: H264: Configuring option h264minimumqpvalue (00990a61) = 16
device/v4l2/device_options.c: H264: Configuring option h264maximumqpvalue (00990a62) = 32
device/links.c: ?: Link 0: CAMERA:capture[1640x1232/YUYV/2] => [H264:output:mplane[1632x1216/YUYV/2], JPEG:output:mplane[1632x1216/YUYV/2]]
device/links.c: ?: Link 1: H264:capture:mplane[1632x1216/H264/2] => [H264-CAPTURE]
device/links.c: ?: Link 2: JPEG:capture:mplane[1632x1216/JPEG/2] => [JPEG-CAPTURE]
[0:04:43.646075325] [2099] ERROR V4L2 v4l2_videodevice.cpp:1903 /dev/video0[23:cap]: Failed to start streaming: Input/output error
device/libcamera/buffer_list.cc: CAMERA:capture: Failed to start camera.
device/links.c: CAMERA:capture: Failed to start streaming
```

Trackhe, 2022-10-27:
Seems to be a problem on the raspberry pi with the screen attached.

closed 2022-10-27 by Trackhe

ayufan, 2022-10-27:
Yes, if you have dsi or dpi screen you need to configure it properly. What kind of screen you have?

Trackhe, 2022-10-27:
@ayufan https://biqu.equipment/collections/lcd/products/bigtreetech-pi-tft43-v2-0-screen-board?variant=39337700851810 the Bigtreetech Pi TFT70 V2.0

ayufan, 2022-10-28:
Nah. If you have the `Bigtreetech Pi TFT70 V2.0` it works as Raspberry PI display. The problem is that camera requires i2c, the same as a display.

The simplest way to configure it is to have the `/boot/config.txt` similar to this. It should be good to have both working just fine. You need to use `kms` driver that allows for both of them share relevant bits:

```
dtoverlay=vc4-kms-v3d,cma-256
dtoverlay=vc4-kms-dsi-7inch
dtoverlay= imx219
```

Trackhe, 2022-10-29:
Thank you that hint was so good, because i tried already vc4-kms and fkms - v3d but i change it only in the pi4 section, but this has no effect because of the cm4 module. So i change it to

```
# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
dtparam=spi=on

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

[cm4]
dtoverlay=vc4-kms-v3d,cma-256
dtoverlay=vc4-kms-dsi-7inch

dtoverlay=dwc2,dr_mode=host
dtoverlay=imx219,cam1
dtoverlay=imx219,cam0

# Enable Hardware UART for Serial Communication
enable_uart=1
dtoverlay=disable-bt

# Enable Raspicam devices at boot
start_x=1
gpu_mem=256
```

So with that the Display and Both cams are working but the touch isnt working, but this seems a bug that is confirmed. https://bugs.launchpad.net/ubuntu/+source/linux-raspi/+bug/1970603