ayufan/camera-streamer #23: Issue with Manual Focus on Arducam 16MP

jinnn1989, 2022-11-26:
Hello,

Thanks for this fantastic streamer. I am not very much into code and development. I just a normal 3d printer guy. 

I got this install and working on Klipper. but the only issue which i am facing is focus. I am not able to focus using the guide on the main page. Please can you provide more detailed guide on how to get focus working.

Thank you

kbingham, 2022-11-28:
We have controls for AF with libcamera, but they're not actually merged yet. I'm trying to get them progressing.

jinnn1989, 2022-11-28:
Thats great. I am trying a work around. Trying to install Mainsail and Octoprint on same pi and then controlling the focus using Octoprint, while timelapse using Klipper / Mainsail. Still its getting installed. let me see how this goes

ayufan, 2022-11-28:
@jinnn1989 The auto focus will work out of box, but you unfortunately have to install ArduCAM libcamera version. This of course only works on RPI.

kbingham, 2022-11-28:
@ayufan Do you already use the focus controls for retrigger or manual focus from Arducam? or are you just relying on their continuous autofocus?

jinnn1989, 2022-11-28:
I use Manual focus. I havent tried AF or Continuous Focus yet.

ayufan, 2022-11-28:
@kbingham They provide `AfTrigger=1` to force refocus, this also to some extent (quirky way) the last I checked controlled if continuous autofocus is there. The manual focus can be set via `focus_absolute=` from v4l2-ctl.

jinnn1989, 2022-11-28:
My workaround did not work. The issue was, I had first installed Octopi RC2 Image and then installed Klipper using Kiauh. this broke both my Octopi and Klipper. 

Now again installing Raspberry pi image and then would install Klipper and Octopi using Kiauh. so that they have different port number assigned. 

if you guys have better idea to control focus. do let me know. I can do the test as required.

ayufan, 2022-11-28:
@jinnn1989 Do you want fixed or auto? I usually for printer run on fixed, to avoid auto.

jinnn1989, 2022-11-28:
I want it to be fixed. Auto ruin the Timelapse

ayufan, 2022-11-28:
Do you see `focus_absolute` under `/option`?

jinnn1989, 2022-11-28:
Not sure how to access this options. can you guide me here. we even had this chat on Discord Mainsail forum

ayufan, 2022-11-28:
How you see the content from `camera-streamer`?

jinnn1989, 2022-11-28:
I never tried to, as installing and starting the streamer worked for me. I will give you detailed logs / errors in few mins

jinnn1989, 2022-11-28:
pi@raspberrypi:~/camera-streamer $ ./camera-streamer -camera-fps=60
bind: Address already in use

I am not sure if this is the correct way of passing the command. but this is the error which i get

jinnn1989, 2022-11-28:
No options passed.

Set: /option?name=value

CAMERA Properties:
- property: SensorSensitivity (00000009, type=5): 1.000000
- property: ScalerCropMaximum (00000008, type=7): (0, 0)/4656x3496
- property: ColorFilterArrangement (0000000a, type=3): 0
- property: PixelArrayActiveAreas (00000007, type=7): [ (8, 48)/4656x3496 ]
- property: PixelArraySize (00000005, type=8): 4656x3496
- property: Rotation (00000002, type=3): 0
- property: Location (00000001, type=3): 2
- property: Model (00000003, type=6): imx519

CAMERA Options:
- available option: AwbEnable (0000000c, type=1): [false..true]
- available option: LensPosition (00000023, type=5): [0.000000..1000.000000]
- available option: FrameDurationLimits (00000019, type=4): [33328..112854017]
- available option: ExposureValue (00000006, type=5): [-8.000000..8.000000]
- available option: NoiseReductionMode (00000027, type=3): [0..4]
- available option: ColourGains (0000000f, type=5): [0.000000..32.000000]
- available option: AfMode (0000001c, type=3): [0..2]
- available option: Brightness (00000009, type=5): [-1.000000..1.000000]
- available option: AeEnable (00000001, type=1): [false..true]
- available option: AnalogueGain (00000008, type=5): [1.000000..16.000000]
- available option: Saturation (00000011, type=5): [0.000000..32.000000]
- available option: ExposureTime (00000007, type=3): [13..112853586]
- available option: AeMeteringMode (00000003, type=3): [0..3]
- available option: ColourCorrectionMatrix (00000015, type=5): [-16.000000..16.000000]
- available option: AeConstraintMode (00000004, type=3): [0..3]
- available option: AeExposureMode (00000005, type=3): [0..3]
- available option: AfTrigger (00000021, type=3): [0..1]
- available option: AwbMode (0000000d, type=3): [0..7]
- available option: ScalerCrop (00000016, type=7): [(0, 0)/128x128..(0, 0)/4656x3496]
- available option: Sharpness (00000013, type=5): [0.000000..16.000000]
- available option: Contrast (0000000a, type=5): [0.000000..32.000000]

ISP-LOW Options:
- available option: horizontalflip (00980914, type=2): [0..1]
- available option: verticalflip (00980915, type=2): [0..1]

JPEG Options:
- available option: compressionquality (009d0903, type=1): [1..100]

JPEG-LOW Options:
- available option: compressionquality (009d0903, type=1): [1..100]

H264 Options:
- available option: videobitratemode (009909ce, type=3): [0..1]
		0: Variable Bitrate
		1: Constant Bitrate
- available option: videobitrate (009909cf, type=1): [25000..25000000]
- available option: sequenceheadermode (009909d8, type=3): [0..1]
		0: Separate Buffer
		1: Joined With 1st Frame
- available option: repeatsequenceheader (009909e2, type=2): [0..1]
- available option: forcekeyframe (009909e5, type=4): button
- available option: h264minimumqpvalue (00990a61, type=1): [0..51]
- available option: h264maximumqpvalue (00990a62, type=1): [0..51]
- available option: h264iframeperiod (00990a66, type=1): [0..2147483647]
- available option: h264level (00990a67, type=3): [0..15]
		0: 1
		1: 1b
		2: 1.1
		3: 1.2
		4: 1.3
		5: 2
		6: 2.1
		7: 2.2
		8: 3
		9: 3.1
		10: 3.2
		11: 4
		12: 4.1
		13: 4.2
		14: 5
		15: 5.1
- available option: h264profile (00990a6b, type=3): [0..4]
		0: Baseline
		1: Constrained Baseline
		2: Main
		4: High

H264-LOW Options:
- available option: videobitratemode (009909ce, type=3): [0..1]
		0: Variable Bitrate
		1: Constant Bitrate
- available option: videobitrate (009909cf, type=1): [25000..25000000]
- available option: sequenceheadermode (009909d8, type=3): [0..1]
		0: Separate Buffer
		1: Joined With 1st Frame
- available option: repeatsequenceheader (009909e2, type=2): [0..1]
- available option: forcekeyframe (009909e5, type=4): button
- available option: h264minimumqpvalue (00990a61, type=1): [0..51]
- available option: h264maximumqpvalue (00990a62, type=1): [0..51]
- available option: h264iframeperiod (00990a66, type=1): [0..2147483647]
- available option: h264level (00990a67, type=3): [0..15]
		0: 1
		1: 1b
		2: 1.1
		3: 1.2
		4: 1.3
		5: 2
		6: 2.1
		7: 2.2
		8: 3
		9: 3.1
		10: 3.2
		11: 4
		12: 4.1
		13: 4.2
		14: 5
		15: 5.1
- available option: h264profile (00990a6b, type=3): [0..4]
		0: Baseline
		1: Constrained Baseline
		2: Main
		4: High

ayufan, 2022-11-28:
Maybe `LensPosition` I need to check on mine later.

jinnn1989, 2022-11-28:
Finally it worked. Thank for your help. I am able to focus now. 

Just one more question. Does the value get saved once we hit enter. or after reboot this will reset ?

jinnn1989, 2022-11-28:
I did edit the libcamera_camera.sh and set the -camera-options=LensPosition=200

but i do get the below error. Any idea what the error could be "bind: Address already in use"

pi@raspberrypi:~ $ cd camera-streamer/
pi@raspberrypi:~/camera-streamer $ tools/libcamera_camera.sh -log-verbose
++ nproc
+ make -j4
make: 'camera-streamer' is up to date.
+ ./camera-streamer -camera-path=/base/soc/i2c0mux/i2c@1/imx519@1a -camera-type=libcamera -camera-format=YUYV -camera-options=LensPosition=200 -log-verbose
bind: Address already in use

ayufan, 2022-11-28:
You have another service running on a port `8080`.

jinnn1989, 2022-11-28:
I have freshly installed Klipper and mainsail, no other services I have installed. only the Arducam drivers and the camera streamer. Not sure what is being used on port 8080

Also I am able to see the options on  http://192.168.100.121:8080/

ayufan, 2022-11-28:
So, you have another process with camera-streamer running. Another terminal?

```
netstat -lnp | grep 8080
```

jinnn1989, 2022-11-28:
pi@raspberrypi:~/camera-streamer $ sudo netstat -lnp | grep 8080
tcp        0      0 0.0.0.0:8080            0.0.0.0:*               LISTEN      519/camera-streamer 

Only a single service

ayufan, 2022-11-28:
So, it is running already. Is it run via `systemd` service?

ayufan, 2022-11-28:
@jinnn1989 But did you see `auto_focus=1` at the end?

ayufan, 2022-11-28:
Unsure then. You would have to see `journalctl` log to see how service started.

jinnn1989, 2022-11-28:
Sure will check the logs and report if I find anything.

robbalmbra, 2022-12-01:
I did notice while manually running camera-streamer using Lens Position:

device/v4l2/device_options.c: CAMERA: The 'LensPosition=400' was failed to find.

Specifying AFTrigger:
device/v4l2/device_options.c: CAMERA: The 'AfTrigger=1' was failed to find.

jinnn1989, 2022-12-03:
go to camera-streamer folder > then Services , edit the 16mp arducam file 

and try this 

 -camera-options=lensposition=360 \
  -camera-options=sharpness=2 \
  -camera-options=saturation=1.1 \

this is working for me,

if you want to change the lensposition via browser check the options menu, you will find the key to be edited

@robbalmbra

issic, 2022-12-06:
I have problem that focus or autofocus doesn't move at all on **Arducam B0389**. 

I found the IMX519

`[    6.638070] imx519: loading out-of-tree module taints kernel.
[    6.687138] imx519 10-001a: Device found is imx519
[    6.687901] imx519 10-001a: Consider updating driver imx519 to match on endpoints`

I have I2C for the CSI device
`[    3.113811] i2c_dev: i2c /dev entries driver
[    4.799193] i2c i2c-22: Added multiplexed i2c bus 0
[    4.801320] i2c 10-001a: Fixing up cyclic dependency with fe801000.csi
[    4.802044] i2c i2c-22: Added multiplexed i2c bus 10`

I do see Lens Position on camera-streamer output.
`The 'LensPosition' was set to '0'.

Set: /option?name=value

CAMERA Properties:
- property: SensorSensitivity (00000009, type=5): 1.000000
- property: ScalerCropMaximum (00000008, type=7): (0, 0)/4656x3496
- property: ColorFilterArrangement (0000000a, type=3): 0
- property: PixelArrayActiveAreas (00000007, type=7): [ (8, 48)/4656x3496 ]
- property: PixelArraySize (00000005, type=8): 4656x3496
- property: Rotation (00000002, type=3): 0
- property: Location (00000001, type=3): 2
- property: Model (00000003, type=6): imx519

CAMERA Options:
- available option: ExposureValue (00000006, type=5): [-8.000000..8.000000]
- available option: AwbMode (0000000d, type=3): [0..7]
- available option: AwbEnable (0000000c, type=1): [false..true]
- available option: Sharpness (00000013, type=5): [0.000000..16.000000]
- available option: Saturation (00000011, type=5): [0.000000..32.000000]
- available option: Contrast (0000000a, type=5): [0.000000..32.000000]
- available option: AeMeteringMode (00000003, type=3): [0..3]
- available option: LensPosition (00000023, type=5): [0.000000..1000.000000]
- available option: AeEnable (00000001, type=1): [false..true]
- available option: FrameDurationLimits (00000019, type=4): [33328..112854017]
- available option: ExposureTime (00000007, type=3): [269..112853586]
- available option: AfMode (0000001c, type=3): [0..2]
- available option: AeConstraintMode (00000004, type=3): [0..3]
- available option: AeExposureMode (00000005, type=3): [0..3]
- available option: AfTrigger (00000021, type=3): [0..1]
- available option: NoiseReductionMode (00000027, type=3): [0..4]
- available option: ScalerCrop (00000016, type=7): [(0, 0)/128x128..(0, 0)/4656x3496]
- available option: ColourCorrectionMatrix (00000015, type=5): [-16.000000..16.000000]
- available option: Brightness (00000009, type=5): [-1.000000..1.000000]
- available option: ColourGains (0000000f, type=5): [0.000000..32.000000]
- available option: AnalogueGain (00000008, type=5): [1.000000..16.000000]`

and this is my start script

`./camera-streamer -camera-path=/base/soc/i2c0mux/i2c@1/imx519@1a -camera-type=libcamera -camera-format=YUYV -camera-width=2328 -camera-height=1748 -camera-fps=30 -camera-nbufs=3 -camera-high_res_factor=1.4 -camera-low_res_factor=2.42 -camera-options=brightness=0.1 -camera-vflip=1 -camera-hflip=1 -camera-options=lensposition=360 -camera-auto_focus=0`

issic, 2022-12-06:
And I might be idiot, cos this modul is probably fix focus.

AMFkuna, 2022-12-12:
I wrote little script for manual focus
#!/bin/bash

VALUE=$(printf "%d\n" "$(i2ctransfer -y 11 w1@0x0c 0 r1)")

case $1 in
        [i])    if [[ $VALUE -lt 256 ]]; then
                        ((VALUE ++))
                        i2ctransfer -y 11 w4@0x0c 0 "$VALUE" 0 0
                fi
                ;;

        [d])    if [[ $VALUE -gt -1 ]]; then
                        ((VALUE --))
                        i2ctransfer -y 11 w4@0x0c 0 "$VALUE" 0 0
                fi
                ;;

        [v])    if [[ $2 -lt 256 ]] && [[ $2 -gt -1 ]]; then
                        VALUE=$2
                        i2ctransfer -y 11 w4@0x0c 0 "$VALUE" 0 0
                fi
                ;;
        *)      echo "Arducam 16MP IMX519 manual focus script by AmfKuna#5125"
                echo
                echo "Usage: arducamfocu.sh i|d|v 0-255"
                echo "i - increment, d - decrement, v - 0-255"
                ;;
esac

echo Focus "$VALUE"

and some macros to klipper

[gcode_macro ACAM_FOCUS_PLUS]
description: Focus Arducam plus
gcode:
    RUN_SHELL_COMMAND CMD=arducamfocus PARAMS=i

[gcode_macro ACAM_FOCUS_MINUS]
description: Focus Arducam minus
gcode:
    RUN_SHELL_COMMAND CMD=arducamfocus PARAMS=d

[gcode_macro ACAM_FOCUS_VALUE]
description: Focus Arducam to desired value
gcode:
    {% set focus = params.VALUE|default(115) %}
    RUN_SHELL_COMMAND CMD=arducamfocus PARAMS="v "{focus}

[gcode_shell_command arducamfocus]
command: bash /home/pi/arducamfocus.sh
timeout: 5.0

jinnn1989, 2022-12-14:
@AMFkuna can you provide more detailed tutorials on how to set this up, Marcos are fine but the bash script and if there is anything else which needs to be setupup. I am not professional linux, so I need to guide to implement this.

AMFkuna, 2023-01-09:
> @AMFkuna can you provide more detailed tutorials on how to set this up, Marcos are fine but the bash script and if there is anything else which needs to be setupup. I am not professional linux, so I need to guide to implement this.

You need to install ayucamstream branch of crowsnest to support camera-streamer https://github.com/mainsail-crew/crowsnest/tree/feat/ayucamstream

next 
in crowsnest.conf
[crowsnest]
log_path: ~/printer_data/logs/crowsnest.log
log_level: verbose                      # Valid Options are quiet/verbose/debug
delete_log: false                       # Deletes log on every restart, if set to true

[cam 1]
mode: multi                             # mjpg/rtsp
port: 8080                              # Port
device: /dev/video0                     # See Log for available ...
resolution: 2328x1748                    # widthxheight format
max_fps: 10                             # If Hardware Supports this it will be forced, ohterwise ignored/coerced.
custom_flags: -camera-path=/base/soc/i2c0mux/i2c@1/imx519@1a -camera-type=libcamera -camera-format=YUYV -camera-nbufs=2 -camera-high_res_factor=1.5 -camera-low_res_factor=3.0 -camera-options=brightness=0.1 -camera-options=sharpness=2

next
you need to install gcode_shell_command from KIAUH script
and put this into printer.cfg

[gcode_shell_command arducamfocus]
command: bash /home/pi/arducamfocus.sh
timeout: 5.0
verbose: True

If you are not using crowsnest, just run camera-streamer service from camera-streamer/service/camera-streamer-arducam-16MP.service