ayufan/camera-streamer #27: Raspberry Pi HQ Camera

ETE-Design, 2023-01-07:
Hi Kamil.

Support for the Official Raspberry Pi HQ Camera, is that something you can add to the Streamer? Or is it already supported?

Regards

kbingham, 2023-01-09:
It should work directly. You'll possibly need to identify the camera name. (Try cam --list, or find it from libcamera-hello).
tools/libcamera_camera.sh shows how to call the streamer.

I've just used this camera-streamer with the new Camera V3, and it worked out of the box, so I don't expect there would be anything specific for the HQ camera.

ayufan, 2023-01-09:
Yup. It should just work. Also with latest changes to `libcamera` it is also very likely that all ArduCAMs can be supported "out of box".

kbingham, 2023-01-09:
Not quite - it's getting close. But the Arducam cameras may need a bit more work for AF. Also - they only have contrast based AF - while the V3 is PDAF. So there are some subtle and not so subtle differences to keep aware of.

Arducam still need to do some more work on upstreaming their camera support.

ETE-Design, 2023-01-09:
@ayufan Will it also work with "High-performance mode via ISP for CSI" to get even better FPS?

ayufan, 2023-01-09:
@ETE-Design It should, but then you have to manually setup ISP first.

ETE-Design, 2023-01-09:
@ayufan I'm really not into this stuff... Is there a guide for that somewhere?

ayufan, 2023-01-09:
Sorry, no, as this is not trivial and libcamera offers quite amount of features related also to brightness control.

kbingham, 2023-01-09:
We'd be better off fixing the latency issue that shows up for libcamera. I would expect this is something occurring in the raspberry pi pipeline handler - and I think the RPi team are working on ways to make that configurable to reduce it.

MAM-Org, 2023-02-24:
Hi there, I have ordered the 64MP beast (I'd love to use ePTZ)... but in the meantime the only cam' I had was a HQ Camera.

I was surprised you didn't have a service file for the HQ cam already, so here are the necessary infos needed to create one :
```
# libcamera-still --list-cameras
Available cameras
-----------------
0 : imx477 [4056x3040] (/base/soc/i2c0mux/i2c@1/imx477@1a)
    Modes: 'SBGGR10_CSI2P' : 1332x990 [30.00 fps - (0, 0)/0x0 crop]
           'SBGGR12_CSI2P' : 2028x1080 [30.00 fps - (0, 0)/0x0 crop]
                             2028x1520 [30.00 fps - (0, 0)/0x0 crop]
                             4056x3040 [30.00 fps - (0, 0)/0x0 crop]
# libcamera-vid --list-cameras
Available cameras
-----------------
0 : imx477 [4056x3040] (/base/soc/i2c0mux/i2c@1/imx477@1a)
    Modes: 'SRGGB10_CSI2P' : 1332x990 [120.05 fps - (696, 528)/2664x1980 crop]
           'SRGGB12_CSI2P' : 2028x1080 [50.03 fps - (0, 440)/4056x2160 crop]
                             2028x1520 [40.01 fps - (0, 0)/4056x3040 crop]
                             4056x3040 [10.00 fps - (0, 0)/4056x3040 crop]
```

For me, this works :
```
# diff camera-streamer-raspi-v3-12MP.service /etc/systemd/system/camera-streamer-raspi-HQ.service
8c8
< ConditionPathExists=/sys/bus/i2c/drivers/imx708/10-001a/video4linux
---
> ConditionPathExists=/sys/bus/i2c/drivers/imx477/10-001a/video4linux
12c12
<   -camera-path=/base/soc/i2c0mux/i2c@1/imx708@1a \
---
>   -camera-path=/base/soc/i2c0mux/i2c@1/imx477@1a \
15c15
<   -camera-width=2304 -camera-height=1296 \
---
>   -camera-width=1332 -camera-height=990 \
26,27c26,27
<   -camera-options=AfMode=2 \
<   -camera-options=AfRange=2 \
---
>   ; -camera-options=AfMode=2 \
>   ; -camera-options=AfRange=2 \
```

philippebourcier, 2023-02-24:
Hi there, I have ordered the 64MP beast (I'd love to use ePTZ)... but in the meantime the only cam' I had was a HQ Camera.

I was surprised you didn't have a service file for the HQ cam already, so here are the necessary infos needed to create one :
```
# libcamera-still --list-cameras
Available cameras
-----------------
0 : imx477 [4056x3040] (/base/soc/i2c0mux/i2c@1/imx477@1a)
    Modes: 'SBGGR10_CSI2P' : 1332x990 [30.00 fps - (0, 0)/0x0 crop]
           'SBGGR12_CSI2P' : 2028x1080 [30.00 fps - (0, 0)/0x0 crop]
                             2028x1520 [30.00 fps - (0, 0)/0x0 crop]
                             4056x3040 [30.00 fps - (0, 0)/0x0 crop]
# libcamera-vid --list-cameras
Available cameras
-----------------
0 : imx477 [4056x3040] (/base/soc/i2c0mux/i2c@1/imx477@1a)
    Modes: 'SRGGB10_CSI2P' : 1332x990 [120.05 fps - (696, 528)/2664x1980 crop]
           'SRGGB12_CSI2P' : 2028x1080 [50.03 fps - (0, 440)/4056x2160 crop]
                             2028x1520 [40.01 fps - (0, 0)/4056x3040 crop]
                             4056x3040 [10.00 fps - (0, 0)/4056x3040 crop]
```

For me, this works :
```
# diff camera-streamer-raspi-v3-12MP.service /etc/systemd/system/camera-streamer-raspi-HQ.service
8c8
< ConditionPathExists=/sys/bus/i2c/drivers/imx708/10-001a/video4linux
---
> ConditionPathExists=/sys/bus/i2c/drivers/imx477/10-001a/video4linux
12c12
<   -camera-path=/base/soc/i2c0mux/i2c@1/imx708@1a \
---
>   -camera-path=/base/soc/i2c0mux/i2c@1/imx477@1a \
15c15
<   -camera-width=2304 -camera-height=1296 \
---
>   -camera-width=1332 -camera-height=990 \
26,27c26,27
<   -camera-options=AfMode=2 \
<   -camera-options=AfRange=2 \
---
>   ; -camera-options=AfMode=2 \
>   ; -camera-options=AfRange=2 \
```

ayufan, 2023-02-24:
Interesting. I don't see the `SRGGB12_CSI2P` supported and this might be limiting factor here.

ETE-Design, 2023-02-25:
@philippebourcier Now that you have the HQ Camera, can you maby help with "High-performance mode via ISP for CSI" for it?

philippebourcier, 2023-02-25:
I tried and it doesn't work...
Sorry but since I have no plans on using this camera, I won't bother. 🤷‍♂️

ayufan, 2023-02-25:
Can you give output of tools dump cameras?

philippebourcier, 2023-02-25:
```
# ./dump_cameras.sh
+ v4l2-ctl --list-devices
bcm2835-codec-decode (platform:bcm2835-codec):
        /dev/video10
        /dev/video11
        /dev/video12
        /dev/video18
        /dev/video31
        /dev/media4

bcm2835-isp (platform:bcm2835-isp):
        /dev/video13
        /dev/video14
        /dev/video15
        /dev/video16
        /dev/video20
        /dev/video21
        /dev/video22
        /dev/video23
        /dev/media0
        /dev/media1

unicam (platform:fe801000.csi):
        /dev/video0
        /dev/video1
        /dev/media2

rpivid (platform:rpivid):
        /dev/video19
        /dev/media3

+ for device in /dev/video*
+ v4l2-ctl -d /dev/video0 -L
+ v4l2-ctl -d /dev/video0 --list-formats-out
ioctl: VIDIOC_ENUM_FMT
        Type: Video Output

+ v4l2-ctl -d /dev/video0 --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture

        [0]: 'YUYV' (YUYV 4:2:2)
                Size: Stepwise 16x16 - 16376x16376 with step 1/1
        [1]: 'UYVY' (UYVY 4:2:2)
                Size: Stepwise 16x16 - 16376x16376 with step 1/1
        [2]: 'YVYU' (YVYU 4:2:2)
                Size: Stepwise 16x16 - 16376x16376 with step 1/1
        [3]: 'VYUY' (VYUY 4:2:2)
                Size: Stepwise 16x16 - 16376x16376 with step 1/1
        [4]: 'RGBP' (16-bit RGB 5-6-5)
                Size: Stepwise 16x16 - 16376x16376 with step 1/1
        [5]: 'RGBR' (16-bit RGB 5-6-5 BE)
                Size: Stepwise 16x16 - 16376x16376 with step 1/1
        [6]: 'RGBO' (16-bit A/XRGB 1-5-5-5)
                Size: Stepwise 16x16 - 16376x16376 with step 1/1
        [7]: 'RGBQ' (16-bit A/XRGB 1-5-5-5 BE)
                Size: Stepwise 16x16 - 16376x16376 with step 1/1
        [8]: 'RGB3' (24-bit RGB 8-8-8)
                Size: Stepwise 16x16 - 16376x16376 with step 1/1
        [9]: 'BGR3' (24-bit BGR 8-8-8)
                Size: Stepwise 16x16 - 16376x16376 with step 1/1
        [10]: 'RGB4' (32-bit A/XRGB 8-8-8-8)
                Size: Stepwise 16x16 - 16376x16376 with step 1/1
        [11]: 'BA81' (8-bit Bayer BGBG/GRGR)
                Size: Stepwise 16x16 - 16376x16376 with step 1/1
        [12]: 'GBRG' (8-bit Bayer GBGB/RGRG)
                Size: Stepwise 16x16 - 16376x16376 with step 1/1
        [13]: 'GRBG' (8-bit Bayer GRGR/BGBG)
                Size: Stepwise 16x16 - 16376x16376 with step 1/1
[... 620 lines cut ...]
JPEG Compression Controls

            compression_quality 0x009d0903 (int)    : min=1 max=100 step=1 default=80 value=80
+ v4l2-ctl -d /dev/video31 --list-formats-out
ioctl: VIDIOC_ENUM_FMT
        Type: Video Output Multiplanar

        [0]: 'RGBP' (16-bit RGB 5-6-5)
        [1]: 'RGB3' (24-bit RGB 8-8-8)
        [2]: 'BGR3' (24-bit BGR 8-8-8)
        [3]: 'AB24' (32-bit RGBA 8-8-8-8)
        [4]: 'BGR4' (32-bit BGRA/X 8-8-8-8)
        [5]: 'YU12' (Planar YUV 4:2:0)
        [6]: 'YUYV' (YUYV 4:2:2)
        [7]: 'YVYU' (YVYU 4:2:2)
        [8]: 'VYUY' (VYUY 4:2:2)
        [9]: 'UYVY' (UYVY 4:2:2)
+ v4l2-ctl -d /dev/video31 --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture Multiplanar

        [0]: 'JPEG' (JFIF JPEG, compressed)
                Size: Stepwise 32x32 - 1920x1920 with step 2/2

```

ETE-Design, 2023-05-02:
@ayufan Can you use the Dump to add support for the RPI HQ Camera in "High-performance mode via ISP for CSI"?

philippebourcier, 2023-05-02:
As a more generic question :
**What are the steps to follow to get a camera working in ISP mode ?**

I have an imx708_wide, the imx477 and neither seem to work in ISP mode. 🤷‍♂️
Will it really change something in termes of performance on a Pi 4 ?
And on a Pi Zero 2 ?

DjofPlusPlus, 2023-08-06:
I have the imx477 working with the following service config:
```
;
; Official Raspberry Pi High Quality Camera based on the Sony IMX477
; https://www.raspberrypi.com/products/raspberry-pi-high-quality-camera/
;
[Unit]
Description=camera-streamer web camera for HQ Camera IMX477
After=network.target
ConditionPathExists=/sys/bus/i2c/drivers/imx477/10-001a/video4linux

[Service]
ExecStart=/usr/bin/camera-streamer \
  -camera-path=/base/soc/i2c0mux/i2c@1/imx477@1a \
  -camera-type=libcamera \
  -camera-format=YUYV \
  -camera-width=2028 -camera-height=1520 \
  -camera-fps=30 \
  ; use two memory buffers to optimise usage
  -camera-nbufs=2 \
  ; the snapshot at full resolution
  -camera-snapshot.height=1520 \
  ; the video/webrtc is 1280x720
  -camera-video.height=720 \
  ; the stream is 1280x720
  -camera-stream.height=720 \
  ; no autofocus
  --http-listen=0.0.0.0 \
  --http-port=8080 \
  -rtsp-port

DynamicUser=yes
SupplementaryGroups=video i2c
Restart=always
RestartSec=10
Nice=10
IOSchedulingClass=idle
IOSchedulingPriority=7
CPUWeight=20
AllowedCPUs=1-3
MemoryMax=250M

[Install]
WantedBy=multi-user.target

```

cearka, 2023-09-13:
> ;
> ; Official Raspberry Pi High Quality Camera based on the Sony IMX477
> ; https://www.raspberrypi.com/products/raspberry-pi-high-quality-camera/
> ;
> [Unit]
> Description=camera-streamer web camera for HQ [...]

Hi.  I created a service with the above and this doesn't seem to work for me.  
I've added 
dtoverlay=imx477 
to the /boot/config.txt  
libcamera-hello --list-cameras definately sees the piHQ  camera.
but after copying the above service into the service director and running a enable and start. I still can't access the control or anything.

cearka, 2023-09-14:
@DjofPlusPlus 

Would you mind sharing what, if anything else, you may have needed to do get the Pi HQ working?  I used your service file and it wasn't enough.  The web page won't even launch because I don't think camera is activating. 

For context, I did the following:
1. fresh  install of bullseye 32 bit lite from the raspberry pi imager.
2. followed @ayufan 's install instructions.  This in itself is enough to get it running with the pi cam module 3 for me.
3. I added an dtoverlay=imx477 in the /boot/config.txt
4. tested libcamera-still --list-cameras.  The pi HQ does show up with that command.
5. copied your service in to the service folder, enabled and started it.  

And it doesn't work.  Is there any other weird quirks required to get pi HQ to run?  like some weird update command? or additional lines in the /boot/config.txt or elsewhere?  Some strange option that needs to be toggled in raspi-config?  I've struggled to find the solution for a few hours now and have had no luck.  

would appreciate any insight, thanks.

DjofPlusPlus, 2023-09-15:
Hello @cearka 

Sorry I don't have much input to provide, I'm no linux expert... However if you want to compare
- I use Bullseye Aarch64, kernel `6.1.21-v8+`
- I installed from deb `camera-streamer-raspi_0.2.6.bullseye_arm64.deb`, for some reason compiling from source as in the past was causing a crash
- I don't have the imx477 overlay in my /boot/confix.txt, but `camera_auto_detect=1`
- Have you tested capturing with something like `libcamera-still -o test.jpg` ?
- I copied mine in my home folder, enabled and start/stop work as expected

For the service file specifically, I would say double check the paths are correct:
- `/sys/bus/i2c/drivers/imx477/10-001a/video4linux` should exist on the file system
`/base/soc/i2c0mux/i2c@1/imx477@1a` should match the libcamera output

I don't remember doing that much beyond the fresh install, looking in my notes I did see the following:
> Rpi OS: Use raspi-config to switch to NetworkManager
So I probably didn't change much else

cearka, 2023-09-15:
@DjofPlusPlus thank for the response!  I'll give this a try.

cearka, 2023-09-24:
@DjofPlusPlus it was the installing from the precompiled that fixed it ! thanks!

ETE-Design, 2024-01-02:
@ayufan Do you have enough info to make  support for the RPI HQ Camera in "High-performance mode via ISP for CSI"?

ETE-Design, 2024-02-17:
@DjofPlusPlus Have you got the HQ Camera work for "RPI HQ "High-performance mode via ISP for CSI"?
Or any chance I can ask you to look at it?

cearka, 2024-02-18:
I eventually got the HQ cam to work.  I had to to use the prebuilt install instead of compiling manually.  That was pretty much the only difference

ETE-Design, 2024-02-29:
@DjofPlusPlus Do you know how to make it work with: "High-performance mode via ISP for CSI"?

ETE-Design, 2024-02-29:
> As a more generic question : **What are the steps to follow to get a camera working in ISP mode ?**
> 
> I have an imx708_wide, the imx477 and neither seem to work in ISP mode. 🤷‍♂️ Will it really change something in termes of performance on a Pi 4 ? And on a Pi Zero 2 ?

@philippebourcier Did you ever find out what: ** What are the steps to follow to get a camera working in ISP mode ?**

ETE-Design, 2024-04-02:
@ayufan Can you pls. help us, at least tell:
**What are the steps to follow to get a camera working in ISP mode ?**

ETE-Design, 2024-05-12:
@ayufan Nice to see you are working on this project again - At least a little... Can you pls. "Release" again so we can update Raspberry Pi OS without need to compile it by our own?

And also take your time to tell us:
What are the steps to follow to get a camera working in ISP mode?
Or make the chances needed for the HQ Camera in ISP Mode?

ayufan, 2025-07-03:
> @ayufan <https://github.com/ayufan> Nice to see a new release from you :-)
> Can I ask you to take your time to tell us:
> Or make the chances needed for the HQ Camera in to work in ISP Mode? And
> then tell how to use / run it each time?

It is tricky. As libcamera does set ISP sensor. The only way that I managed
to make it work was to run libcamera, capture all v4l2 parameters, and then
use fixed ISP config to the sensor. This will only work in stable
conditions, as you loose auto exposure controls, etc.

I used this:
https://github.com/ayufan/camera-streamer/blob/main/tools/csi_camera.sh. It
should work, but it was long time ago.

ETE-Design, 2025-07-03:
@ayufan  Can you guide me in the direction on how to get these parameters?
I need it to get lowest possible latency at highest possible resolution - Cause would like to use it with OBS Studio to stream swim competition in a club where I'm a volunteer.
Later on I have to figure out how to make it work like a PTZ so I can control the directions direct from OBS Studio.

ayufan, 2025-07-03:
> @ayufan <https://github.com/ayufan> As I read it in your Readme it should
> be possible to get higher resolution and FPS when using "High-performance
> mode via ISP for CSI" so that's why ;-)

Libcamera one is the less of an issue. If this has acceptable latency, you
should use this, as it provides a better control and AE. Overall RasPI is
not best for this usage, since the limited h264/1080p/30fps and in general
outdated encoder producing average quality output.

ETE-Design, 2025-07-03:
@ayufan Would like to have the performance like you got on a Raspberry PI Zero 2W as described here:

"This mode allows to provide significantly better performance for camera sensors than described in specs. For example for Arducam 16MP (using IMX519) it is possible to achieve:

120fps on 1280x720, with latency of about 50ms, where-as documentation says 1280x720x60p

60fps on 1920x1080, with latency of about 90ms, where-as documentation says 1920x1080x30p

30fps on 2328x1748, with latency of about 140ms, where-as documentation says 2328x1748x15p

The above was tested with camera connected to Raspberry PI Zero 2W, streaming MJPEG over WiFi with camera recording"

I need as high FPS as possible and high resolution since it will be streamed "Live"
I can see you both got 120FPS and 60FPS at really low latency

ayufan, 2025-07-03:
> @ayufan <https://github.com/ayufan> Another thing - How to even tell your
> camera streamer to use the .sh file on each startup?

The .sh file is an example to run from command line. You have to figure out
how you want to start service yourself. This can be systemd, or calling sh
manually, or automatically.