ayufan/camera-streamer #28: ScalerCrop option

martiege, 2023-01-13:
Hello! 

I've been trying to use the ScalerCrop option to zoom in on the image, but I'm not seeing any changes to the stream, any reason why this would be? 

I'm trying e.g. `option?ScalerCrop=(0,0)/1000x1000`, and nothing changes. Anything specific I'm doing wrong with the format, or is there something else I'm not understanding regarding this option? 

Using other options work fine, and I'm using an arducam 64MP

ayufan, 2023-01-13:
Passing those options is currently not supported. It would have to be programmed first.

martiege, 2023-01-13:
That would explain it, haha... I haven't looked much through the repo, but some quick searches didn't show me where the options are parsed, can you point me in the right direction?

Sendt fra Outlook for Android<https://aka.ms/AAb9ysg>

closed 2023-02-22 by ayufan