ayufan/camera-streamer #32: Pi Camera Module 3 Autofocus

eprouty, 2023-01-19:
I just found this project while trying to setup an efficient stream using the new camera module 3 for OctoPrint.

I have gotten this running successfully with the following command: `sudo tools/libcamera_camera.sh -camera-format=YUYV -camera-path=/base/soc/i2c0mux/i2c@1/imx708@1a`

However, I am trying to get it to enable autofocus.  This works on the most recent version of `libcamera` like so: `libcamera-hello -t 0 --autofocus-mode continuous`

Is there a way to set the autofocus options that I am missing?

charredchar, 2023-01-19:
I have a module 3 in the mail as well, I am interested in this.

ayufan, 2023-01-19:
@eprouty @charredchar There's not a lot of magic in there. Just check what `Af*` settings are there, and set proper options for those. Likely for the above is `AfMode=<value>`.

charredchar, 2023-01-19:
> @eprouty @charredchar There's not a lot of magic in there. Just check what `Af*` settings are there, and set proper options for those. Likely for the above is `AfMode=<value>`.

Thanks! As soon as mine comes in I can take a look.

eprouty, 2023-01-19:
Messed around with that a bit seems like `AfMode=2` gets it working! Not sure what `AfMode=1` is doing, but 2 seems to do what's expected.

Thank you!

ayufan, 2023-01-19:
@eprouty https://github.com/raspberrypi/libcamera/blob/7c855784a28d0c9b742c6137bf311ac3779373df/src/libcamera/control_ids.yaml#L402 here are values for various options

eprouty, 2023-01-19:
> @eprouty https://github.com/raspberrypi/libcamera/blob/7c855784a28d0c9b742c6137bf311ac3779373df/src/libcamera/control_ids.yaml#L402 here are values for various options

This is the bit I was sorely missing! Thanks for pointing out the documentation for those options.

eprouty, 2023-01-19:
This project is an awesome replacement for mjpg-streamer, and I hope it'll see a lot more use as more people switch to camera module 3 which doesn't seem to be supported by that project.

Do you have a buymeacoffee page or anything? Would love to contribute a bit for the work you've done and help you just gave me.

ayufan, 2023-01-20:
I do have https://ko-fi.com/ayufan :)

closed 2023-01-20 by ayufan