ayufan/camera-streamer #35: Low latency with libcamera

kbingham, 2023-01-26:
I've been looking into this with the Raspberry Pi guys, to see what we can do to fix the latency issues you report so you don't need the manual configuration of the ISP.

It may be that we could add a control or such to 'request' low-latency configurations from a camera.

I can think of a couple of ways to measure the latency, but to be able to compare against your numbers can you confirm how you are measuring the latency ?

Are you just capturing an image with the camera of a stopwatch and the stream view in the background?

ayufan, 2023-01-26:
> Are you just capturing an image with the camera of a stopwatch and the stream view in the background?

Yes, this is what I was doing. Playing YouTube video, capturing this with camera, and having another tab open with streaming.

kbingham, 2023-01-26:
Aha a youtube stopwatch :smile: I hadn't thought of that. I'm not sure I'd trust it for always being an accurate time keeper - but simple to get initial measurements.

ayufan, 2023-01-26:
> I'm not sure I'd trust it for always being an accurate time keeper - but simple to get initial measurements.

I don't have anything better.

kbingham, 2023-01-26:
haha It's fine - I think we're looking for details and gains of 10's of milliseconds, and not too worried about the singles at the moment. :-)

clearRawPath22, 2023-10-28:
Sorry for bumping this old thread. I have two questions:

But, first: respect for this library. Really cool. 

1 - About the performance mentioned in the documentation
I am trying to reproduce the numbers mentioned in https://github.com/ayufan/camera-streamer/blob/main/docs/v4l2-isp-mode.md

I get at best 130ms latency for 120fps on 1280x720, instead of the mentioned 50ms (same camera model, RPi 4 1GB, with fresh last installation of Raspian. Viewing the MJPEG stream in a Firefox browser in an Linux laptop. Raspi and laptop are next to the router). I know that it depends on the router. And in my case, the tests were done at relatively dark environment, so the camera had probably larger exposure time. 

Now I am confused, since that documentation page and this thread have conflicting information:
- Here: "Playing YouTube video, capturing this with camera, and having another tab open with streaming." - I assume this has NO streaming over Wifi involved
- There: "... was tested with camera connected to Raspberry PI Zero 2W, streaming MJPEG over WiFi with camera recording https://www.youtube.com/watch?v=e8ZtPSIfWPc from desktop monitor."

How can I understand that?
If this involves streaming over Wifi to another Raspi/PC/Tablet/etc, is is possible to describe the viewing setup (which device, program and/or commands were used to view the stream)?

Thank you,