ayufan/camera-streamer #36: Using AfWindows

charredchar, 2023-02-01:
I am attempting to use the `AfWindows` parameter to adjust where autofocus is triggered from on a Pi Camera 3. I was attempting to test using the URL method first but I can not seem to get it to take properly. Example; `/option?AfWindows=(0,0,4608,250)`
Also, is there any method to view what the parameters are currently set to for verification?

ayufan, 2023-02-01:
Take a look at this: https://github.com/ayufan/camera-streamer/pull/30

charredchar, 2023-02-01:
Oh, I didn't see that because I only searched issues, thanks! I'll wait for it to get merged then update. Keep up the amazing work, I just got a Pi Camera 3 and it is working great with Cramera-Streamer!

closed 2023-02-01 by charredchar