ayufan/camera-streamer #37: IMX462 Low Light

Tormund1985, 2023-02-06:
How can I make it work with an IMX462 sensor, bullseye and kernel 5.15?

Tormund1985, 2023-02-07:
I modified service file in order to get it work with arducam IMX462, however y can't set any options being set up by device_options.c file. The output of status service:

```
root@raspberrypi:/# sudo service camera-streamer-arducam-16MP status
● camera-streamer-arducam-16MP.service - camera-streamer web camera
     Loaded: loaded (/home/pi/camera-streamer/service/camera-streamer-arducam-16MP.service; enabled; vendor preset: enabled)
     Active: active (running) since Tue 2023-02-07 10:36:24 CET; 6s ago
   Main PID: 1632 (camera-streamer)
      Tasks: 16 (limit: 255)
        CPU: 555ms
     CGroup: /system.slice/camera-streamer-arducam-16MP.service
             └─1632 /usr/local/bin/camera-streamer -camera-path=/base/soc/i2c0mux/i2c@1/imx290@1a -camera-type=libcamera -camera-for>

feb 07 10:36:24 raspberrypi camera-streamer[1632]: device/v4l2/device_options.c: H264: Configuring option h264minimumqpvalue (00990a>
feb 07 10:36:24 raspberrypi camera-streamer[1632]: device/v4l2/device_options.c: H264: Configuring option h264maximumqpvalue (00990a>
feb 07 10:36:24 raspberrypi camera-streamer[1632]: device/links.c: ?: Link 0: CAMERA:capture[1920x1080/YUYV/2] => [H264:output:mplan>
feb 07 10:36:24 raspberrypi camera-streamer[1632]: device/links.c: ?: Link 1: H264:capture:mplane[1920x1080/H264/2] => [H264-CAPTURE]
feb 07 10:36:24 raspberrypi camera-streamer[1632]: device/links.c: ?: Link 2: JPEG:capture:mplane[1920x1056/JPEG/2] => [JPEG-CAPTURE]
feb 07 10:36:24 raspberrypi camera-streamer[1632]: device/buffer_list.c: CAMERA:capture: Streaming started... Was 0 of 2 enqueud
feb 07 10:36:25 raspberrypi camera-streamer[1632]: device/buffer_list.c: H264:output:mplane: Streaming started... Was 0 of 2 enqueud
feb 07 10:36:25 raspberrypi camera-streamer[1632]: device/buffer_list.c: JPEG:output:mplane: Streaming started... Was 0 of 2 enqueud
feb 07 10:36:25 raspberrypi camera-streamer[1632]: device/buffer_list.c: H264:capture:mplane: Streaming started... Was 0 of 2 enqueud
feb 07 10:36:25 raspberrypi camera-streamer[1632]: device/buffer_list.c: JPEG:capture:mplane: Streaming started... Was 0 of 2 enqueud
```

Many thanks.

ayufan, 2023-02-07:
@Tormund1985 Would you mind contributing additional `service/` into project. This will help others later :)

Tormund1985, 2023-02-07:
> @Tormund1985 Would you mind contributing additional `service/` into project. This will help others later :)

Of course:
```

[Unit]
Description=camera-streamer web camera
After=network.target
ConditionPathExists=/sys/bus/i2c/drivers/imx290/10-001a/video4linux

[Service]
ExecStart=/usr/local/bin/camera-streamer \
  -camera-path=/base/soc/i2c0mux/i2c@1/imx290@1a \
  -camera-type=libcamera \
  -camera-format=YUYV \
  -camera-width=1920 -camera-height=1080 \
  -camera-options=rotation=180 \
  -camera-fps=10 \
  ; use two memory buffers to optimise usage
  -camera-nbufs=2 \
  ; the high-res is 1920x1080
  -camera-high_res_factor=1 \
  ; the low-res is 640x480
  ;-camera-low_res_factor=3.0 \
  ; bump brightness slightly
  ;-camera-options=brightness=0.1 \
  ; disable auto-focus
  ;-camera-auto_focus=1 \
  -rtsp-port
  ; bitrate H264 \
  ; -camera-h264.options=bitrate=1000000

DynamicUser=yes
SupplementaryGroups=video i2c
Restart=always
RestartSec=10
Nice=10
IOSchedulingClass=idle
IOSchedulingPriority=7
CPUWeight=50
AllowedCPUs=1-4
MemoryMax=250M

[Install]
WantedBy=multi-user.target
```

ayufan, 2023-02-07:
Hmm, why `imx290@1a`?

ayufan, 2023-02-07:
Can you open a PR to add this new service?

Tormund1985, 2023-02-07:
> Hmm, why `imx290@1a`?

Is the output of libcamera-hello, I think use the same driver than IMX462

```
root@raspberrypi:/# libcamera-hello
Made DRM preview window
[0:43:46.024318908] [1705]  INFO Camera camera_manager.cpp:299 libcamera v0.0.0+4185-8c2b3b43
[0:43:46.066964120] [1708]  INFO RPI raspberrypi.cpp:1425 Registered camera /base/soc/i2c0mux/i2c@1/imx290@1a to Unicam device /dev/media3 and ISP device /dev/media0
[0:43:46.068144753] [1705]  INFO Camera camera.cpp:1028 configuring streams: (0) 968x548-YUV420
[0:43:46.068722179] [1708]  INFO RPI raspberrypi.cpp:805 Sensor: /base/soc/i2c0mux/i2c@1/imx290@1a - Selected sensor format: 1280x720-SRGGB12_1X12 - Selected unicam format: 1280x720-pRCC
```

ayufan, 2023-02-07:
Hmm. Interesting. The `imx290` is dependent on `dtoverlay` that you enabled.

Tormund1985, 2023-02-07:
> Hmm. Interesting. The `imx290` is dependent on `dtoverlay` that you enabled in `/boot/config.txt`. What you have there?

```
[all]
enable_uart=1
gpu_mem=128
dtoverlay=imx462
```

ayufan, 2023-02-07:
Yup. It uses `imx290@1a`. Can you follow those services (by the name of camera) and open PR https://github.com/ayufan/camera-streamer/tree/master/service?

Tormund1985, 2023-02-07:
> 

What do you mean about follow those services? Tell me what do you need and i will do it.

ayufan, 2023-02-07:
@Tormund1985 OK. I will create. Please give me links to what exactly this camera is and how to buy it, so I could reference it by product name instead of sensor (IMX462).

Tormund1985, 2023-02-07:
> @Tormund1985 OK. I will create. Please give me links to what exactly this camera is and how to buy it, so I could reference it by product name instead of sensor (IMX462).

Is the IMX462 Arducam Ultra low light, I ordered in UCtronics.

I get it work but I need to set options because the image is continuously pixelated.

Do you know how I can do it? I see that the file that modifies these values is the device_options.c and not the values entered in the .service file.

ayufan, 2023-02-07:
`-camera-options` switch is enough, it will call methods from `device_options.c`

Tormund1985, 2023-02-07:
> `-camera-options` switch is enough, it will call methods from `device_options.c`

camera-options set up from .service file are not applying

ayufan, 2023-02-07:
They should, but without a full log it is hard to know if they are or not.

Tormund1985, 2023-02-07:
> 

Couldn't you edit the device_options.c file directly to set fps, resolution, bitrate etc?

I have opened the file but I don't understand the code very well so I don't know how to enter a certain bitrate or fps.

Thank you very much for your kind help.

ayufan, 2023-02-07:
> Couldn't you edit the device_options.c file directly to set fps, resolution, bitrate etc?

No. This is why there are options that you pass via service meant for.

Tormund1985, 2023-02-07:
I attached a journalctl log if it could help:

```
feb 07 11:23:28 raspberrypi camera-streamer[1669]: output/rtsp/rtsp.cc: ?: Running RTSP server on '8554'
feb 07 11:23:28 raspberrypi camera-streamer[1669]: [0:35:38.864150076] [1669]  INFO Camera camera_manager.cpp:299 libcamera v0.0.0+4185-8c2b3b43
feb 07 11:23:28 raspberrypi camera-streamer[1669]: [0:35:38.964156226] [1682]  INFO RPI raspberrypi.cpp:1425 Registered camera /base/soc/i2c0mux/i2c@1/imx290@1a to Unicam device /dev/media3 and ISP device /dev/media0
feb 07 11:23:28 raspberrypi camera-streamer[1669]: device/libcamera/device.cc: CAMERA: Device path=/base/soc/i2c0mux/i2c@1/imx290@1a opened
feb 07 11:23:28 raspberrypi camera-streamer[1669]: [0:35:38.966107151] [1669]  INFO Camera camera.cpp:1028 configuring streams: (0) 1920x1080-YUYV
feb 07 11:23:28 raspberrypi camera-streamer[1669]: [0:35:38.967276706] [1682]  INFO RPI raspberrypi.cpp:805 Sensor: /base/soc/i2c0mux/i2c@1/imx290@1a - Selected sensor format: 1920x1080-SRGGB12_1X12 - Selected unicam format: 1920x1080-pRCC
feb 07 11:23:28 raspberrypi camera-streamer[1669]: device/buffer_list.c: CAMERA:capture: Using: 1920x1080/YUYV, bytesperline=3840
feb 07 11:23:28 raspberrypi camera-streamer[1669]: device/v4l2/device.c: H264: Device path=/dev/video11 fd=33 opened
feb 07 11:23:28 raspberrypi camera-streamer[1669]: device/buffer_list.c: H264:output:mplane: Using: 1920x1080/YUYV, bytesperline=3840
feb 07 11:23:28 raspberrypi camera-streamer[1669]: device/buffer_list.c: H264:capture:mplane: Using: 1920x1080/H264, bytesperline=0
feb 07 11:23:28 raspberrypi camera-streamer[1669]: device/v4l2/device.c: JPEG: Device path=/dev/video31 fd=36 opened
feb 07 11:23:28 raspberrypi camera-streamer[1669]: device/buffer_list.c: JPEG:output:mplane: Using: 1920x1056/YUYV, bytesperline=3840
feb 07 11:23:28 raspberrypi camera-streamer[1669]: device/buffer_list.c: JPEG:capture:mplane: Using: 1920x1056/JPEG, bytesperline=0
feb 07 11:23:28 raspberrypi camera-streamer[1669]: device/v4l2/device_options.c: JPEG: Configuring option compressionquality (009d0903) = 80
feb 07 11:23:28 raspberrypi camera-streamer[1669]: device/v4l2/device_options.c: H264: Configuring option repeatsequenceheader (009909e2) = 1
feb 07 11:23:28 raspberrypi camera-streamer[1669]: device/v4l2/device_options.c: H264: Configuring option videobitratemode (009909ce) = 0
feb 07 11:23:28 raspberrypi camera-streamer[1669]: device/v4l2/device_options.c: H264: Configuring option videobitrate (009909cf) = 2000000
feb 07 11:23:28 raspberrypi camera-streamer[1669]: device/v4l2/device_options.c: H264: Configuring option repeatsequenceheader (009909e2) = 5000000
feb 07 11:23:28 raspberrypi camera-streamer[1669]: device/v4l2/device_options.c: H264: Configuring option h264iframeperiod (00990a66) = 30
feb 07 11:23:28 raspberrypi camera-streamer[1669]: device/v4l2/device_options.c: H264: Configuring option h264level (00990a67) = 11
feb 07 11:23:28 raspberrypi camera-streamer[1669]: device/v4l2/device_options.c: H264: Configuring option h264profile (00990a6b) = 4
feb 07 11:23:28 raspberrypi camera-streamer[1669]: device/v4l2/device_options.c: H264: Configuring option h264minimumqpvalue (00990a61) = 16
feb 07 11:23:28 raspberrypi camera-streamer[1669]: device/v4l2/device_options.c: H264: Configuring option h264maximumqpvalue (00990a62) = 32
feb 07 11:23:28 raspberrypi camera-streamer[1669]: device/links.c: ?: Link 0: CAMERA:capture[1920x1080/YUYV/2] => [H264:output:mplane[1920x1080/YUYV/2], JPEG:output:mplane[1920x1056/YUYV/2]]
feb 07 11:23:28 raspberrypi camera-streamer[1669]: device/links.c: ?: Link 1: H264:capture:mplane[1920x1080/H264/2] => [H264-CAPTURE]
feb 07 11:23:28 raspberrypi camera-streamer[1669]: device/links.c: ?: Link 2: JPEG:capture:mplane[1920x1056/JPEG/2] => [JPEG-CAPTURE]
feb 07 11:23:28 raspberrypi camera-streamer[1669]: device/buffer_list.c: CAMERA:capture: Streaming started... Was 0 of 2 enqueud
feb 07 11:23:28 raspberrypi camera-streamer[1669]: device/buffer_list.c: H264:output:mplane: Streaming started... Was 0 of 2 enqueud
feb 07 11:23:28 raspberrypi camera-streamer[1669]: device/buffer_list.c: JPEG:output:mplane: Streaming started... Was 0 of 2 enqueud
feb 07 11:23:28 raspberrypi camera-streamer[1669]: device/buffer_list.c: H264:capture:mplane: Streaming started... Was 0 of 2 enqueud
feb 07 11:23:28 raspberrypi camera-streamer[1669]: device/buffer_list.c: JPEG:capture:mplane: Streaming started... Was 0 of 2 enqueud
feb 07 11:23:37 raspberrypi camera-streamer[1669]: output/rtsp/rtsp.cc: ?: Requesting stream.h264 stream...
feb 07 11:23:37 raspberrypi camera-streamer[1669]: output/rtsp/rtsp.cc: ?: Requesting stream.h264 stream...
feb 07 11:37:19 raspberrypi camera-streamer[1669]: output/rtsp/rtsp.cc: ?: Requesting stream.h264 stream...
feb 07 11:37:19 raspberrypi camera-streamer[1669]: output/rtsp/rtsp.cc: ?: Requesting stream.h264 stream...
feb 07 11:37:40 raspberrypi camera-streamer[1669]: output/rtsp/rtsp.cc: ?: Requesting stream.h264 stream...
feb 07 11:37:40 raspberrypi camera-streamer[1669]: output/rtsp/rtsp.cc: ?: Requesting stream.h264 stream...
feb 07 11:38:01 raspberrypi camera-streamer[1669]: output/rtsp/rtsp.cc: ?: Requesting stream.h264 stream...
feb 07 11:38:01 raspberrypi camera-streamer[1669]: output/rtsp/rtsp.cc: ?: Requesting stream.h264 stream...
feb 07 11:38:22 raspberrypi camera-streamer[1669]: output/rtsp/rtsp.cc: ?: Requesting stream.h264 stream...
[... cut ...]
feb 07 11:43:01 raspberrypi camera-streamer[1811]: device/buffer_list.c: JPEG:output:mplane: Streaming started... Was 0 of 2 enqueud
feb 07 11:43:01 raspberrypi camera-streamer[1811]: device/buffer_list.c: H264:capture:mplane: Streaming started... Was 0 of 2 enqueud
feb 07 11:43:01 raspberrypi camera-streamer[1811]: device/buffer_list.c: JPEG:capture:mplane: Streaming started... Was 0 of 2 enqueud
feb 07 11:43:10 raspberrypi camera-streamer[1811]: output/rtsp/rtsp.cc: ?: Requesting stream.h264 stream...
feb 07 11:43:10 raspberrypi camera-streamer[1811]: output/rtsp/rtsp.cc: ?: Requesting stream.h264 stream...
```

ayufan, 2023-02-07:
I don't see any problems in here. It seems that you need to pass camera options that you need.

Tormund1985, 2023-02-07:
> I don't see any problems in here. It seems that you need to pass camera options that you need.

I don't know why, but it doesn't work any option that y set in .service file. Maybe the sentences should be different...

Tormund1985, 2023-02-07:
> What you set then?

For example:

  -camera-format=H264 \
  -camera-options=rotation=180 \
  -camera-fps=10 \

They don't do anything.

ayufan, 2023-02-07:
Show the full service with everything. It is important to add `\` at end of each line.

Tormund1985, 2023-02-07:
> Show the full service with everything. It is important to add ` \` at end of each line.

is the same than before:
```

  GNU nano 5.4                           /etc/systemd/system/camera-streamer-arducam-16MP.service                                    
[Unit]
Description=camera-streamer web camera
After=network.target
ConditionPathExists=/sys/bus/i2c/drivers/imx290/10-001a/video4linux

[Service]
ExecStart=/usr/local/bin/camera-streamer \
  -camera-path=/base/soc/i2c0mux/i2c@1/imx290@1a \
  -camera-type=libcamera \
  -camera-format=H264 \
  -camera-width=1920 -camera-height=1080 \
  -camera-options=rotation=180 \
  -camera-fps=10 \
  ; use two memory buffers to optimise usage
  -camera-nbufs=2 \
  ; the high-res is 1920x1080
  -camera-high_res_factor=1 \
  ; the low-res is 640x480
  ;-camera-low_res_factor=3.0 \
  ; bump brightness slightly
  ;-camera-options=brightness=0.1 \
  ; disable auto-focus
  ;-camera-auto_focus=1 \
  -rtsp-port \

DynamicUser=yes
SupplementaryGroups=video i2c
Restart=always
RestartSec=10
Nice=10
IOSchedulingClass=idle
IOSchedulingPriority=7
CPUWeight=50
AllowedCPUs=1-4
MemoryMax=250M

[Install]
WantedBy=multi-user.target

```

ayufan, 2023-02-07:
Well, but where are all these additional options?

`-camera-format=H264` this does not seem right, it should rather be YUYV.

Tormund1985, 2023-02-08:
> Well, but where are all these additional options?
> 
> `-camera-format=H264` this does not seem right, it should rather be YUYV.

This camera support H264:
[(https://www.uctronics.com/download/Amazon/B0333_Manual.pdf)]

But I just put it in to see if it would work better that way. The image is very pixelated, and the RTSP stream fails after a few minutes.

The rest of the sentences don't apply:
  -camera-options=rotation=180 \
  -camera-fps=10 \

ayufan, 2023-02-08:
H264 is inflexible to do tranformations and other stuff. This is why you
should be using the closest native format to the camera.

Tormund1985, 2023-02-09:
> H264 is inflexible to do tranformations and other stuff. This is why you should be using the closest native format to the camera.
> […](#)

I tried to change in case I was doing better. With the default settings the image is very bad, with continuous pixelation and delays.... maybe the service is not optimised for my camera model... I've been trying to apply changes and still can't. With other sensors I apply the changes with v4l2-ctl but with this sensor I don't know how to do it...

In addition, the rtsp transmission drops after a few minutes, which is precisely what I am interested in so that I can introduce it into a video manager for motion detection.

Tormund1985, 2023-02-28:
The same thing keeps happening to me, after a while the rtsp video stream drops showing the following message:

`pi@raspberrypi:~ $ sudo service camera-streamer-arducam-16MP status
Warning: The unit file, source configuration file or drop-ins of camera-streamer-arducam-16MP.service changed on disk. Run 'systemctl >
● camera-streamer-arducam-16MP.service - camera-streamer web camera
     Loaded: loaded (/home/pi/camera-streamer/service/camera-streamer-arducam-16MP.service; enabled; vendor preset: enabled)
     Active: active (running) since Mon 2023-02-27 13:54:11 CET; 18h ago
    Process: 30801 ExecStartPre=/bin/sleep 60 (code=exited, status=0/SUCCESS)
   Main PID: 30803 (camera-streamer)
      Tasks: 16 (limit: 255)
        CPU: 13min 48.803s
     CGroup: /system.slice/camera-streamer-arducam-16MP.service
             └─30803 /usr/local/bin/camera-streamer -camera-path=/base/soc/i2c0mux/i2c@1/imx290@1a -camera-type=libcamera -camera-form>

feb 28 08:12:55 raspberrypi camera-streamer[30803]: device/v4l2/buffer.c: H264:capture:mplane:buf0: ioctl(ret=-1): Can't queue buffer.
feb 28 08:12:55 raspberrypi camera-streamer[30803]: device/v4l2/buffer.c: H264:capture:mplane:buf0: ioctl(ret=-1): Can't queue buffer.
feb 28 08:12:55 raspberrypi camera-streamer[30803]: device/v4l2/buffer.c: H264:capture:mplane:buf0: ioctl(ret=-1): Can't queue buffer.
feb 28 08:12:56 raspberrypi camera-streamer[30803]: device/v4l2/buffer.c: H264:capture:mplane:buf0: ioctl(ret=-1): Can't queue buffer.
feb 28 08:12:56 raspberrypi camera-streamer[30803]: device/v4l2/buffer.c: H264:capture:mplane:buf0: ioctl(ret=-1): Can't queue buffer.
feb 28 08:12:56 raspberrypi camera-streamer[30803]: device/v4l2/buffer.c: H264:capture:mplane:buf0: ioctl(ret=-1): Can't queue buffer.
feb 28 08:12:56 raspberrypi camera-streamer[30803]: device/v4l2/buffer.c: H264:capture:mplane:buf0: ioctl(ret=-1): Can't queue buffer.
feb 28 08:12:56 raspberrypi camera-streamer[30803]: device/v4l2/buffer.c: H264:capture:mplane:buf0: ioctl(ret=-1): Can't queue buffer.
feb 28 08:12:56 raspberrypi camera-streamer[30803]: device/v4l2/buffer.c: H264:capture:mplane:buf0: ioctl(ret=-1): Can't queue buffer.
feb 28 08:12:56 raspberrypi camera-streamer[30803]: device/v4l2/buffer.c: H264:capture:mplane:buf0: ioctl(ret=-1): Can't queue buffer.
feb 28 08:12:56 raspberrypi camera-streamer[30803]: device/v4l2/buffer.c: H264:capture:mplane:buf0: ioctl(ret=-1): Can't queue buffer.
feb 28 08:12:56 raspberrypi camera-streamer[30803]: device/v4l2/buffer.c: H264:capture:mplane:buf0: ioctl(ret=-1): Can't queue buffer.
feb 28 08:12:56 raspberrypi camera-streamer[30803]: device/v4l2/buffer.c: H264:capture:mplane:buf0: ioctl(ret=-1): Can't queue buffer.
feb 28 08:12:57 raspberrypi camera-streamer[30803]: device/v4l2/buffer.c: H264:capture:mplane:buf0: ioctl(ret=-1): Can't queue buffer.
lines 1-25/25 (END)
`

I can't edit anything with camera-options either, neither bitrate, nor h264 level, rotation, brightness...

Can you give me a hand? It is very important for me to be able to use the IMX462 sensor for its virtues in low light.

ayufan, 2023-02-28:
@Tormund1985 There was a critical bug (now fixed on master) that caused this weird behavior. Update `master`, or even better switch to `develop` which also improves significantly performance.

Tormund1985, 2023-02-28:
> @Tormund1985 There was a critical bug (now fixed on master) that caused this weird behavior. Update `master`, or even better switch to `develop` which also improves significantly performance.

I am installing but I get next error when type make:

`cmd/camera-streamer/status.cc:19:10: fatal error: nlohmann/json.hpp: No existe el fichero o el directorio`

ayufan, 2023-02-28:
`git submodule update --init --recursive`

Tormund1985, 2023-02-28:
> `git submodule update --init --recursive`

perfect, so far it's working, thank you very much!

What still doesn't work are the bitrate settings, rotation...

Tormund1985, 2023-03-01:
The image keeps going away in my manager and it is impossible to make a reliable motion detection. Through journalctl I see a strange behaviour, restarting the service every so often:

```

**mar 01 06:18:50 raspberrypi systemd[1]: camera-streamer-arducam-16MP.service: Main process exited, code=killed, status=11/SEGV**
mar 01 06:18:50 raspberrypi systemd[1]: camera-streamer-arducam-16MP.service: Failed with result 'signal'.
mar 01 06:18:50 raspberrypi systemd[1]: camera-streamer-arducam-16MP.service: Consumed 19.299s CPU time.
mar 01 06:19:00 raspberrypi systemd[1]: camera-streamer-arducam-16MP.service: Scheduled restart job, restart counter is at 46.
mar 01 06:19:00 raspberrypi systemd[1]: Stopped camera-streamer web camera.
mar 01 06:19:00 raspberrypi systemd[1]: camera-streamer-arducam-16MP.service: Consumed 19.299s CPU time.
mar 01 06:19:00 raspberrypi systemd[1]: Started camera-streamer web camera.
mar 01 06:19:00 raspberrypi camera-streamer[30658]: output/rtsp/rtsp.cc: ?: Running RTSP server on '8554'
mar 01 06:19:00 raspberrypi camera-streamer[30658]: [17:31:07.539803700] [30658]  INFO Camera camera_manager.cpp:299 libcamera v0.0.0+4185-8c2b3b43
mar 01 06:19:00 raspberrypi camera-streamer[30658]: [17:31:07.592363873] [30670]  INFO RPI raspberrypi.cpp:1425 Registered camera /base/soc/i2c0mux/i2c@1/imx290@1a to Unicam device /dev/media3 and ISP device /dev/media1
mar 01 06:19:00 raspberrypi camera-streamer[30658]: device/libcamera/device.cc: CAMERA: Device path=/base/soc/i2c0mux/i2c@1/imx290@1a opened
mar 01 06:19:00 raspberrypi camera-streamer[30658]: [17:31:07.593839082] [30658]  INFO Camera camera.cpp:1028 configuring streams: (0) 1920x1080-SRGGB12_CSI2P
mar 01 06:19:00 raspberrypi camera-streamer[30658]: [17:31:07.594668457] [30670]  INFO RPI raspberrypi.cpp:805 Sensor: /base/soc/i2c0mux/i2c@1/imx290@1a - Selected sensor format: 1920x1080-SRGGB12_1X12 - Selected unicam format: 1920x1080-pRCC
mar 01 06:19:00 raspberrypi camera-streamer[30658]: device/buffer_list.c: CAMERA:capture: Using: 1920x1080/RG12, bytesperline=2880, sizeimage=0.0MiB
mar 01 06:19:00 raspberrypi camera-streamer[30658]: device/buffer_list.c: CAMERA:capture: Opened 2 buffers. Memory used: 5.9 MiB
mar 01 06:19:00 raspberrypi camera-streamer[30658]: [17:31:07.612104296] [30658]  INFO Camera camera.cpp:1028 configuring streams: (0) 1920x1080-SRGGB12_CSI2P (1) 1920x1088-YUYV
mar 01 06:19:00 raspberrypi camera-streamer[30658]: [17:31:07.612881796] [30670]  INFO RPI raspberrypi.cpp:805 Sensor: /base/soc/i2c0mux/i2c@1/imx290@1a - Selected sensor format: 1920x1080-SRGGB12_1X12 - Selected unicam format: 1920x1080-pRCC
mar 01 06:19:00 raspberrypi camera-streamer[30658]: device/buffer_list.c: CAMERA:capture:1: Using: 1920x1088/YUYV, bytesperline=3840, sizeimage=0.0MiB
mar 01 06:19:00 raspberrypi camera-streamer[30658]: device/buffer_list.c: CAMERA:capture:1: Opened 2 buffers. Memory used: 8.0 MiB
mar 01 06:19:00 raspberrypi camera-streamer[30658]: device/v4l2/device.c: SNAPSHOT: Device path=/dev/video31 fd=37 opened
mar 01 06:19:00 raspberrypi camera-streamer[30658]: device/buffer_list.c: SNAPSHOT:output:mplane: Using: 1920x1088/YUYV, bytesperline=3840, sizeimage=4.0MiB
mar 01 06:19:00 raspberrypi camera-streamer[30658]: device/buffer_list.c: SNAPSHOT:output:mplane: Opened 2 buffers. Memory used: 0.0 MiB
mar 01 06:19:00 raspberrypi camera-streamer[30658]: device/buffer_list.c: SNAPSHOT:capture:mplane: Using: 1920x1088/JPEG, bytesperline=0, sizeimage=4.0MiB
mar 01 06:19:00 raspberrypi camera-streamer[30658]: device/buffer_list.c: SNAPSHOT:capture:mplane: Opened 2 buffers. Memory used: 8.0 MiB
mar 01 06:19:00 raspberrypi camera-streamer[30658]: device/v4l2/device.c: RESCALLER:STREAM: Device path=/dev/video12 fd=40 opened
mar 01 06:19:00 raspberrypi camera-streamer[30658]: device/buffer_list.c: RESCALLER:STREAM:output:mplane: Using: 1920x1088/YUYV, bytesperline=3840, sizeimage=4.0MiB
mar 01 06:19:00 raspberrypi camera-streamer[30658]: device/buffer_list.c: RESCALLER:STREAM:output:mplane: Opened 2 buffers. Memory used: 0.0 MiB
mar 01 06:19:00 raspberrypi camera-streamer[30658]: device/buffer_list.c: RESCALLER:STREAM:capture:mplane: Using: 864x480/YUYV, bytesperline=1728, sizeimage=0.8MiB
mar 01 06:19:00 raspberrypi camera-streamer[30658]: device/buffer_list.c: RESCALLER:STREAM:capture:mplane: Opened 2 buffers. Memory used: 1.6 MiB
mar 01 06:19:00 raspberrypi camera-streamer[30658]: device/v4l2/device.c: STREAM: Device path=/dev/video31 fd=43 opened
mar 01 06:19:00 raspberrypi camera-streamer[30658]: device/buffer_list.c: STREAM:output:mplane: Using: 864x480/YUYV, bytesperline=1728, sizeimage=0.8MiB
mar 01 06:19:00 raspberrypi camera-streamer[30658]: device/buffer_list.c: STREAM:output:mplane: Opened 2 buffers. Memory used: 0.0 MiB
mar 01 06:19:01 raspberrypi camera-streamer[30658]: device/buffer_list.c: STREAM:capture:mplane: Using: 864x480/JPEG, bytesperline=0, sizeimage=4.0MiB
mar 01 06:19:01 raspberrypi camera-streamer[30658]: device/buffer_list.c: STREAM:capture:mplane: Opened 2 buffers. Memory used: 8.0 MiB
mar 01 06:19:01 raspberrypi camera-streamer[30658]: device/v4l2/device.c: RESCALLER:VIDEO: Device path=/dev/video12 fd=46 opened
mar 01 06:19:01 raspberrypi camera-streamer[30658]: device/buffer_list.c: RESCALLER:VIDEO:output:mplane: Using: 1920x1088/YUYV, bytesperline=3840, sizeimage=4.0MiB
[... cut ...]
mar 01 06:41:06 raspberrypi systemd[1]: camera-streamer-arducam-16MP.service: Consumed 1min 59.414s CPU time.
mar 01 06:41:16 raspberrypi systemd[1]: camera-streamer-arducam-16MP.service: Scheduled restart job, restart counter is at 47.
mar 01 06:41:16 raspberrypi systemd[1]: Stopped camera-streamer web camera.
mar 01 06:41:16 raspberrypi systemd[1]: camera-streamer-arducam-16MP.service: Consumed 1min 59.414s CPU time.
mar 01 06:41:16 raspberrypi systemd[1]: Started camera-streamer web camera.**

```

ayufan, 2023-03-01:
@Tormund1985 It seems that process crashed with `Main process exited, code=killed, status=11/SEGV`. Which is access violation. Can you get a core dump of the process when it happens, and also send me a compiled `camera-streamer` alongside the `core`?

Tormund1985, 2023-03-01:
> @Tormund1985 It seems that process crashed with `Main process exited, code=killed, status=11/SEGV`. Which is access violation. Can you get a core dump of the process when it happens, and also send me a compiled `camera-streamer` alongside the `core`?

I am not an expert in the matter, if you tell me how to do it, I will pass you what you want. Thank you very much for your interest.

ayufan, 2023-03-01:
You expect something like this:

```
$ sysctl grep kernel.core_pattern
kernel.core_pattern = core
```

Then hopefully the `core` file should be alongside the `camera-streamer` binary depending from where it was run or in `/` folder. I need those two files send to me.

Tormund1985, 2023-03-01:
> 

Output is the same:

```
pi@raspberrypi:~ $ sysctl kernel.core_pattern
kernel.core_pattern = core
```

I have camera-streamer folder in /home/pi but i dont see any file named core... sorry for my ignorance.

ayufan, 2023-03-01:
Look for file named core anywhere for now.

Tormund1985, 2023-03-01:
> Look for file named core anywhere for now.

There is no file called core
```

pi@raspberrypi:/ $ sudo find / -type f -name "core"
/sys/kernel/debug/vchiq/log/core
/usr/share/dwagent/LICENSES/core
```

kbingham, 2023-03-01:
You might also need to make sure core dumps are enabled in that process space:

ulimit -c unlimited

Tormund1985, 2023-03-08:
I've been trying but I have no way to get that core file. I will try to find time to see what I am doing wrong.... thanks for your interest...

ayufan, 2023-03-08:
@Tormund1985 

I was able to reproduce it. The RTSP is fixed on latest develop. It should no longer crash. Will close this for now, please re-open if you still find it broken.

closed 2023-03-08 by ayufan