ayufan/camera-streamer #39: systemd unit configuration file for Pi Camera Module V3

giddyhup, 2023-02-11:
Not a bug, just a hint for people trying to use camera-streamer with the Pi Camera Module V3. The following works for my as a systemd unit configuration file. I use it with the file/service name `camera-streamer-pi3.service`.

```;
; Official Raspberry Pi v3 12MP camera based on the Sony IMX708 chip
; with autofocus
; https://www.raspberrypi.com/products/camera-module-3/
;
[Unit]
Description=camera-streamer web camera
After=network.target
ConditionPathExists=/sys/bus/i2c/drivers/imx708/10-001a/video4linux

[Service]
ExecStart=/usr/local/bin/camera-streamer \
  -camera-path=/base/soc/i2c0mux/i2c@1/imx708@1a \
  -camera-type=libcamera \
  -camera-format=YUYV \
  -camera-fps=25 \
  ; use two memory buffers to optimise usage
  -camera-nbufs=2 \
  ; the high-res is 1640x1232
  -camera-high_res_factor=2 \
  ; the low-res is 820x616
  -camera-low_res_factor=4 \
 -camera-options="AfMode=2" \
 -rtsp-port

DynamicUser=yes
SupplementaryGroups=video i2c
Restart=always
RestartSec=10
Nice=10
IOSchedulingClass=idle
IOSchedulingPriority=7
CPUWeight=20
AllowedCPUs=1-2
MemoryMax=250M

[Install]
WantedBy=multi-user.target
```

@ayufan you could add this to your examples.

ayufan, 2023-02-11:
@giddyhup How is so far your experience with this project and this camera?

giddyhup, 2023-02-11:
Your project is well documented, easy to use. Thank you for that. I wonder if it could become a drop-in replacement which could be used in RPI Web Cam Interface (I assume it is possible but there probably is some HowTo needed).  

The V3 Camera module works nicely and picture quality is very good. With autofocus you notice how it is chasing back and forth a bit before locking in. I haven't had it running for long time yet, so there are certainly people with more experience.

ltlowe, 2023-02-11:
There's a discussion ongoing on the octopi forums about using camera-streamer with the PI Camera Module V3.
https://community.octoprint.org/t/pi-camera-v3-imx-chipset-based-cameras-not-working/49022

I'm hosting the install script and .service file we arrived at in that thread.
https://github.com/ltlowe/misc/tree/main/picam3

This is the current service file we arrived, but it is likely "good enough" rather than "fully correct." :-)
I changed it from using `ConditionPathExists` to `ExecStartPre` because we found there was a startup race condition where systemd would sometimes try to start this service before the driver was ready. `ConditionPathExists` would just abort where `ExecStartPre` using test will trigger the systemd retries.
```
;
; Official Raspberry Pi v3 12MP camera based on the Sony IMX708 chip
; https://www.raspberrypi.com/products/camera-module-3/
;
[Unit]
Description=camera-streamer web camera
After=network.target

[Service]
;Fail if path doesn't exist.  Restarts will be attempted at 10s intervals.
ExecStartPre=/usr/bin/test -e /sys/bus/i2c/drivers/imx708/10-001a/video4linux

; raw still max is 4608x2592
; regular still is 2304x1296
; Max video resolution is 1920x1080p30 (720p60)

ExecStart=/usr/local/bin/camera-streamer \
  -camera-path=/base/soc/i2c0mux/i2c@1/imx708@1a \
  -camera-type=libcamera \
  ; YUYV offers best quality in camera-streamer
  -camera-format=YUYV \
  -camera-width=1920 -camera-height=1080 \
  ; 4608 (H) × 2592 is full res - don't know how to enable it 
  ; frame rate 30 for North America.  25 may work better in 50Hz countries
  -camera-fps=30 \
  ; use two memory buffers to optimise usage
  -camera-nbufs=2 \
  ; high-res video is 1920x1080
  ; high-res snapshot is 1920x1056
  -camera-high_res_factor=1 \
  ; the low-res is 960x512
  -camera-low_res_factor=2 \
  ; Enable Continuous Autofocus
  -camera-options=AfMode=2 \
  -camera-options=AfRange=2 \

DynamicUser=yes
SupplementaryGroups=video i2c
Restart=always
RestartSec=10
Nice=10
IOSchedulingClass=idle
IOSchedulingPriority=7
CPUWeight=20
AllowedCPUs=1-2
MemoryMax=250M

[Install]
WantedBy=multi-user.target
```

giddyhup, 2023-02-12:
@ltlowe thank you for sharing your unit file. Congrats on the work done for Octoprint!

No, I have not done much with the camera yet. I noticed some flickering, even in natural light and with the proper frame rate/frequency setting.

Gonioul, 2023-02-12:
> @giddyhup How is so far your experience with this project and this camera?

Nothing to do with this thread, but at least basic auth would be nice :-p

ayufan, 2023-02-22:
@ltlowe The `master` branch was updated with removal of `?res=low` and introduction of https://github.com/ayufan/camera-streamer#resolution.

My proposed settings are:
- snapshot = 1080p (jpeg)
- video = 720p (h264)
- stream = 480p (mjpeg)

Those are defaults for the included `v3 camera service`.

The `v3 camera service` uses half resolution to reduce BW requirements since it is further downscaled anyway.

closed 2023-02-22 by ayufan

ltlowe, 2023-02-22:
#response_container_BBPPID{font-family: initial; font-size:initial; color: initial;}  Cool. I'll adjust my octopi script to try your updates and let you know how it goes.  It may be a few days though.                                                                                                                                                                                                                                                                                                                         From: ***@***.***: February 22, 2023 11:40 a.m.To: ***@***.***: ***@***.***: ***@***.***; ***@***.***: Re: [ayufan/camera-streamer] systemd unit configuration file for Pi Camera Module V3 (Issue #39)  
@ltlowe The master branch was updated with removal of ?res=low and introduction of https://github.com/ayufan/camera-streamer#resolution.
My proposed settings are:
snapshot = 1080p (jpeg)video = 720p (h264)stream = 480p (mjpeg)
Those are defaults for the included v3 camera service.
The v3 camera service uses half resolution to reduce BW requirements since it is further downscaled anyway.

ltlowe, 2023-02-28:
@ayufan I'm not planning to update my script anymore since @foosel is working on a direct integration with OctoPi, but I noticed that your service file is still using 
```
ConditionPathExists=/sys/bus/i2c/drivers/imx708/10-001a/video4linux
```
In my previous testing there was a race condition where this was not always ready before systemd tried to launch camera-streamer.   Since `ConditionPathExists` just skips the service rather than retrying, the streamer would occasionally not start.

I didn't investigate further and simply switched to using ExecPreStart instead since it lets the retry mechanism kick in.
```
ExecStartPre=/usr/bin/test -e /sys/bus/i2c/drivers/imx708/10-001a/video4linux
```
Is this something you should be doing as well, or is your method considered reliable and whatever I had encountered was an anomaly?

foosel, 2023-02-28:
> In my previous testing there was a race condition where video4linux was not always ready before systemd tried to launch camera-streamer. Since `ConditionPathExists` just skips the service rather than retrying, the streamer would occasionally not start.

For the record, I'm pretty sure I encountered something similar, which is why I also went with `ExecStartPre`.

kbingham, 2023-02-28:
I've always wondered why the camera name is hardcoded in this project. one of the goals of libcamera is to abstract this so that it shouldn't matter which camera is connected, and all that you need could be determined at runtime.