ayufan/camera-streamer #45: No h264 Bitrate adjustment

andybrinkley, 2023-03-01:
I have an application where I want to minimize the network bandwidth as much as possible.

When using libcamera_camera.sh with the option:
--camera-video.options=video_bitrate=1000000 (and using all sorts of different bitrates)
None of the bitrates actually affect the stream rate.
I'm checking the network stream with the command:
iftop -i wlan0
I'm wondering if the video_bitrate option isn't being passed on to libcamera?
Also, 
-camera-h264.options=bitrate=1000000
throws an error too.

andybrinkley, 2023-03-01:
I've tried it on both RPi4 and RPi3 with Camera module v2.
I video device is open, and I get a good picture over the network. The frame rate and resolution affect bandwidth (about 5Mb for full HD, right down to tiny 200pixel-high videos being about 400Kb), but the bandwidth doesn't seem to affect quality or bandwidth of stream.

ayufan, 2023-03-01:
Post a full log.

ayufan, 2023-03-01:
I just tested the `--camera-video.options=videobitrate=100000`. It worked for me. This is my pipeline:

```
device/links.c: ?: Link 0: CAMERA:capture:1[1920x1088/YUYV/3] => [SNAPSHOT:output:mplane[1920x1088/YUYV/3], VIDEO:output:mplane[1920x1088/YUYV/3]]
device/links.c: ?: Link 1: SNAPSHOT:capture:mplane[1920x1088/JPEG/3] => [SNAPSHOT-CAPTURE, STREAM-CAPTURE]
device/links.c: ?: Link 2: VIDEO:capture:mplane[1920x1088/H264/3] => [VIDEO-CAPTURE]
```

andybrinkley, 2023-03-01:
If I run `./camera_streamer` from the directory /camera_streamer, I actually get a buffering error:

```
device/v4l2/device.c: CAMERA: Device path=/dev/video0 fd=14 opened
device/v4l2/device_options.c: CAMERA: Configuring option horizontalflip (00980914) = 0
device/v4l2/device_options.c: CAMERA: Configuring option verticalflip (00980915) = 0
device/buffer_list.c: CAMERA:capture: Using: 1920x1080/pBAA, bytesperline=2400, sizeimage=2.5MiB
device/buffer_list.c: CAMERA:capture: Opened 3 buffers. Memory used: 7.4 MiB
device/v4l2/device.c: ISP: Device path=/dev/video13 fd=19 opened
device/buffer_list.c: ISP:output: Using: 1920x1080/pBAA, bytesperline=2400, sizeimage=2.5MiB
device/buffer_list.c: ISP:output: Opened 3 buffers. Memory used: 0.0 MiB
device/v4l2/buffer_list.c: ISP:capture: Device path=/dev/video14 fd=20 opened
device/buffer_list.c: ISP:capture: Using: 1920x1080/YUYV, bytesperline=3840, sizeimage=4.0MiB
device/buffer_list.c: ISP:capture: Opened 3 buffers. Memory used: 11.9 MiB
device/v4l2/device.c: SNAPSHOT: Device path=/dev/video31 fd=24 opened
device/v4l2/buffer_list.c: SNAPSHOT:output:mplane: Requested resolution=1920x1080 is unavailable. Got 1920x1088.
device/buffer_list.c: SNAPSHOT:output:mplane: Using: 1920x1056/YUYV, bytesperline=3840, sizeimage=3.9MiB
device/buffer_list.c: SNAPSHOT:output:mplane: Opened 3 buffers. Memory used: 0.0 MiB
device/buffer_list.c: SNAPSHOT:capture:mplane: Using: 1920x1056/JPEG, bytesperline=0, sizeimage=4.0MiB
device/buffer_list.c: SNAPSHOT:capture:mplane: Opened 3 buffers. Memory used: 12.0 MiB
device/v4l2/device.c: VIDEO: Device path=/dev/video11 fd=28 opened
device/buffer_list.c: VIDEO:output:mplane: Using: 1920x1080/YUYV, bytesperline=3840, sizeimage=4.0MiB
device/buffer_list.c: VIDEO:output:mplane: Opened 3 buffers. Memory used: 0.0 MiB
device/buffer_list.c: VIDEO:capture:mplane: Using: 1920x1080/H264, bytesperline=0, sizeimage=0.8MiB
device/buffer_list.c: VIDEO:capture:mplane: Opened 3 buffers. Memory used: 2.2 MiB
device/v4l2/device.c: CAMERA: ioctl(ret=-1, errno=25): Can't set FPS
device/device.c: CAMERA: Setting frame interval_us=33333 for FPS=30
device/v4l2/device_options.c: CAMERA: The 'AfTrigger=1' was failed to find.
device/v4l2/device_options.c: SNAPSHOT: Configuring option compressionquality (009d0903) = 80
device/v4l2/device_options.c: VIDEO: Configuring option repeatsequenceheader (009909e2) = 1
device/v4l2/device_options.c: VIDEO: Configuring option videobitratemode (009909ce) = 0
device/v4l2/device_options.c: VIDEO: Configuring option videobitrate (009909cf) = 2000000
device/v4l2/device_options.c: VIDEO: Configuring option repeatsequenceheader (009909e2) = 5000000
device/v4l2/device_options.c: VIDEO: Configuring option h264iframeperiod (00990a66) = 30
device/v4l2/device_options.c: VIDEO: Configuring option h264level (00990a67) = 11
device/v4l2/device_options.c: VIDEO: Configuring option h264profile (00990a6b) = 4
device/v4l2/device_options.c: VIDEO: Configuring option h264minimumqpvalue (00990a61) = 16
device/v4l2/device_options.c: VIDEO: Configuring option h264maximumqpvalue (00990a62) = 32
device/links.c: ?: Link 0: CAMERA:capture[1920x1080/pBAA/3] => [ISP:output[1920x1080/pBAA/3]]
device/links.c: ?: Link 1: ISP:capture[1920x1080/YUYV/3] => [SNAPSHOT:output:mplane[1920x1056/YUYV/3], VIDEO:output:mplane[1920x1080/YUYV/3]]
device/links.c: ?: Link 2: SNAPSHOT:capture:mplane[1920x1056/JPEG/3] => [SNAPSHOT-CAPTURE, STREAM-CAPTURE]
device/links.c: ?: Link 3: VIDEO:capture:mplane[1920x1080/H264/3] => [VIDEO-CAPTURE]
device/buffer_list.c: CAMERA:capture: Streaming started... Was 0 of 3 enqueud
device/buffer_list.c: ISP:output: Streaming started... Was 0 of 3 enqueud
device/buffer_list.c: ISP:capture: Streaming started... Was 0 of 3 enqueud
device/buffer_list.c: SNAPSHOT:output:mplane: Streaming started... Was 0 of 3 enqueud
device/buffer_list.c: VIDEO:output:mplane: Streaming started... Was 0 of 3 enqueud
device/buffer_list.c: SNAPSHOT:capture:mplane: Streaming started... Was 0 of 3 enqueud
device/buffer_list.c: VIDEO:capture:mplane: Streaming started... Was 0 of 3 enqueud
util/http/http.c: HTTP8080/0: Client connected 192.168.13.64 (fd=4).
util/http/http.c: HTTP8080/0: Request 'GET' '/' ''
util/http/http.c: HTTP8080/0: Client disconnected 192.168.13.64.
util/http/http.c: HTTP8080/1: Client connected 192.168.13.64 (fd=5).
util/http/http.c: HTTP8080/1: Request 'GET' '/webrtc' ''
util/http/http.c: HTTP8080/1: Client disconnected 192.168.13.64.
util/http/http.c: HTTP8080/2: Client connected 192.168.13.64 (fd=6).
util/http/http.c: HTTP8080/2: Request 'POST' '/webrtc' ''
output/webrtc/webrtc.cc: rtc-yzfihqoovuldpxsfpbdl: Stream requested.
util/http/http.c: HTTP8080/2: Client disconnected 192.168.13.64.
util/http/http.c: HTTP8080/3: Client connected 192.168.13.64 (fd=7).
util/http/http.c: HTTP8080/3: Request 'POST' '/webrtc' ''
output/webrtc/webrtc.cc: rtc-yzfihqoovuldpxsfpbdl: Answer received.
util/http/http.c: HTTP8080/3: Client disconnected 192.168.13.64.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
[... cut ...]
util/http/http.c: HTTP8080/2: Request 'POST' '/webrtc' ''
output/webrtc/webrtc.cc: rtc-qsqooeorttzzwyxfqxdm: Stream requested.
util/http/http.c: HTTP8080/2: Client disconnected 192.168.13.64.
util/http/http.c: HTTP8080/3: Client connected 192.168.13.64 (fd=7).
util/http/http.c: HTTP8080/3: Request 'POST' '/webrtc' ''
output/webrtc/webrtc.cc: rtc-qsqooeorttzzwyxfqxdm: Answer received.
util/http/http.c: HTTP8080/3: Client disconnected 192.168.13.64.
```
But I can't change bitrate.

I must be doing something wrong.

ayufan, 2023-03-01:
> If I run ./camera_streamer from the directory /camera_streamer, I actually get a buffering error:

This is expected and is more involved to run CSI cameras without libcamera, possible but tricky. Don't do it unless you have very specific requirements.

> If I go to camera_streamer/tools and run:

But here you did not specify anywhere a bitrate. How you do it?

andybrinkley, 2023-03-01:
Ok, This is the response when I run with bitrate command (I chose 256 just to make sure):

```
pi@raspberrypi:~/camera-streamer/tools $ ./libcamera_camera.sh -camera-video.options=video_bitrate=256
++ nproc
+ make -j4
make: 'camera-streamer' is up to date.
+ ./camera-streamer -camera-type=libcamera -camera-format=YUYV -camera-video.options=video_bitrate=256
[0:51:00.894919070] [2808]  INFO Camera camera_manager.cpp:299 libcamera v0.0.3+40-9b860a66
[0:51:00.925973624] [2819]  WARN RPI raspberrypi.cpp:1308 Mismatch between Unicam and CamHelper for embedded data usage!
[0:51:00.927098240] [2819]  INFO RPI raspberrypi.cpp:1425 Registered camera /base/soc/i2c0mux/i2c@1/imx219@10 to Unicam device /dev/media4 and ISP device /dev/media1
device/libcamera/device.cc: CAMERA: Device path=/base/soc/i2c0mux/i2c@1/imx219@10 opened
[0:51:00.927756640] [2808]  INFO Camera camera.cpp:1028 configuring streams: (0) 1920x1080-SBGGR10_CSI2P
[0:51:00.928196264] [2819]  INFO RPI raspberrypi.cpp:805 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected sensor format: 1920x1080-SBGGR10_1X10 - Selected unicam format: 1920x1080-pBAA
device/buffer_list.c: CAMERA:capture: Using: 1920x1080/BG10, bytesperline=2400, sizeimage=0.0MiB
device/buffer_list.c: CAMERA:capture: Opened 3 buffers. Memory used: 7.4 MiB
[0:51:00.936202650] [2808]  INFO Camera camera.cpp:1028 configuring streams: (0) 1920x1080-SBGGR10_CSI2P (1) 1920x1088-YUYV
[0:51:00.936608719] [2819]  INFO RPI raspberrypi.cpp:805 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected sensor format: 1920x1080-SBGGR10_1X10 - Selected unicam format: 1920x1080-pBAA
device/buffer_list.c: CAMERA:capture:1: Using: 1920x1088/YUYV, bytesperline=3840, sizeimage=0.0MiB
device/buffer_list.c: CAMERA:capture:1: Opened 3 buffers. Memory used: 12.0 MiB
device/v4l2/device.c: SNAPSHOT: Device path=/dev/video31 fd=39 opened
device/buffer_list.c: SNAPSHOT:output:mplane: Using: 1920x1088/YUYV, bytesperline=3840, sizeimage=4.0MiB
device/buffer_list.c: SNAPSHOT:output:mplane: Opened 3 buffers. Memory used: 0.0 MiB
device/buffer_list.c: SNAPSHOT:capture:mplane: Using: 1920x1088/JPEG, bytesperline=0, sizeimage=4.0MiB
device/buffer_list.c: SNAPSHOT:capture:mplane: Opened 3 buffers. Memory used: 12.0 MiB
device/v4l2/device.c: VIDEO: Device path=/dev/video11 fd=43 opened
device/buffer_list.c: VIDEO:output:mplane: Using: 1920x1088/YUYV, bytesperline=3840, sizeimage=4.0MiB
device/buffer_list.c: VIDEO:output:mplane: Opened 3 buffers. Memory used: 0.0 MiB
device/buffer_list.c: VIDEO:capture:mplane: Using: 1920x1088/H264, bytesperline=0, sizeimage=0.8MiB
device/buffer_list.c: VIDEO:capture:mplane: Opened 3 buffers. Memory used: 2.2 MiB
device/device.c: CAMERA: Setting frame interval_us=0 for FPS=30
device/v4l2/device_options.c: SNAPSHOT: Configuring option compressionquality (009d0903) = 80
device/v4l2/device_options.c: VIDEO: Configuring option repeatsequenceheader (009909e2) = 1
device/v4l2/device_options.c: VIDEO: Configuring option videobitratemode (009909ce) = 0
device/v4l2/device_options.c: VIDEO: Configuring option videobitrate (009909cf) = 2000000
device/v4l2/device_options.c: VIDEO: Configuring option repeatsequenceheader (009909e2) = 5000000
device/v4l2/device_options.c: VIDEO: Configuring option h264iframeperiod (00990a66) = 30
device/v4l2/device_options.c: VIDEO: Configuring option h264level (00990a67) = 11
device/v4l2/device_options.c: VIDEO: Configuring option h264profile (00990a6b) = 4
device/v4l2/device_options.c: VIDEO: Configuring option h264minimumqpvalue (00990a61) = 16
device/v4l2/device_options.c: VIDEO: Configuring option h264maximumqpvalue (00990a62) = 32
device/v4l2/device_options.c: VIDEO: Configuring option videobitrate (009909cf) = 256
device/links.c: ?: Link 0: CAMERA:capture:1[1920x1088/YUYV/3] => [SNAPSHOT:output:mplane[1920x1088/YUYV/3], VIDEO:output:mplane[1920x1088/YUYV/3]]
device/links.c: ?: Link 1: SNAPSHOT:capture:mplane[1920x1088/JPEG/3] => [SNAPSHOT-CAPTURE, STREAM-CAPTURE]
device/links.c: ?: Link 2: VIDEO:capture:mplane[1920x1088/H264/3] => [VIDEO-CAPTURE]
device/buffer_list.c: CAMERA:capture:1: Streaming started... Was 0 of 3 enqueud
device/buffer_list.c: SNAPSHOT:output:mplane: Streaming started... Was 0 of 3 enqueud
device/buffer_list.c: VIDEO:output:mplane: Streaming started... Was 0 of 3 enqueud
device/buffer_list.c: SNAPSHOT:capture:mplane: Streaming started... Was 0 of 3 enqueud
device/buffer_list.c: VIDEO:capture:mplane: Streaming started... Was 0 of 3 enqueud
util/http/http.c: HTTP8080/0: Client connected 192.168.13.64 (fd=4).
util/http/http.c: HTTP8080/0: Request 'GET' '/webrtc' ''
util/http/http.c: HTTP8080/0: Client disconnected 192.168.13.64.
util/http/http.c: HTTP8080/1: Client connected 192.168.13.64 (fd=5).
util/http/http.c: HTTP8080/1: Request 'POST' '/webrtc' ''
output/webrtc/webrtc.cc: rtc-fsvdwmhtgmtvxqwppdhv: Stream requested.
util/http/http.c: HTTP8080/1: Client disconnected 192.168.13.64.
util/http/http.c: HTTP8080/2: Client connected 192.168.13.64 (fd=6).
util/http/http.c: HTTP8080/2: Request 'POST' '/webrtc' ''
output/webrtc/webrtc.cc: rtc-fsvdwmhtgmtvxqwppdhv: Answer received.
util/http/http.c: HTTP8080/2: Client disconnected 192.168.13.64.
output/webrtc/webrtc.cc: rtc-fsvdwmhtgmtvxqwppdhv: Client removed: stream closed.

```

ayufan, 2023-03-01:
And then what happens how you validate this? Because I do basically the same.

andybrinkley, 2023-03-01:
Well, for a start, the image looks exactly the same, I was expecting some sort of degrading of the picture. either we hang ups, or low quality. Secondly, I monitor the network traffic with iftop, seeing the different bandwidth at different resolutions and framerates. But the bitrate doesn't affect it. 
This is a screengrab of the network, with the above stream running (webrtc)
[https://imgur.com/a/vPgqDaN](url)

andybrinkley, 2023-03-01:
It says 3.5Mb.

ayufan, 2023-03-01:
Interesting. I clearly see a difference in BW. There is a lower limit. I also assume that it will naturally consume more with higher resolution.

```
                  video_bitrate 0x009909cf (int)    : min=25000 max=25000000 step=25000 default=10000000 value=10000000
```

I can change the bitrate dynamically and have see a big difference in image quality and BW:
- http://mainsailos.home:8080/option?videobitrate=25000000
- http://mainsailos.home:8080/option?videobitrate=2500000
- http://mainsailos.home:8080/option?videobitrate=250000

It is not exactly as requested, but roughly around that.

ayufan, 2023-03-01:
> When I change the options as you do, the log says:
>
> util/http/http.c: HTTP8080/1: Request 'GET' '/option' 'videobitrate=25000'
> device/v4l2/device_options.c: SNAPSHOT: The 'videobitrate=25000' was failed to find.
> device/v4l2/device_options.c: VIDEO: Configuring option videobitrate (009909cf) = 25000
> util/http/http.c: HTTP8080/1: Client disconnected 192.168.13.64.

Seems OK to me.

andybrinkley, 2023-03-01:
How are you viewing the feed, webrtc?

ayufan, 2023-03-01:
Yes

andybrinkley, 2023-03-01:
OK, thanks for your help, I'll do some further digging, and try some other things. Maybe I'll try recompiling...