ayufan/camera-streamer #46: Work with Arducam Multicam?

ZakEspley, 2023-03-04:
Hi, this looks like an amazing setup.

I was wondering if you know if it would work with a multicamera from arducam? One of the older models, not the newer camarray. Particularly this one:

https://www.arducam.com/product/multi-camera-v2-1-adapter-raspberry-pi/

I am trying to use the libcamera or the csi -camera scripts but neither is able to see the camera to show me the output, but I can get the camera working via other means, so I know the board and camera work. Do you have any idea how I might go modify the scripts to work with this?

Additionally, is the /video feed able to be seen by multiple clients at once or is it one client at a time?

Thanks,
Zak

kbingham, 2023-03-06:
The camera id will be different for each camera when connected through the arducam multiplexor. try something like 'cam --list' to see what the camera ids are.

kbingham, 2023-03-06:
Only one camera can be operated at a time through this device.

ayufan, 2023-03-06:
> Additionally, is the /video feed able to be seen by multiple clients at once or is it one client at a time?

Multiple