ayufan/camera-streamer #48: SPS ans PPS Header on RTSP

Tormund1985, 2023-03-08:
Good evening again,

I add a problem that I am also experiencing.

I normally burn RTSP streams to disk with the following command:

`ffmpeg -hide_banner -y -loglevel error -rtsp_transport tcp -use_wallclock_as_timestamps 1 -i rtsp://localhost:8554/stream.h264 -vcodec copy -acodec copy -f segment -reset_timestamps 1 -segment_time 3600 -segment_format mkv -segment_atclocktime 1 -strftime 1 /home/pi/Videos/IMX462_%Y%m%d_%H%M%S.mkv`

It is functional with rtsp from other streamers like v4l2rtspserver or from manufacturers like Dahua, Hikvision, Avigilon etc....

The following output is obtained:
```

[h264 @ 0xf92db0] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0xf92db0] decode_slice_header error
[h264 @ 0xf92db0] no frame!
[h264 @ 0xf92db0] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0xf92db0] decode_slice_header error
[h264 @ 0xf92db0] no frame!
[h264 @ 0xf92db0] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0xf92db0] decode_slice_header error
[h264 @ 0xf92db0] no frame!
[h264 @ 0xf92db0] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0xf92db0] decode_slice_header error
[h264 @ 0xf92db0] no frame!
[h264 @ 0xf92db0] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0xf92db0] decode_slice_header error
[h264 @ 0xf92db0] no frame!
[h264 @ 0xf92db0] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0xf92db0] decode_slice_header error
[h264 @ 0xf92db0] no frame!
[h264 @ 0xf92db0] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0xf92db0] decode_slice_header error
[h264 @ 0xf92db0] no frame!
[h264 @ 0xf92db0] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0xf92db0] decode_slice_header error
[h264 @ 0xf92db0] no frame!
[h264 @ 0xf92db0] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0xf92db0] decode_slice_header error
[h264 @ 0xf92db0] no frame!
[h264 @ 0xf92db0] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0xf92db0] decode_slice_header error
[h264 @ 0xf92db0] no frame!
[h264 @ 0xf92db0] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0xf92db0] decode_slice_header error
[h264 @ 0xf92db0] no frame!
[h264 @ 0xf92db0] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0xf92db0] decode_slice_header error
[h264 @ 0xf92db0] no frame!
[h264 @ 0xf92db0] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0xf92db0] decode_slice_header error
[h264 @ 0xf92db0] no frame!
[h264 @ 0xf92db0] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0xf92db0] decode_slice_header error
[h264 @ 0xf92db0] no frame!
[h264 @ 0xf92db0] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0xf92db0] decode_slice_header error
[h264 @ 0xf92db0] no frame!
[h264 @ 0xf92db0] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0xf92db0] decode_slice_header error
[h264 @ 0xf92db0] no frame!
[h264 @ 0xf92db0] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0xf92db0] decode_slice_header error
[h264 @ 0xf92db0] no frame!
[h264 @ 0xf92db0] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0xf92db0] decode_slice_header error
[h264 @ 0xf92db0] no frame!
[h264 @ 0xf92db0] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0xf92db0] decode_slice_header error
[h264 @ 0xf92db0] no frame!
[h264 @ 0xf92db0] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0xf92db0] decode_slice_header error
[h264 @ 0xf92db0] no frame!
[h264 @ 0xf92db0] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0xf92db0] decode_slice_header error
[h264 @ 0xf92db0] no frame!
[h264 @ 0xf92db0] error while decoding MB 1 19, bytestream -7
[segment @ 0xfa9d70] Failed to open segment '/home/pi/Videos/IMX462_20230308_180712.mkv'
Could not write header for output file #0 (incorrect codec parameters ?): No such file or directory

```

However with this streamer it doesn't, because apparently the SPS and PPS header data is missing, which I think would be added with the "--inline" command in libcamera-vid.

`--inline, -ih       #Insert PPS, SPS header`

Is there any other way to record this video stream locally? Obviously with libcamera-vid I can't since it is in use.

Many thanks for your support!

ayufan, 2023-03-08:
Interesting. I do not see this, but will check.

ayufan, 2023-03-08:
It does in fact happen early:

```
[h264 @ 0xfbec10] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0xfbec10] decode_slice_header error
[h264 @ 0xfbec10] no frame!
[h264 @ 0xfbec10] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0xfbec10] decode_slice_header error
[h264 @ 0xfbec10] no frame!
[h264 @ 0xfbec10] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0xfbec10] decode_slice_header error
[h264 @ 0xfbec10] no frame!
[h264 @ 0xfbec10] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0xfbec10] decode_slice_header error
[h264 @ 0xfbec10] no frame!
[h264 @ 0xfbec10] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0xfbec10] decode_slice_header error
[h264 @ 0xfbec10] no frame!
[h264 @ 0xfbec10] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0xfbec10] decode_slice_header error
[h264 @ 0xfbec10] no frame!
[h264 @ 0xfbec10] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0xfbec10] decode_slice_header error
[h264 @ 0xfbec10] no frame!
[h264 @ 0xfbec10] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0xfbec10] decode_slice_header error
[h264 @ 0xfbec10] no frame!
[h264 @ 0xfbec10] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0xfbec10] decode_slice_header error
[h264 @ 0xfbec10] no frame!
[h264 @ 0xfbec10] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0xfbec10] decode_slice_header error
[h264 @ 0xfbec10] no frame!
[h264 @ 0xfbec10] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0xfbec10] decode_slice_header error
[h264 @ 0xfbec10] no frame!
[h264 @ 0xfbec10] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0xfbec10] decode_slice_header error
[h264 @ 0xfbec10] no frame!
Input #0, rtsp, from 'rtsp://localhost:8554/stream.h264':
  Metadata:
    title           : streamed by the LIVE555 Media Server
    comment         : stream.h264
  Duration: N/A, start: 1678298192.223089, bitrate: N/A
    Stream #0:0: Video: h264 (High), yuv420p(progressive), 1920x1080, 30 fps, 30 tbr, 90k tbn, 60 tbc
[segment @ 0x11dc8f0] Opening '/run/IMX462_20230308_185633.mkv' for writing
Output #0, segment, to '/run/IMX462_%Y%m%d_%H%M%S.mkv':
  Metadata:
    title           : streamed by the LIVE555 Media Server
    comment         : stream.h264
    encoder         : Lavf58.45.100
    Stream #0:0: Video: h264 (High), yuv420p(progressive), 1920x1080, q=2-31, 30 fps, 30 tbr, 1k tbn, 30 tbc
Stream mapping:
  Stream #0:0 -> #0:0 (copy)
Press [q] to stop, [?] for help
frame=  328 fps= 31 q=-1.0 size=N/A time=00:00:11.33 bitrate=N/A speed=1.08x    
```

But seems to be resolved after about 1s.

ayufan, 2023-03-08:
@Tormund1985 I found a bug related to keyframe handling. I also found a deadlock in RTSP stream. Both are fixed on `master`.

closed 2023-03-08 by ayufan

Tormund1985, 2023-03-09:
> Do `git pull` and this should be enough.

After do git pull now it's working! It shows "non-existing PPS 0 referenced" but finally start recording.

Many Thanks!

ayufan, 2023-03-09:
Interesting. I don't see it. I think you might have hit this deadlock yourself that got fixed.

Tormund1985, 2023-03-10:
> Interesting. I don't see it. I think you might have hit this deadlock yourself that got fixed.
Nothing... it's the same as with RTSP transmission, it finally drops, the quality is very poor. I attach a video so you can see for yourself.

Thanks for the support. 

https://drive.google.com/file/d/1zl5EHL0tMLmu_1_TXpAn4j_eWDJNCKUm/view?usp=sharing

Tormund1985, 2023-03-10:
> @Tormund1985 Do you maybe have connection that might not be able to transmit all frames? If yes, some of those might be truncated.

Not possible I am connected by Wifi one meter lenght.

ayufan, 2023-03-10:
@Tormund1985 Something like this: `curl http://mainsailos.home:8080/status.json | jq '.endpoints.rtsp'`.

Tormund1985, 2023-03-10:
> curl http://mainsailos.home:8080/status.json | jq '.endpoints.rtsp'

```
pi@raspberrypi:~ $ sudo curl http://localhost:8080/status.json | jq '.endpoints.rtsp'
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  1883  100  1883    0     0  38428      0 --:--:-- --:--:-- --:--:-- 39229
{
  "enabled": true,
  "output": "video",
  "uri": "rtsp://localhost:8554/stream.h264"
```