ayufan/camera-streamer #50: webrtc not working anymore

philippebourcier, 2023-03-18:
Everything was working fine and for whatever reasons my 15 days old release stopped working for webrtc (the rest is fine)...
So I tried a newer version, just in case, but same problem.

Here is a strace :
[pid  1632] write(2, "util/http/http.c: HTTP8080/1: Client connected 192.168... (fd=5).\n", 69util/http/http.c: HTTP8080/1: Client connected 192.168... (fd=5).) = 69
...
[pid  1632] read(5, "POST /webrtc HTTP/1.1\r\nHost: 192.168...:8080\r\nConnection: keep-alive\r\nContent-Length: 18\r\nPragma: no-cache\r\nCache-Control: no-cache\r\nUser-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/111.0.0.0 Safari/537.36\r\nContent-Type: application/json\r\nAccept: */*\r\nOrigin: http://192.168...:8080\r\nReferer: http://192.168...:8080/webrtc\r\nAccept-Encoding: gzip, deflate\r\nAccept-Language: fr,en-US;q=0.9,en;q=0.8\r\nCookie: motion_detected_1=false; monitor_info_1=\"\"; capture_fps_1=5.0\r\n\r\n{\"type\":\"request\"}", 4096) = 549
[pid  1632] write(2, "util/http/http.c: HTTP8080/1: Request 'POST' '/webrtc' ''\n", 58util/http/http.c: HTTP8080/1: Request 'POST' '/webrtc''') = 58
[pid  1632] lseek(5, -18, SEEK_CUR)     = -1 ESPIPE (Illegal seek)
[pid  1632] close(5)                    = 0
[pid  1632] write(2, "util/http/http.c: HTTP8080/1: Client disconnected 192.168...\n", 65util/http/http.c: HTTP8080/1: Client disconnected 192.168...) = 65

[image]

I tried to debug a bit more... we go through http_404 and http_write_response successfully, so I don't get what's going on...
All other streams and snapshots are working just fine.

Here is the payload, just in case :
[image]

philippebourcier, 2023-04-25:
@ayufan Could you look into this ?
I had really great results in terms of latency when webrtc was working.
ps: It seems to be a duplicate of #43 btw...

ayufan, 2023-04-25:
Interesting, what else changed? Maybe you updated system, or something?

Did you recompile the `camera-streamer` fully?

philippebourcier, 2023-04-26:
Oh man, how stupid I feel... 🤦‍♂️

I have been debugging this, this morning... I looked at the code and I was like... Humm, this is strange I'm going through code that is behind a "else" where I wouldn't be compiling with LIBDATACHANNEL.
So I looked at the previous build that worked well and found out I had libdatachannel installed... but not in the recent build.

So, I just added the "--recursive" option to git and recompiled and everything is obviously back to normal ! 😅🤯

I guess it would have helped me to have a big warning, like "USE_LIBDATACHANNEL not set, WebRTC will be disabled !!!"... in red. 😬😅

closed 2023-04-26 by philippebourcier