ayufan/camera-streamer #58: Cannot set Video Resolution for PiCamera Module 3

JoelBecker1998, 2023-05-09:
I have a Camera Module 3 that should support 2304x1296 with 30fps. [see here](https://www.raspberrypi.com/documentation/accessories/camera.html#hardware-specification)

I'm using this command:
`camera-streamer/tools/libcamera_camera.sh --camera-type=libcamera --camera-format=YUYV --camera-fps=30 --camera-path=/base/soc/i2c0mux/i2c@0/imx708@1a --http-port=8080 --camera-width=2304 --camera-height=1296 --camera-hflip=1 --camera-vflip=1 `

The stream launches, but only with 1920x1080 with 25 fps.

The output looks like this:
```pi@camerapi1:~ $ camera-streamer/tools/libcamera_camera.sh --camera-path=/base/soc/i2c0mux/i2c@1/imx708@1a --http-port=8081 --camera-width=2304 --camera-height=1296 --camera-hflip=1 --camera-vflip=1 --log-stats=1
++ nproc
+ make -j4
echo "#define GIT_VERSION \"v0.1-21-g54f538e\"\n#define GIT_REVISION \"54f538e\"" > version.h.tmp; \
        diff -u version.h version.h.tmp || mv version.h.tmp version.h; \
        rm -f version.h.tmp
make camera-streamer
make[1]: Entering directory '/home/pi/camera-streamer'
make[1]: 'camera-streamer' is up to date.
make[1]: Leaving directory '/home/pi/camera-streamer'
+ ./camera-streamer -camera-type=libcamera -camera-format=YUYV --camera-path=/base/soc/i2c0mux/i2c@1/imx708@1a --http-port=8081 --camera-width=2304 --camera-height=1296 --camera-hflip=1 --camera-vflip=1 --log-stats=1
util/opts/opts.c: ?: Usage of '-camera-type=libcamera' is deprecated change to '--camera-type=libcamera'.
util/opts/opts.c: ?: Usage of '-camera-format=YUYV' is deprecated change to '--camera-format=YUYV'.
./camera-streamer Version: v0.1-21-g54f538e (54f538e)
[1:14:22.748467097] [3530]  INFO Camera camera_manager.cpp:299 libcamera v0.0.4+22-923f5d70
[1:14:22.878250245] [3541]  INFO RPI raspberrypi.cpp:1476 Registered camera /base/soc/i2c0mux/i2c@0/imx708@1a to Unicam device /dev/media4 and ISP device /dev/media1
[1:14:22.891141912] [3541]  INFO RPI raspberrypi.cpp:1476 Registered camera /base/soc/i2c0mux/i2c@1/imx708@1a to Unicam device /dev/media5 and ISP device /dev/media3
device/libcamera/device.cc: CAMERA: Device path=/base/soc/i2c0mux/i2c@1/imx708@1a opened
[1:14:22.892265264] [3530]  INFO Camera camera.cpp:1028 configuring streams: (0) 1920x1080-YUYV
[1:14:22.892616801] [3541]  INFO RPI raspberrypi.cpp:851 Sensor: /base/soc/i2c0mux/i2c@1/imx708@1a - Selected sensor format: 2304x1296-SRGGB10_1X10 - Selected unicam format: 2304x1296-pRAA
[1:14:22.894984116] [3530]  INFO Camera camera.cpp:1028 configuring streams: (0) 1920x1080-YUYV (1) 2304x1296-SRGGB10_CSI2P
[1:14:22.895295597] [3541]  INFO RPI raspberrypi.cpp:851 Sensor: /base/soc/i2c0mux/i2c@1/imx708@1a - Selected sensor format: 2304x1296-SRGGB10_1X10 - Selected unicam format: 2304x1296-pRAA
device/buffer_list.c: CAMERA:capture: Using: 1920x1080/YUYV, buffers=3, bytesperline=3840, sizeimage=0.0MiB
device/buffer_list.c: CAMERA:capture: Opened 3 buffers. Memory used: 11.9 MiB
device/buffer_list.c: CAMERA:capture:1: Using: 2304x1296/RG10, buffers=3, bytesperline=2880, sizeimage=0.0MiB
device/buffer_list.c: CAMERA:capture:1: Opened 3 buffers. Memory used: 10.7 MiB
device/v4l2/device.c: SNAPSHOT: Device path=/dev/video31 fd=52 opened
device/v4l2/buffer_list.c: SNAPSHOT:output:mplane: Requested resolution=1920x1080 is unavailable. Got 1920x1088.
device/buffer_list.c: SNAPSHOT:output:mplane: Using: 1920x1056/YUYV, buffers=3, bytesperline=3840, sizeimage=3.9MiB
device/buffer_list.c: SNAPSHOT:output:mplane: Opened 3 buffers. Memory used: 0.0 MiB
device/buffer_list.c: SNAPSHOT:capture:mplane: Using: 1920x1056/JPEG, buffers=3, bytesperline=0, sizeimage=4.0MiB
device/buffer_list.c: SNAPSHOT:capture:mplane: Opened 3 buffers. Memory used: 12.0 MiB
device/v4l2/device.c: VIDEO: Device path=/dev/video11 fd=56 opened
device/buffer_list.c: VIDEO:output:mplane: Using: 1920x1080/YUYV, buffers=3, bytesperline=3840, sizeimage=4.0MiB
device/buffer_list.c: VIDEO:output:mplane: Opened 3 buffers. Memory used: 0.0 MiB
device/buffer_list.c: VIDEO:capture:mplane: Using: 1920x1080/H264, buffers=3, bytesperline=0, sizeimage=0.8MiB
device/buffer_list.c: VIDEO:capture:mplane: Opened 3 buffers. Memory used: 2.2 MiB
device/device.c: CAMERA: Setting frame interval_us=0 for FPS=30
device/libcamera/device.cc: CAMERA: Configuring option aftrigger (00000021, type=3) = 1
device/v4l2/device_options.c: SNAPSHOT: Configuring option compressionquality (009d0903) = 80
device/v4l2/device_options.c: VIDEO: Configuring option repeatsequenceheader (009909e2) = 1
device/v4l2/device_options.c: VIDEO: Configuring option videobitratemode (009909ce) = 0
device/v4l2/device_options.c: VIDEO: Configuring option videobitrate (009909cf) = 2000000
device/v4l2/device_options.c: VIDEO: Configuring option repeatsequenceheader (009909e2) = 5000000
device/v4l2/device_options.c: VIDEO: Configuring option h264iframeperiod (00990a66) = 30
device/v4l2/device_options.c: VIDEO: Configuring option h264level (00990a67) = 11
device/v4l2/device_options.c: VIDEO: Configuring option h264profile (00990a6b) = 4
device/v4l2/device_options.c: VIDEO: Configuring option h264minimumqpvalue (00990a61) = 16
device/v4l2/device_options.c: VIDEO: Configuring option h264maximumqpvalue (00990a62) = 32
device/links.c: ?: Link 0: CAMERA:capture[1920x1080/YUYV/3] => [SNAPSHOT:output:mplane[1920x1056/YUYV/3], VIDEO:output:mplane[1920x1080/YUYV/3]]
device/links.c: ?: Link 1: SNAPSHOT:capture:mplane[1920x1056/JPEG/3] => [SNAPSHOT-CAPTURE, STREAM-CAPTURE]
device/links.c: ?: Link 2: VIDEO:capture:mplane[1920x1080/H264/3] => [VIDEO-CAPTURE]
[1:14:22.925715819] [3546]  WARN IPARPI raspberrypi.cpp:1183 Could not set AF_TRIGGER - no AF algorithm or not Auto
device/buffer_list.c: CAMERA:capture: Streaming started... Was 0 of 3 enqueud
device/buffer_list.c: SNAPSHOT:output:mplane: Streaming started... Was 0 of 3 enqueud
device/buffer_list.c: VIDEO:output:mplane: Streaming started... Was 0 of 3 enqueud
device/buffer_list.c: SNAPSHOT:capture:mplane: Streaming started... Was 0 of 3 enqueud
device/buffer_list.c: VIDEO:capture:mplane: Streaming started... Was 0 of 3 enqueud
Statistics: [  CAMERA  0 FPS/ 0 D/ -1ms/ -1ms/P/Q0:O0:C0] [SNAPSHOT  0 FPS/ 0 D/ -1ms/ -1ms/P/Q0:O0:C0] [   VIDEO  0 FPS/ 0 D/ -1ms/ -1ms/P/Q0:O0:C0]
Statistics: [  CAMERA  0 FPS/ 0 D/ -1ms/ -1ms/P/Q0:O0:C0] [SNAPSHOT  0 FPS/ 0 D/ -1ms/ -1ms/P/Q0:O0:C0] [   VIDEO  0 FPS/ 0 D/ -1ms/ -1ms/P/Q0:O0:C0]
Statistics: [  CAMERA  0 FPS/ 0 D/ -1ms/ -1ms/P/Q0:O0:C0] [SNAPSHOT  0 FPS/ 0 D/ -1ms/ -1ms/P/Q0:O0:C0] [   VIDEO  0 FPS/ 0 D/ -1ms/ -1ms/P/Q0:O0:C0]
util/http/http.c: HTTP8081/0: Client connected 172.16.10.88 (fd=4).
util/http/http.c: HTTP8081/0: Request 'GET' '/webrtc' ''
util/http/http.c: HTTP8081/0: Client disconnected 172.16.10.88.
util/http/http.c: HTTP8081/1: Client connected 172.16.10.88 (fd=5).
util/http/http.c: HTTP8081/1: Request 'POST' '/webrtc' ''
Statistics: [  CAMERA  0 FPS/ 0 D/ -1ms/ -1ms/P/Q0:O0:C0] [SNAPSHOT  0 FPS/ 0 D/ -1ms/ -1ms/P/Q0:O0:C0] [   VIDEO  0 FPS/ 0 D/ -1ms/ -1ms/P/Q0:O0:C0]
output/webrtc/webrtc.cc: rtc-csabmfqnvqtojkvhwrin: Stream requested.
util/http/http.c: HTTP8081/1: Client disconnected 172.16.10.88.
util/http/http.c: HTTP8081/2: Client connected 172.16.10.88 (fd=6).
util/http/http.c: HTTP8081/2: Request 'POST' '/webrtc' ''
output/webrtc/webrtc.cc: rtc-csabmfqnvqtojkvhwrin: Answer received.
util/http/http.c: HTTP8081/2: Client disconnected 172.16.10.88.
Statistics: [  CAMERA 15 FPS/ 0 D/ 40ms/ 86ms/S/Q0:O0:C1] [SNAPSHOT  0 FPS/ 0 D/ -1ms/ -1ms/P/Q0:O0:C0] [   VIDEO 13 FPS/ 0 D/ 76ms/ 35ms/S/Q1:O1:C1]
Statistics: [  CAMERA 22 FPS/ 0 D/ 39ms/ 98ms/S/Q0:O0:C2] [SNAPSHOT  0 FPS/ 0 D/ -1ms/ -1ms/P/Q0:O0:C0] [   VIDEO 24 FPS/ 0 D/ 73ms/ 33ms/S/Q0:O1:C0]
Statistics: [  CAMERA 24 FPS/ 0 D/ 42ms/ 69ms/S/Q0:O0:C2] [SNAPSHOT  0 FPS/ 0 D/ -1ms/ -1ms/P/Q0:O0:C0] [   VIDEO 24 FPS/ 0 D/ 81ms/ 37ms/S/Q0:O1:C0]
Statistics: [  CAMERA 24 FPS/ 0 D/ 38ms/ 61ms/S/Q0:O0:C2] [SNAPSHOT  0 FPS/ 0 D/ -1ms/ -1ms/P/Q0:O0:C0] [   VIDEO 24 FPS/ 0 D/ 73ms/ 34ms/S/Q0:O1:C0]
Statistics: [  CAMERA 24 FPS/ 0 D/ 38ms/ 88ms/S/Q0:O0:C1] [SNAPSHOT  0 FPS/ 0 D/ -1ms/ -1ms/P/Q0:O0:C0] [   VIDEO 22 FPS/ 0 D/ 73ms/ 34ms/S/Q1:O1:C1]
Statistics: [  CAMERA 23 FPS/ 0 D/ 39ms/ 88ms/S/Q0:O0:C1] [SNAPSHOT  0 FPS/ 0 D/ -1ms/ -1ms/P/Q0:O0:C0] [   VIDEO 24 FPS/ 0 D/ 74ms/ 34ms/S/Q1:O0:C0]
Statistics: [  CAMERA 23 FPS/ 0 D/ 38ms/ 65ms/S/Q0:O0:C2] [SNAPSHOT  0 FPS/ 0 D/ -1ms/ -1ms/P/Q0:O0:C0] [   VIDEO 24 FPS/ 0 D/ 70ms/ 30ms/S/Q0:O1:C0]
[... cut ...] s/S/Q0:O1:C0]
Statistics: [  CAMERA 25 FPS/ 0 D/ 38ms/ 64ms/S/Q0:O0:C1] [SNAPSHOT  0 FPS/ 0 D/ -1ms/ -1ms/P/Q0:O0:C0] [   VIDEO 24 FPS/ 0 D/ 79ms/ 41ms/S/Q0:O2:C1]
Statistics: [  CAMERA 23 FPS/ 0 D/ 39ms/ 88ms/S/Q0:O0:C1] [SNAPSHOT  0 FPS/ 0 D/ -1ms/ -1ms/P/Q0:O0:C0] [   VIDEO 22 FPS/ 0 D/ 74ms/ 33ms/S/Q1:O1:C1]
Statistics: [  CAMERA 23 FPS/ 0 D/ 39ms/ 98ms/S/Q0:O0:C1] [SNAPSHOT  0 FPS/ 0 D/ -1ms/ -1ms/P/Q0:O0:C0] [   VIDEO 24 FPS/ 0 D/ 81ms/ 41ms/S/Q0:O2:C1]```

Any help is appreciated, thanks!

ayufan, 2023-05-09:
@JoelBecker1998 The HW encoder does require max 1920x1920 on output resolution. What happens here: camera sensor is capturing with `2304x1296`, but ISP is producing `1920x1080` (downscaling) to fit the HW encoder requirements.

It works as expected. The only tricky part is you should get 30FPS. Can you try with `--camera-nbufs=4` if this produces 30PFS?

JoelBecker1998, 2023-05-09:
Ah alright thanks, so no way to get a higher resolution stream over webrtc?

With `--camera-nbufs=4` i get 30 fps, thanks!

JoelBecker1998, 2023-05-09:
Thanks, no worries. Great project btw.

closed 2023-05-09 by JoelBecker1998