ayufan/camera-streamer #62: device/camera/camera_decoder.c: CAMERA: Cannot find 'MJPG' decoder

jattie-ire, 2023-05-26:
Hi Kamil,

I am working on the migration process for Octoprint to camera-streamer as the defacto standard, implementing that as a Linuc container using LXC.

* I am running in a Linux container(LXC)
* I am running on Ubuntu 22.04 LTS
* I could not install liblivemedia-dev, not sure how to resolve it and if it is only relevant to raspberry Pi hardware.
* The build steps went fine, and I can build the streamer

It just fails at running and I have been at it for days now, so was hoping to get some guidance.

```bash
root@ustream:~/camera-streamer# ./camera-streamer 
util/opts/opts.c: ?: Usage of '-camera-list_options' is deprecated change to '--camera-list_options'.
./camera-streamer Version: v0.1-23-gf0634f9 (f0634f9)
device/v4l2/device.c: CAMERA: Device path=/dev/video0 fd=3 opened
device/v4l2/device_options.c: CAMERA: The 'horizontal_flip=0' was failed to find.
device/v4l2/device_options.c: CAMERA: The 'vertical_flip=0' was failed to find.
device/camera/camera_input.c: CAMERA: Disabling DMA since device uses USB (which is likely not working properly).
device/v4l2/buffer_list.c: CAMERA:capture: Requested resolution=1920x1080 is unavailable. Got 1280x720. Accepted
device/buffer_list.c: CAMERA:capture: Using: 1280x720/MJPG, buffers=3, bytesperline=0, sizeimage=1.8MiB
device/buffer_list.c: CAMERA:capture: Opened 3 buffers. Memory used: 5.3 MiB
device/camera/camera_decoder.c: CAMERA: Cannot find 'MJPG' decoder
device/camera/camera_output.c: CAMERA: Cannot find source for 'VIDEO' for one of the formats 'H264'.
```

Is my issue related to this missing component `liblivemedia-dev` or do I just miss some basic options passed in?

Thanks

Jattie

ayufan, 2023-05-26:
@jattie-ire Is this RPI?

jattie-ire, 2023-05-26:
No, it's a Linux PC

ayufan, 2023-05-26:
@jattie-ire Currently the application will start to run and decode MJPG to re-encode it into H264. Since there's no HW decoder you have to run it with: `--camera-video.disabled`. Only MJPG stream will be available then.

jattie-ire, 2023-05-26:
Hi Kamil, I just want to use the same streamer to allow me to be compatible and portable for pi support, so will try to get it working with the 'disabled' flag now.

ayufan, 2023-05-26:
The `disabled` will make it work, as this is something that I tested on other platforms.

jattie-ire, 2023-05-26:
It does indeed, thanks for the help. 
```
root@ustream:~/camera-streamer# ./camera-streamer --camera-video.disabled
./camera-streamer Version: v0.1-23-gf0634f9 (f0634f9)
device/v4l2/device.c: CAMERA: Device path=/dev/video0 fd=14 opened
device/v4l2/device_options.c: CAMERA: The 'horizontal_flip=0' was failed to find.
device/v4l2/device_options.c: CAMERA: The 'vertical_flip=0' was failed to find.
device/camera/camera_input.c: CAMERA: Disabling DMA since device uses USB (which is likely not working properly).
device/v4l2/buffer_list.c: CAMERA:capture: Requested resolution=1920x1080 is unavailable. Got 1280x720. Accepted
device/buffer_list.c: CAMERA:capture: Using: 1280x720/MJPG, buffers=3, bytesperline=0, sizeimage=1.8MiB
device/buffer_list.c: CAMERA:capture: Opened 3 buffers. Memory used: 5.3 MiB
device/device.c: CAMERA: Setting frame interval_us=0 for FPS=30
device/v4l2/device_options.c: CAMERA: The 'AfTrigger=1' was failed to find.
device/links.c: ?: Link 0: CAMERA:capture[1280x720/MJPG/3] => [SNAPSHOT-CAPTURE, STREAM-CAPTURE]
device/buffer_list.c: CAMERA:capture: Streaming started... Was 0 of 3 enqueud
```

ayufan, 2023-05-26:
Glad that I could help.

closed 2023-05-26 by ayufan