ayufan/camera-streamer #67: AeExposureMode=2 - AeExposureMode=2' was failed to find.

webgps, 2023-06-02:
Getting this error and service restart when try to set various options (not all, but too few)

Jun 02 13:21:17 wgrpinlcam01 camera-streamer[44004]: device/v4l2/device_options.c: RESCALLER:STREAM: The 'AeExposureMode=2' was failed to find.
Jun 02 13:21:17 wgrpinlcam01 camera-streamer[44004]: device/v4l2/device_options.c: RESCALLER:VIDEO: The 'AeExposureMode=2' was failed to find.
Jun 02 13:21:17 wgrpinlcam01 camera-streamer[44004]: device/v4l2/device_options.c: SNAPSHOT: The 'AeExposureMode=2' was failed to find.
Jun 02 13:21:17 wgrpinlcam01 camera-streamer[44004]: device/v4l2/device_options.c: STREAM: The 'AeExposureMode=2' was failed to find.
Jun 02 13:21:17 wgrpinlcam01 camera-streamer[44004]: device/v4l2/device_options.c: VIDEO: The 'AeExposureMode=2' was failed to find.
Jun 02 13:21:17 wgrpinlcam01 camera-streamer[44004]: util/http/http.c: HTTP8080/6: Client disconnected 192.168.2.6.
Jun 02 13:21:17 wgrpinlcam01 camera-streamer[44004]: [518:41:50.377648261] [44019] FATAL RPiAgc agc.cpp:540 No exposure profile long
Jun 02 13:21:17 wgrpinlcam01 camera-streamer[44004]: Backtrace:
Jun 02 13:21:17 wgrpinlcam01 systemd[1]: camera-streamer-raspi-v3-12MP.service: Main process exited, code=killed, status=6/ABRT
Jun 02 13:21:17 wgrpinlcam01 systemd[1]: camera-streamer-raspi-v3-12MP.service: Failed with result 'signal'.
Jun 02 13:21:17 wgrpinlcam01 systemd[1]: camera-streamer-raspi-v3-12MP.service: Consumed 1min 32.532s CPU time.
Jun 02 13:21:27 wgrpinlcam01 systemd[1]: camera-streamer-raspi-v3-12MP.service: Scheduled restart job, restart counter is at 5.

ayufan, 2023-06-02:
Unfortunately, libcamera (upstream) fails when setting some of the options. Not all are exposed. In this case it is `long` exposure. So it fails, and crashes the whole process.

webgps, 2023-06-04:
Yes, i can use bunch of the options without error, unfortunatelly i need thoose which are failing :)
Should i use libcamera-dev? Or....?

kbingham, 2023-06-05:
"Crashes the whole process" shouldn't happen. Only controls exposed by the Camera->controls() are valid to be set. Is there a specific control you are setting that is crashing on the RPi ?

ayufan, 2023-06-05:
@kbingham Thanks for looking into this :) Let me find one.

ayufan, 2023-06-05:
@kbingham If I recall correctly:

- for `dtoverlay=imx519` when setting `AfMode=` it would crash process
- this is code fetching all available controls: https://github.com/ayufan/camera-streamer/blob/develop/device/libcamera/options.cc
- whether this is develop or master, does not make difference

```
libcamera-dev is already the newest version (0~git20230302+923f5d70-1).
libcamera0 is already the newest version (0~git20230302+923f5d70-1).
```

```
device/libcamera/options.cc: CAMERA: Configuring option 'AfMode' (0000001c, type=3) = 1
util/http/http.c: HTTP8080/5: Client disconnected 192.168.4.1.
[51:25:20.179121287] [4991]  WARN IPARPI raspberrypi.cpp:801 Could not set AF_MODE - no AF algorithm

Thread 15 "camera-streamer" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fe329b600 (LWP 4991)]
0x0000007fe980fb24 in libcamera::ipa::RPi::IPARPi::queueRequest(libcamera::ControlList const&) () from /usr/lib/aarch64-linux-gnu/libcamera/ipa_rpi.so
(gdb) backtrace
#0  0x0000007fe980fb24 in libcamera::ipa::RPi::IPARPi::queueRequest(libcamera::ControlList const&) () at /usr/lib/aarch64-linux-gnu/libcamera/ipa_rpi.so
#1  0x0000007ff6346028 in libcamera::Object::message(libcamera::Message*) () at /lib/aarch64-linux-gnu/libcamera-base.so.0.0.4
#2  0x0000007ff6348430 in libcamera::Thread::dispatchMessages(libcamera::Message::Type) () at /lib/aarch64-linux-gnu/libcamera-base.so.0.0.4
#3  0x0000007ff633f804 in libcamera::EventDispatcherPoll::processEvents() () at /lib/aarch64-linux-gnu/libcamera-base.so.0.0.4
#4  0x0000007ff6348158 in libcamera::Thread::exec() () at /lib/aarch64-linux-gnu/libcamera-base.so.0.0.4
#5  0x0000007ff7e7ccac in  () at /lib/aarch64-linux-gnu/libstdc++.so.6
#6  0x0000007ff7f92648 in start_thread (arg=0x7fe329af00) at pthread_create.c:477
#7  0x0000007ff5cf2c1c in thread_start () at ../sysdeps/unix/sysv/linux/aarch64/clone.S:78
```

ayufan, 2023-06-05:
Or.:

```
device/libcamera/options.cc: CAMERA: Configuring option 'AeExposureMode' (00000005, type=3) = 2
util/http/http.c: HTTP8080/0: Client disconnected 192.168.4.1.
[51:27:05.507557502] [5477] FATAL RPiAgc agc.cpp:540 No exposure profile long
Backtrace:

Aborted
```

This fetches the available control, but in this case it seems to crash on the lack of the given enum value.

kbingham, 2023-06-07:
Just as an update - we're looking into this on the libcamera/rpi side. There's a couple of issues. The IMX519 got missed from an IPA tuning data file update, so that needs fixing (you could do this locally yourself to get past this if you look at the differences between imx477.json and imx519.json in the exposure_modes section), and maybe in that event we shouldn't be calling a fatal assertion, just disallowing that exposure profile.

@webgps Can you confirm what sensor module you are using please?

kbingham, 2023-06-07:
Oh I posted here 'minutes' too soon, as just after I sent this Naush has posted relevant patches.

Could you test the patches from https://patchwork.libcamera.org/project/libcamera/list/?series=3912 please?

kbingham, 2023-06-08:
> Thank you @kbingham. I will be able to test it next week.

Patches for this issue now merged to libcamera to https://git.libcamera.org/libcamera/libcamera.git/commit/?id=0ee9339331c648232e87d2de2ccd5a92cc61cab2

> Would it be possible for you folks to also fix the [#67 (comment)](https://github.com/ayufan/camera-streamer/issues/67#issuecomment-1577385614)? It seems to be NULL pointer access violation.

Could this be broken out to a separate issue please?