ayufan/camera-streamer #70: VERSION file not updated. Shouldn't it have 0.2.1 instead of 0.2?

joaomajesus, 2023-06-09:
Hi,

Just a minor issue I found while trying to compare the version of camera-streamer being used by my crowsnest installation and the latest version. 

The file ./VERSION still has "0.2" shouldn't id be updated to 0.2.1 to reflect the latest release?

I'm not familiar with the scheme for this file, so if that's how it's supposed to be then disregard.

ayufan, 2023-06-09:
@joaomajesus No. The `VERSION` indicates a stable major version, so currently development happens in `0.2.x`. What is released is automatically incremented by GitHub Workflow to add `0.2.N` and push associated tag: https://github.com/ayufan/camera-streamer/blob/master/.github/workflows/build_release.yaml#L39.

We will be moving `crowsnest` to use prebuilts that are created with GitHub CI.

joaomajesus, 2023-06-09:
Thanks for clarifying! Sorry to bother! Closing.

closed 2023-06-09 by joaomajesus