ayufan/camera-streamer #85: "ioctl(ret=-1, errno=22): Can't queue buffer." with v4l2 camera interface

CelloVerb, 2023-07-07:
Looking for help and suggestions:

libcamera streaming works properly with camera-streamer, but v4l2 doesn't, and gives "ioctl(ret=-1, errno=22): Can't queue buffer." when trying to start capture.     Am wanting that very low latency with v4l2.   I don't see any errors on stdout with v4l2 startup.

Hardware: raspberry pi zero W
System: Raspbian GNU/Linux 11 (bullseye) / 11.6
Camera: Arducam imx219 8MP

Any suggestions for what would cause this, or other places to look? 

Thanks!
Cello

stdout output:
pi@tarmocam:~/camera-streamer $ ./camera-streamer --http-listen=0.0.0.0 
./camera-streamer Version: v0.2.5 (98aa208)
util/http/http.c: ?: HTTP listening on 0.0.0.0:8080.
device/v4l2/device.c: CAMERA: Device path=/dev/video0 fd=14 opened
device/v4l2/device_options.c: CAMERA: Configuring option 'Horizontal Flip' (00980914) = 0
device/v4l2/device_options.c: CAMERA: Configuring option 'Vertical Flip' (00980915) = 0
device/buffer_list.c: CAMERA:capture: Using: 1920x1080/YUYV, buffers=3, bytesperline=3840, sizeimage=4.0MiB
device/buffer_list.c: CAMERA:capture: Opened 3 buffers. Memory used: 11.9 MiB
device/v4l2/device.c: SNAPSHOT: Device path=/dev/video31 fd=19 opened
device/v4l2/buffer_list.c: SNAPSHOT:output:mplane: Requested resolution=1920x1080 is unavailable. Got 1920x1088.
device/buffer_list.c: SNAPSHOT:output:mplane: Using: 1920x1056/YUYV, buffers=3, bytesperline=3840, sizeimage=3.9MiB
device/buffer_list.c: SNAPSHOT:output:mplane: Opened 3 buffers. Memory used: 0.0 MiB
device/buffer_list.c: SNAPSHOT:capture:mplane: Using: 1920x1056/JPEG, buffers=3, bytesperline=0, sizeimage=4.0MiB
device/buffer_list.c: SNAPSHOT:capture:mplane: Opened 3 buffers. Memory used: 12.0 MiB
device/v4l2/device.c: VIDEO: Device path=/dev/video11 fd=23 opened
device/buffer_list.c: VIDEO:output:mplane: Using: 1920x1080/YUYV, buffers=3, bytesperline=3840, sizeimage=4.0MiB
device/buffer_list.c: VIDEO:output:mplane: Opened 3 buffers. Memory used: 0.0 MiB
device/buffer_list.c: VIDEO:capture:mplane: Using: 1920x1080/H264, buffers=3, bytesperline=0, sizeimage=0.8MiB
device/buffer_list.c: VIDEO:capture:mplane: Opened 3 buffers. Memory used: 2.2 MiB
device/v4l2/device.c: CAMERA: ioctl(ret=-1, errno=25): Can't set FPS
device/device.c: CAMERA: Setting frame interval_us=33333 for FPS=30
device/v4l2/device_options.c: CAMERA: The 'AfTrigger=1' was failed to find.
device/v4l2/device_options.c: SNAPSHOT: Configuring option 'Compression Quality' (009d0903) = 80
device/v4l2/device_options.c: VIDEO: Configuring option 'Repeat Sequence Header' (009909e2) = 1
device/v4l2/device_options.c: VIDEO: Configuring option 'Video Bitrate Mode' (009909ce) = 0
device/v4l2/device_options.c: VIDEO: Configuring option 'Video Bitrate' (009909cf) = 2000000
device/v4l2/device_options.c: VIDEO: Configuring option 'Repeat Sequence Header' (009909e2) = 5000000
device/v4l2/device_options.c: VIDEO: Configuring option 'H264 I-Frame Period' (00990a66) = 30
device/v4l2/device_options.c: VIDEO: Configuring option 'H264 Level' (00990a67) = 11
device/v4l2/device_options.c: VIDEO: Configuring option 'H264 Profile' (00990a6b) = 4
device/v4l2/device_options.c: VIDEO: Configuring option 'H264 Minimum QP Value' (00990a61) = 16
device/v4l2/device_options.c: VIDEO: Configuring option 'H264 Maximum QP Value' (00990a62) = 32
device/links.c: ?: Link 0: CAMERA:capture[1920x1080/YUYV/3] => [SNAPSHOT:output:mplane[1920x1056/YUYV/3], VIDEO:output:mplane[1920x1080/YUYV/3]]
device/links.c: ?: Link 1: SNAPSHOT:capture:mplane[1920x1056/JPEG/3] => [SNAPSHOT-CAPTURE, STREAM-CAPTURE]
device/links.c: ?: Link 2: VIDEO:capture:mplane[1920x1080/H264/3] => [VIDEO-CAPTURE]
device/buffer_list.c: CAMERA:capture: Streaming started... Was 0 of 3 enqueud
device/buffer_list.c: SNAPSHOT:output:mplane: Streaming started... Was 0 of 3 enqueud
device/buffer_list.c: VIDEO:output:mplane: Streaming started... Was 0 of 3 enqueud
device/buffer_list.c: SNAPSHOT:capture:mplane: Streaming started... Was 0 of 3 enqueud
device/buffer_list.c: VIDEO:capture:mplane: Streaming started... Was 0 of 3 enqueud
util/http/http.c: HTTP8080/1: Client connected 192.168.86.21 (fd=5).
util/http/http.c: HTTP8080/1: Request 'GET' '/' ''
util/http/http.c: HTTP8080/0: Client connected 192.168.86.21 (fd=4).
util/http/http.c: HTTP8080/1: Client disconnected 192.168.86.21.
util/http/http.c: HTTP8080/0: Client disconnected 192.168.86.21.
util/http/http.c: HTTP8080/2: Client connected 192.168.86.21 (fd=6).
util/http/http.c: HTTP8080/2: Request 'GET' '/snapshot' ''
util/http/http.c: HTTP8080/3: Client connected 192.168.86.21 (fd=7).
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
util/http/http.c: HTTP8080/2: Client disconnected 192.168.86.21.
util/http/http.c: HTTP8080/3: Client disconnected 192.168.86.21.
util/http/http.c: HTTP8080/4: Client connected 192.168.86.21 (fd=8).
util/http/http.c: HTTP8080/5: Client connected 192.168.86.21 (fd=9).
util/http/http.c: HTTP8080/4: Client disconnected 192.168.86.21.
util/http/http.c: HTTP8080/5: Client disconnected 192.168.86.21.

CelloVerb, 2023-07-07:
OK, I've dug further and realized that disabling the media-controller interface in /boot/config.txt fixes this - changing dtoverlay=imx219 to

- dtoverlay=imx219,media-controller=0

But now I can't use libcamera apps - is there a way to have both?

YaphetS1, 2023-11-10:
same issue while use camera-streamer in Docker container... But I don't know how can we fix /boot/config.txt in any image? I'd try to use that images: `balenalib/rpi-raspbian:latest`, `balenalib/raspberrypi4-64-debian`

kbingham, 2023-11-13:
It's not possible to have both the legacy camera stack and the libcamera stack configured at the same time.
It would probably be more beneficial to look at how to lower the latency with libcamera.

I could envisage a low latency mode in libcamera being implemented.
I think at the moment libcamera on RPi prefers extra buffering to ensure there is no loss of data over a leaky but shorter pipe. 

Ultimately that's a design choice by Raspberry Pi in the RPi pipeline handler presently, but could be configurable.