ayufan/camera-streamer #94: Stale Streams detected but not fixed

JoelBecker1998, 2023-08-13:
Sometimes, after a while streams just die. WebRTC shows nothing, mjpg streams throws a server error.
This happens if the camera is left running for a while, sometimes an hour, sometimes a day.

Anyway, I think the software detects a stale stream and tries to restart it but it still doesn't work. Restarting the entire service works just fine. Any idea what could cause this?

Here is the log:

``` 
ug 13 15:41:10 armcam0 bash[601]: device/buffer_list.c: CAMERA:capture: Streaming started... Was 0 of 4 enqueud
Aug 13 15:41:10 armcam0 bash[601]: device/buffer_list.c: SNAPSHOT:output:mplane: Streaming started... Was 0 of 4 enqueud
Aug 13 15:41:10 armcam0 bash[601]: device/buffer_list.c: VIDEO:output:mplane: Streaming started... Was 0 of 4 enqueud
Aug 13 15:41:10 armcam0 bash[601]: device/buffer_list.c: SNAPSHOT:capture:mplane: Streaming started... Was 0 of 4 enqueud
Aug 13 15:41:10 armcam0 bash[601]: device/buffer_list.c: VIDEO:capture:mplane: Streaming started... Was 0 of 4 enqueud
Aug 13 19:12:11 armcam0 bash[601]: util/http/http.c: HTTP8081/1: Client connected 172.16.10.4 (fd=7).
Aug 13 19:12:11 armcam0 bash[601]: util/http/http.c: HTTP8081/1: Request 'GET' '/webrtc' ''
Aug 13 19:12:11 armcam0 bash[601]: util/http/http.c: HTTP8081/1: Client disconnected 172.16.10.4.
Aug 13 19:12:11 armcam0 bash[601]: util/http/http.c: HTTP8081/2: Client connected 172.16.10.4 (fd=8).
Aug 13 19:12:11 armcam0 bash[601]: util/http/http.c: HTTP8081/2: Request 'POST' '/webrtc' ''
Aug 13 19:12:11 armcam0 bash[601]: output/webrtc/webrtc.cc: rtc-mborhwnxgtxllrbydrka: Stream requested.
Aug 13 19:12:15 armcam0 bash[601]: util/http/http.c: HTTP8081/2: Client disconnected 172.16.10.4.
Aug 13 19:12:15 armcam0 bash[601]: util/http/http.c: HTTP8081/0: Client connected 172.16.10.4 (fd=9).
Aug 13 19:12:15 armcam0 bash[601]: util/http/http.c: HTTP8081/0: Request 'POST' '/webrtc' ''
Aug 13 19:12:15 armcam0 bash[601]: output/webrtc/webrtc.cc: rtc-mborhwnxgtxllrbydrka: Answer received.
Aug 13 19:12:15 armcam0 bash[601]: util/http/http.c: HTTP8081/0: Client disconnected 172.16.10.4.
Aug 13 19:12:15 armcam0 bash[601]: device/links.c: CAMERA:capture: Stale detected. Restarting streaming...
Aug 13 19:12:15 armcam0 bash[601]: device/buffer_list.c: CAMERA:capture: Streaming stopped... Was 0 of 4 enqueud
Aug 13 19:12:15 armcam0 bash[601]: device/buffer_list.c: CAMERA:capture: Streaming started... Was 0 of 4 enqueud
```

foosel, 2023-10-02:
We are seeing a lot of those as well on the camera-streamer enabled OctoPi image. This is currently a blocker for general roll-out.

ayufan, 2023-10-02:
> Exactly. I temporarily fixed it by just adding an "exit(0)" where the log
> message "detected stale streams" is logged and just auto restart the
> service. However, i believe the issue also occures at different places, as
> this does not fix all issues.
> With MJPG or H264 you at least get a "server error message" when you
> reload the screen. With webRTC there is no error, no image nothing.

Let me take a look at this in more detail.

ayufan, 2023-10-02:
Let me take a look at this again. Previously when I checked this:

- `Stale detected. Restarting streaming` is an outcome of jpeg decoder that received a broken jpeg frame
- The broken JPEG frame can be received from USB bus, it appears that this occasionally happens on RPI when the USB bus is underrun?

JoelBecker1998, 2023-10-02:
Bit confused about that. 
Idk where or why I would receive a broken JPEG frame over USB.
My setup is just a CSI camera and i use WebRTC.
If I remember correctly, Stale detected occurs if the timestamp of the frame is too old no?

foosel, 2023-10-02:
Too add some more data points to this, we've also gotten reports about this from CSI cameras (primarily RPi Cam 3, but also others). So I don't think it's just USB stuff. 

In any case, it would be important that this gets handled gracefully some way or another, either by exiting the service altogether since it is unusable when it happens right now (in which case it can be restarted automatically by systemd), or handling it internally by restarting whatever needs to get restarted in the camera pipeline. As things are at the moment, a manual restart is required to fix things, and for a headless system that's an big UX issue sadly.

ayufan, 2023-10-02:
@foosel So, restarting service fixes the issue? Interesting. Do you have maybe `dmesg` from the time it happens?

foosel, 2023-10-02:
Currently not, but I can see that I get one. It's a bit tricky since I haven't found yet a way to trigger it when I want, I just saw it happen myself, and I've gotten (`journalctl`) logs from users for whom it happened. I'll keep an eye out and will also see if I can trigger it here.

ayufan, 2023-10-02:
You can also send my way `journalctl` as it should contain `dmesg` output as well.

foosel, 2023-10-02:
Successful repro. Left the streamer alone for an hour as suggested in #90, tried to access the mjpg stream, "stale" message in log and just a server error in the browser.

<details>
<summary>dmesg</summary>

```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 6.1.21-v7+ (dom@buildbot) (arm-linux-gnueabihf-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1642 SMP Mon Apr  3 17:20:52 BST 2023
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 3 Model B Rev 1.2
[    0.000000] random: crng init done
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Reserved memory: created CMA memory pool at 0x1ec00000, size 256 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x0000000037ffffff]
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000037ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000037ffffff]
[    0.000000] percpu: Embedded 17 pages/cpu s36884 r8192 d24556 u69632
[    0.000000] pcpu-alloc: s36884 r8192 d24556 u69632 alloc=17*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 227360
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_headphones=0 snd_bcm2835.enable_headphones=1 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_hdmi=0 video=Composite-1:720x480@60i vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=69498b49-02 rootfstype=ext4 fsck.repair=yes rootwait
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 628144K/917504K available (10240K kernel code, 1452K rwdata, 2900K rodata, 1024K init, 613K bss, 27216K reserved, 262144K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 34800 entries in 103 pages
[    0.000000] ftrace: allocated 102 pages with 4 groups
[    0.000000] trace event string verifier disabled
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] 	Rude variant of Tasks RCU enabled.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000001] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000018] Switching to timer-based delay loop, resolution 52ns
[    0.000389] Console: colour dummy device 80x30
[    0.001010] printk: console [tty1] enabled
[    0.001074] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.001120] pid_max: default: 32768 minimum: 301
[    0.001283] LSM: Security Framework initializing
[    0.001507] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.001551] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.002735] cgroup: Disabling memory control group subsystem
[    0.002988] CPU: Testing write buffer coherency: ok
[    0.003426] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.004288] cblist_init_generic: Setting adjustable number of callback queues.
[    0.004327] cblist_init_generic: Setting shift to 2 and lim to 1.
[    0.004455] cblist_init_generic: Setting shift to 2 and lim to 1.
[    0.004615] Setting up static identity map for 0x100000 - 0x10003c
[    0.004772] rcu: Hierarchical SRCU implementation.
[    0.004796] rcu: 	Max phase no-delay instances is 1000.
[    0.005526] smp: Bringing up secondary CPUs ...
[    0.006456] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.007415] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.008291] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.008392] smp: Brought up 1 node, 4 CPUs
[    0.008473] SMP: Total of 4 processors activated (153.60 BogoMIPS).
[    0.008500] CPU: All CPU(s) started in HYP mode.
[    0.008520] CPU: Virtualization extensions available.
[... 397 lines cut ...]
Sep 22 01:24:10 octopib sh[775]: [0:00:22.974276865] [894]  WARN IPARPI raspberrypi.cpp:1183 Could not set AF_TRIGGER - no AF algorithm or not Auto
Sep 22 01:24:10 octopib sh[775]: device/buffer_list.c: CAMERA:capture: Streaming started... Was 0 of 2 enqueud
Sep 22 01:24:10 octopib sh[775]: device/buffer_list.c: SNAPSHOT:output:mplane: Streaming started... Was 0 of 2 enqueud
Sep 22 01:24:10 octopib sh[775]: device/buffer_list.c: RESCALLER:STREAM:output:mplane: Streaming started... Was 0 of 2 enqueud
[... cut ...]
Oct 02 16:36:27 octopib sh[775]: device/buffer_list.c: CAMERA:capture: Streaming stopped... Was 0 of 2 enqueud
Oct 02 16:36:27 octopib sh[775]: device/buffer_list.c: CAMERA:capture: Streaming started... Was 0 of 2 enqueud
Oct 02 16:36:29 octopib sh[775]: util/http/http.c: HTTP8080/1: Client disconnected 127.0.0.1.
```
</details>

Note how the stale message in the journalctl output only gets logged once I tried to access the mjpg stream in the browser. The server is now dysfunctional, neither of the streams nor the snapshot work.

After a server restart however, everything works again.

warshoot3r, 2023-10-26:
I hope this can be of use. I have multiple raspberry pi's( pi4, pizero)
- pi4 is wired into ethernet and never gets this problem

Now the wifi ones.
- one of my pizero never gets this problem after I adjusted wifi power.
- another pizero will get the issue but its in a different location.

But what I think might be causing the issue is that :
1. camera streaming server is running.
2. client is connected
3. the camera server  roams to another wifi network/ disconnects
4. server is not aware and this could cause the server error(?) as stream socket might not be handled?
5. You must restart the server or the camera streamer service.

YaphetS1, 2023-12-11:
@foosel Hi!
Did you find to solve issue?
I'd the same problem on my Pi4 with camera OV5647

recrudesce, 2023-12-27:
I have this same issue.  Streams just stop working after a while.  Wondering if there's a fix ?

jeloneal, 2024-01-31:
You could just restart the service via cron on a regular interval like every few hours. That could at least serve as a workaround until it is fixed.

DanielJoyce, 2024-07-26:
After digging around, apparently H264 encoding 1280p can cause voltage dips? On the kodi forums saw a reccomendation to add over_voltage=2 to the boot/config.txt file, and maybe just drop the res of the pi-cam to 720p, or the framerate to 15fps.

I have made these changes on my mainsail system and will update if things stay stable

DanielJoyce, 2024-07-26:
I have noticed that once my Rpi 4 with klipper starts printing, the camera invariably stops working.

DanielJoyce, 2024-07-27:
> After digging around, apparently H264 encoding 1280p can cause voltage dips? On the kodi forums saw a reccomendation to add over_voltage=2 to the boot/config.txt file, and maybe just drop the res of [...]

No crashes of the webcam after these changes so far.

DanielJoyce, 2024-09-04:
@foosel Please check out my comments. The tweaks here have fixed the camera issue for me.

foosel, 2024-09-06:
Saw the comments, that isn't a solution for the issue however that will work for the majority I fear.

hsac, 2025-01-19:
made the changes suggested by @DanielJoyce which unfortunately didn't work, same behavior as before the changes the cam using webrtc(camera-streamer) crashes and eventually the pi crashes
`kernel:[  728.706730] Internal error: Oops: 0000000096000005 [#1] PREEMPT SMP`
`kernel:[  728.707562] Code: d65f03c0 d2800001 52800022 f9800271 (885ffe60)`

mjepg functions okay

DanielJoyce, 2025-01-19:
The camera still sometimes comes and goes for me. But not as often and it sometimes seems to recover.