ayufan/camera-streamer #95: Full Frame capture with PiCam2

msroest, 2023-08-14:
I'm attempting to move my octoprint to the new version using camera-streamer and I've come across a very strange issue. When trying to get full frame capture from my PiCam2.  I've setup the my libcamera.conf as follows based on this issue https://github.com/ayufan/camera-streamer/issues/3
```
### Options for libcamera based cameras (PiCam, Arducam, ...)

# The port on which the webcam server for the camera should listen on. If you have only
# one camera, leave at 8080. If you have more, change to 8081, 8082, etc. The primary
# camera will be considered the one with 8080.
PORT=8080

# The resolution to request on the camera sensor. Defaults to 1280x720.
WIDTH=3280
HEIGHT=2464

# The height to use for the video stream. Defaults to 720.
VIDEO_HEIGHT=720

# The height to use for the snapshots. Defaults to 1080.
SNAPSHOT_HEIGHT=1080

# The framerate to set on the camera. Defaults to 15fps.
FRAMERATE=15

# Additional options. By default enables continuous auto focus (if possible).
#OPTIONS='--camera-options="AfMode=2" --camera-options="AfRange=2"'
OPTIONS='-camera-nbufs=2 --camera-high_res_factor=2 --camera-low_res_factor=4'
```
In this mode the camera-streamer returns a half a frame with the rest being green
[image]

If I update the config to use 1920x1080 capture the streamer works fine but the view is garbage cause of the crop.

If I stop camera-streamer and use `libcamera-jpeg -o image.jpg --nopreview` it correctly pulls the whole frame:
[image]

msroest, 2023-08-18:
@paddymck you are my saviour.  This fixed my issue, my full frame streaming now works

closed 2023-08-18 by msroest

drf5n, 2023-10-06:
Where does the libcamera.conf file go?

paddymck, 2023-10-06:
My libcamera.conf is in '/boot/camera-streamer/ ' and seems to work OK there with v0.2.5.
If that location does not work for you, please say.
Enjoy.
:slightly_smiling_face:

drf5n, 2023-10-07:
Thanks.  I wasn't using the Octopi distribution and wanted to see how the information got from `/boot/camera-streamer/...` into the camera-streamer systemd services. 

Turns out the distribution installs camera-streamer, writes its own services, and then moves the config files to /boot and symlinks them:

https://community.octoprint.org/t/camera-streamer-configuration-on-the-new-camera-stack-for-octopi/49950/45?u=drf5n

If you aren't using the OctoPi distribution image, those systemd files are missing, and the ones you get from here don't know to parse the `/boot/camera_streamer/*` environment variable files into the appropriate camara-streamer CLI options.