ayufan/camera-streamer #99: Can't pass a tuning-file as a parameter

sosagarcia, 2023-08-25:
Hello!,

I use your camera-streamer service in a fluidd web interface to stream what is happening while I'm 3D printing.

I noticed that is not possible to pass a tuning-file as a parameter ( which is something you can do with libcamera )
For example  executing:
`libcamera-still --tuning-file /usr/share/libcamera/ipa/rpi/vc4/ov5647_noir.json -o image.jpg`

As I have a NOIR camera in my Raspberry it would be perfect if I could apply to the streaming a tuning file to see a full-color streaming 

For example, the imagen2.jpg is how the camera captures the image in infrared. In image.jpg is with a -tuning-file="/usr/share/libcamera/ipa/raspberrypi/ov5647_noir.json"
[image]
[image]

The closest thing I could get is when I set `--camera-options=AwbEnable=true` ( see image3.jpg) 
[image]

Do you know if there is a solution for that?
In the past, I could manage it by putting 
`awb_auto_is_greyworld=1` in /boot/config.txt in my raspberry pi. But seems is not working anymore...

I would appreciate any help :) thanks

albatorsssx, 2023-11-29:
Hi,
No one can answer?
I am also interested in my configuration under Octopi.
THANKS

ayufan, 2023-11-29:
Oh. Good hack. I guess, this is best solution then.

Delivator, 2024-02-11:
I set the environment variable but it seems to just be ignored. Using `libcamera-still --tuning-file` also works fine for me.