ayufan/camera-streamer #100: Support 90º degree rotation

calexandre, 2023-08-28:
As the topic states, I would like to request support for 90º degree rotation for the `camera-streamer`.

There's a [topic already opened](https://community.octoprint.org/t/camera-streamer-rotating-90-degree-how/53591) in the OctoPrint forums, but eventually it led us here.

According to libcamera.org, the [library apparently supports the transform function](https://libcamera.org/api-html/namespacelibcamera.html#a371b6d17d531b85c035c4e889b116571) which can be used to apply a 90º degree rotation.

kbingham, 2023-08-29:
libcamera supports a 90* rotation - but the RPi ISP hardware doesn't - so the rotation must be handled by some later layer. That could be the browser perhaps in this instance if we can report the rotation desired?

bra1nDeaD, 2024-02-15:
+1

I've been searching blind to see how I can rotate 90 degrees CCW. I found that the camera has "properties" that show it can have a value for "Rotation", but I have not yet found a way set this with camera-streamer:
`
CAMERA Properties:
- property: SensorSensitivity (00000009, type=Float) = 1.000000
- property: SystemDevices (0000000a, type=Integer64) = [ 20750, 20738, 20739, 20740 ]
- property: ScalerCropMaximum (00000008, type=Rectangle) = (348, 434)/1928x1080
- property: ColorFilterArrangement (0000000b, type=Integer32) = 2
- property: PixelArrayActiveAreas (00000007, type=Rectangle) = [ (16, 6)/2592x1944 ]
- property: PixelArraySize (00000005, type=Size) = 2592x1944
- property: Rotation (00000002, type=Integer32) = 0
- property: Location (00000001, type=Integer32) = 2
- property: UnitCellSize (00000004, type=Size) = 1400x1400
- property: Model (00000003, type=String) = ov5647
`

So far I've only seen how the "options" can be changed. What is the fundamental difference between "options" and "properties" ? Would it be possible to change properties?

jerryfudd, 2025-04-24:
trying to find out the answer to this also