ayufan/camera-streamer #103: camera-streamer breaks udev monitoring

Thugmek, 2023-09-08:
I have a weird error with `camera-streamer` on Raspberry Pi with my custom-built Yocto image. When I stream `ov5647` camera with `camera-streamer --camera-path=/base/soc/i2c0mux/i2c@1/ov5647@36 --camera-type=libcamera --camera-format=YUYV --camera-width=1296 --camera-height=972 --http-port=8081`, it completely breaks udev monitoring. No events are captured by `udevadm monitor --udev`, no new devices are listed in `/dev`. Once I terminate streaming process, udev starts to work again (but actions performed during camera streaming, eg. USB inserts, are lost).
If I stream USB camera with camera-streamer `--camera-path=/dev/v4l/by-id/usb-046d_C270_HD_WEBCAM_200901010001-video-index0 --camera-type=v4l2 --camera-format=MJPEG --camera-width=1280 --camera-height=720 --http-port=8081`, everything works fine. If I create a service with the same command, I use to stream `ov5674` camera, udev monitor works fine. But when I make python script to monitor udev`and switch cameras, udev monitor doesn't work inside this service and I won't get the actual filesystem state, when I run `ls /dev` inside the service.
It seems that the camera streamer breaks udev in some local context. I've been stuck on this for a few days and I'm not able to investigate, what's going on here. Do you have any ideas?

kbingham, 2023-09-11:
This is interesting. libcamera uses udev internally to monitor for events (i.e. camera hotplug events) ... but I wouldn't have expected that to block the main udev monitor.

Might be worth looking through https://git.libcamera.org/libcamera/libcamera.git/tree/src/libcamera/device_enumerator_udev.cpp and the api's used to see if there are any suspicious calls there that are documented to block the system. If so we'd have to look deeper into how we manage it.

Thugmek, 2023-09-12:
By further investigation, I have found this error log in journalctl:
```Sep 12 14:03:14 MichalYocto kernel: usb 1-1.1: new high-speed USB device number 19 using xhci_hcd
Sep 12 14:03:14 MichalYocto kernel: xhci_hcd 0000:01:00.0: ERROR: unexpected setup context command completion code 0x11.
Sep 12 14:03:14 MichalYocto kernel: usb 1-1.1: hub failed to enable device, error -22
Sep 12 14:03:14 MichalYocto kernel: usb 1-1.1: new high-speed USB device number 20 using xhci_hcd
Sep 12 14:03:14 MichalYocto kernel: xhci_hcd 0000:01:00.0: ERROR: unexpected setup context command completion code 0x11.
Sep 12 14:03:14 MichalYocto kernel: usb 1-1.1: hub failed to enable device, error -22
Sep 12 14:03:14 MichalYocto kernel: usb 1-1-port1: attempt power cycle
Sep 12 14:03:15 MichalYocto kernel: usb 1-1.1: new high-speed USB device number 21 using xhci_hcd
Sep 12 14:03:15 MichalYocto kernel: xhci_hcd 0000:01:00.0: ERROR: unexpected setup address command completion code 0x11.
Sep 12 14:03:15 MichalYocto kernel: xhci_hcd 0000:01:00.0: ERROR: unexpected setup address command completion code 0x11.
Sep 12 14:03:15 MichalYocto kernel: usb 1-1.1: device not accepting address 21, error -22
Sep 12 14:03:15 MichalYocto kernel: usb 1-1.1: new high-speed USB device number 22 using xhci_hcd
Sep 12 14:03:15 MichalYocto kernel: xhci_hcd 0000:01:00.0: ERROR: unexpected setup address command completion code 0x11.
Sep 12 14:03:16 MichalYocto kernel: xhci_hcd 0000:01:00.0: ERROR: unexpected setup address command completion code 0x11.
Sep 12 14:03:16 MichalYocto kernel: usb 1-1.1: device not accepting address 22, error -22
Sep 12 14:03:16 MichalYocto kernel: usb 1-1-port1: unable to enumerate USB device
```

It happens every time I plug in something to USB (Flashdisk, USB webcam, wireless mouse dongle).