ayufan/camera-streamer #105: tools/libcamera_camera.sh returns with libcamera is not supported

Dephrilibrium, 2023-10-01:
Hello,

I'm currently searching for **any** working solution which makes a http-camera-stream based on libcamera available to use it by octoprint/haproxy.
camera-streamer seems for now to be the best solution as it is at least compilable.
I checked the docs a bit out, and eded up that there is an easy to use libcamera-script within the tools folder, but when I run it, the script returns:
```
~/camera-streamer $ ./tools/libcamera_camera.sh 
++ nproc
+ make -j4
echo "#define GIT_VERSION \"v0.2.6\"\n#define GIT_REVISION \"ff539cd\"" > version.h.tmp; \
        diff -u version.h version.h.tmp || mv version.h.tmp version.h; \
        rm -f version.h.tmp
make camera-streamer
make[1]: Entering directory '/home/haum/camera-streamer'
make[1]: 'camera-streamer' is up to date.
make[1]: Leaving directory '/home/haum/camera-streamer'
+ ./camera-streamer --http-listen=0.0.0.0 --http-port=8080 --camera-type=libcamera --camera-format=YUYV
./camera-streamer Version: v0.2.6 (ff539cd)
util/http/http.c: ?: HTTP listening on 0.0.0.0:8080.
device/libcamera/libcamera.cc: ?: libcamera is not supported
```
Since the output does not tell my actually what's exactly the problem, I hope someone here does.

Kind regards,
Deph

JoelBecker1998, 2023-10-06:
You have Bullseye correct?
can you try `libcamera-hello` and post the output?
did you compile manually and did you do all of the bellow?
```
git clone https://github.com/ayufan-research/camera-streamer.git --recursive
apt-get -y install libavformat-dev libavutil-dev libavcodec-dev libcamera-dev liblivemedia-dev v4l-utils pkg-config xxd build-essential cmake libssl-dev

cd camera-streamer/
make
sudo make install
```

Dephrilibrium, 2023-10-06:
Yes, all libcamera-tools work as estimated. But nevermind, until there is any working solution for the libcam-stack I use the sloppy way and enabled the legacy camera-stack to use again mjpg-streamer.

But thanks for your help.

closed 2023-10-06 by Dephrilibrium

mryel00, 2023-10-19:
@Dephrilibrium I just came around the same issue as you. It turned out I used the wrong precompiled version. That error appears as soon as you are using a generic build instead of a raspi build.

> But nevermind, until there is any working project for the libcam-stack

I hate to post other projects inside a foreign repo but there are other working options out there like [ustreamer with libcamerify](https://github.com/pikvm/ustreamer#raspberry-pi-camera-example) (`libcamerify` is located inside `libcamera-tools`) and [spyglass](https://github.com/roamingthings/spyglass) (still missing camera controls and highly focused on the module v3).

But camera-streamer is definitely the best option I currently know, if you want a powerhouse with a lot of features. Also it was one of the first, if not even the first one inside this field.

Delivator, 2024-02-11:
@mryel00 thank you!! looks like the little install script from the release page installed the generic version instead of the raspi version.