ayufan/camera-streamer #106: camera-streamer stops streaming after a half picture

Loevion, 2023-10-06:
Hello
I have installed camera-streamer on a new Raspery Bullseye and am using an 8MP camera. It also works almost the http server is reachable. If I look at a stream in the browser, it lasts exactly half an image, the rest is colored green (picture_1(i dont use an ir filter)) and the logs always show: device/links.c: RESCALLER:VIDEO:capture:mplane:buf0: Capture image is outdated. Skipped. Now: 9928386556, vs 9927338788.please help me
LG Loevion

picture_1
[image]

terminal output:
admin@roboraspi:/usr/bin $ ./camera-streamer --camera-path=/base/soc/i2c0mux/i2c@1/imx219@10 --camera-type=libcamera --camera-format=YUYV --camera-width=3280 --camera-height=2464 --camera-fps=30 --camera-nbufs=2 --camera-snapshot.height=1080 --camera-video.height=720 --camera-stream.height=480 --camera-options=brightness=0.1 --http-listen=0.0.0.0 --http-port=8080 --rtsp-port
./camera-streamer Version: 0.2.6 (ff539cd)
util/http/http.c: ?: HTTP listening on 0.0.0.0:8080.
output/rtsp/rtsp.cc: ?: Running RTSP server on '8554'
[2:44:13.451387450] [1414]  INFO Camera camera_manager.cpp:297 libcamera v0.0.5+83-bde9b04f
[2:44:13.502178638] [1426]  WARN RPI vc4.cpp:383 Mismatch between Unicam and CamHelper for embedded data usage!
[2:44:13.503246707] [1426]  INFO RPI vc4.cpp:437 Registered camera /base/soc/i2c0mux/i2c@1/imx219@10 to Unicam device /dev/media3 and ISP device /dev/media0
device/libcamera/device.cc: CAMERA: Device path=/base/soc/i2c0mux/i2c@1/imx219@10 opened
[2:44:13.504331963] [1414]  INFO Camera camera.cpp:1033 configuring streams: (0) 1437x1080-YUYV
[2:44:13.505074564] [1426]  INFO RPI vc4.cpp:565 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected sensor format: 1640x1232-SBGGR10_1X10 - Selected unicam format: 1640x1232-pBAA
[2:44:13.507966530] [1414]  INFO Camera camera.cpp:1033 configuring streams: (0) 1437x1080-YUYV (1) 3280x2464-SBGGR10_CSI2P
[2:44:13.508676840] [1426]  INFO RPI vc4.cpp:565 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected sensor format: 3280x2464-SBGGR10_1X10 - Selected unicam format: 3280x2464-pBAA
device/buffer_list.c: CAMERA:capture: Using: 1437x1080/YUYV, buffers=2, bytesperline=2880, sizeimage=0.0MiB
device/buffer_list.c: CAMERA:capture: Opened 2 buffers. Memory used: 5.9 MiB
device/buffer_list.c: CAMERA:capture:1: Using: 3280x2464/BG10, buffers=2, bytesperline=4128, sizeimage=0.0MiB
device/buffer_list.c: CAMERA:capture:1: Opened 2 buffers. Memory used: 19.4 MiB
device/v4l2/device.c: SNAPSHOT: Device path=/dev/video31 fd=38 opened
device/v4l2/buffer_list.c: SNAPSHOT:output:mplane: Requested resolution=1437x1080 is unavailable. Got 1437x1088.
device/buffer_list.c: SNAPSHOT:output:mplane: Using: 1408x1056/YUYV, buffers=2, bytesperline=2880, sizeimage=2.9MiB
device/buffer_list.c: SNAPSHOT:output:mplane: Opened 2 buffers. Memory used: 0.0 MiB
device/buffer_list.c: SNAPSHOT:capture:mplane: Using: 1408x1056/JPEG, buffers=2, bytesperline=0, sizeimage=4.0MiB
device/buffer_list.c: SNAPSHOT:capture:mplane: Opened 2 buffers. Memory used: 8.0 MiB
device/v4l2/device.c: RESCALLER:STREAM: Device path=/dev/video12 fd=41 opened
device/buffer_list.c: RESCALLER:STREAM:output:mplane: Using: 1437x1080/YUYV, buffers=2, bytesperline=2880, sizeimage=3.0MiB
device/buffer_list.c: RESCALLER:STREAM:output:mplane: Opened 2 buffers. Memory used: 0.0 MiB
device/buffer_list.c: RESCALLER:STREAM:capture:mplane: Using: 640x480/YUYV, buffers=2, bytesperline=1280, sizeimage=0.6MiB
device/buffer_list.c: RESCALLER:STREAM:capture:mplane: Opened 2 buffers. Memory used: 1.2 MiB
device/v4l2/device.c: STREAM: Device path=/dev/video31 fd=44 opened
device/buffer_list.c: STREAM:output:mplane: Using: 640x480/YUYV, buffers=2, bytesperline=1280, sizeimage=0.6MiB
device/buffer_list.c: STREAM:output:mplane: Opened 2 buffers. Memory used: 0.0 MiB
device/buffer_list.c: STREAM:capture:mplane: Using: 640x480/JPEG, buffers=2, bytesperline=0, sizeimage=4.0MiB
device/buffer_list.c: STREAM:capture:mplane: Opened 2 buffers. Memory used: 8.0 MiB
device/v4l2/device.c: RESCALLER:VIDEO: Device path=/dev/video12 fd=47 opened
device/buffer_list.c: RESCALLER:VIDEO:output:mplane: Using: 1437x1080/YUYV, buffers=2, bytesperline=2880, sizeimage=3.0MiB
device/buffer_list.c: RESCALLER:VIDEO:output:mplane: Opened 2 buffers. Memory used: 0.0 MiB
device/buffer_list.c: RESCALLER:VIDEO:capture:mplane: Using: 992x736/YUYV, buffers=2, bytesperline=1984, sizeimage=1.4MiB
device/buffer_list.c: RESCALLER:VIDEO:capture:mplane: Opened 2 buffers. Memory used: 2.8 MiB
device/v4l2/device.c: VIDEO: Device path=/dev/video11 fd=50 opened
device/buffer_list.c: VIDEO:output:mplane: Using: 992x736/YUYV, buffers=2, bytesperline=1984, sizeimage=1.4MiB
device/buffer_list.c: VIDEO:output:mplane: Opened 2 buffers. Memory used: 0.0 MiB
device/buffer_list.c: VIDEO:capture:mplane: Using: 992x736/H264, buffers=2, bytesperline=0, sizeimage=0.5MiB
device/buffer_list.c: VIDEO:capture:mplane: Opened 2 buffers. Memory used: 1.0 MiB
device/device.c: CAMERA: Setting frame interval_us=0 for FPS=30
device/libcamera/options.cc: CAMERA: Configuring option 'Brightness' (00000009, type=5) = 0.100000
device/v4l2/device_options.c: SNAPSHOT: Configuring option 'Compression Quality' (009d0903) = 80
device/v4l2/device_options.c: STREAM: Configuring option 'Compression Quality' (009d0903) = 80
device/v4l2/device_options.c: VIDEO: Configuring option 'Repeat Sequence Header' (009909e2) = 1
device/v4l2/device_options.c: VIDEO: Configuring option 'Video Bitrate Mode' (009909ce) = 0
device/v4l2/device_options.c: VIDEO: Configuring option 'Video Bitrate' (009909cf) = 2000000
device/v4l2/device_options.c: VIDEO: Configuring option 'Repeat Sequence Header' (009909e2) = 5000000
device/v4l2/device_options.c: VIDEO: Configuring option 'H264 I-Frame Period' (00990a66) = 30
device/v4l2/device_options.c: VIDEO: Configuring option 'H264 Level' (00990a67) = 11
device/v4l2/device_options.c: VIDEO: Configuring option 'H264 Profile' (00990a6b) = 4
device/v4l2/device_options.c: VIDEO: Configuring option 'H264 Minimum QP Value' (00990a61) = 16
device/v4l2/device_options.c: VIDEO: Configuring option 'H264 Maximum QP Value' (00990a62) = 32
device/links.c: ?: Link 0: CAMERA:capture[1437x1080/YUYV/2] => [SNAPSHOT:output:mplane[1408x1056/YUYV/2], RESCALLER:STREAM:output:mplane[1437x1080/YUYV/2], RESCALLER:VIDEO:output:mplane[1437x1080/YUYV/2]]
device/links.c: ?: Link 1: SNAPSHOT:capture:mplane[1408x1056/JPEG/2] => [SNAPSHOT-CAPTURE]
device/links.c: ?: Link 2: RESCALLER:STREAM:capture:mplane[640x480/YUYV/2] => [STREAM:output:mplane[640x480/YUYV/2]]
device/links.c: ?: Link 3: STREAM:capture:mplane[640x480/JPEG/2] => [STREAM-CAPTURE]
device/links.c: ?: Link 4: RESCALLER:VIDEO:capture:mplane[992x736/YUYV/2] => [VIDEO:output:mplane[992x736/YUYV/2]]
device/links.c: ?: Link 5: VIDEO:capture:mplane[992x736/H264/2] => [VIDEO-CAPTURE]
device/buffer_list.c: CAMERA:capture: Streaming started... Was 0 of 2 enqueud
[... 51 lines cut ...]
device/links.c: RESCALLER:VIDEO:capture:mplane:buf0: Capture image is outdated. Skipped. Now: 9915886654, vs 9914835153.
device/links.c: RESCALLER:VIDEO:capture:mplane:buf0: Capture image is outdated. Skipped. Now: 9916446650, vs 9915401355.
device/links.c: RESCALLER:VIDEO:capture:mplane:buf0: Capture image is outdated. Skipped. Now: 9916966636, vs 9915920375.
device/links.c: RESCALLER:VIDEO:capture:mplane:buf0: Capture image is outdated. Skipped. Now: 9917536633, vs 9916486577.
device/links.c: RESCALLER:VIDEO:capture:mplane:buf0: Capture image is outdated. Skipped. Now: 9918056639, vs 9917005595.
device/links.c: RESCALLER:VIDEO:capture:mplane:buf0: Capture image is outdated. Skipped. Now: 9918616626, vs 9917571798.
device/links.c: RESCALLER:VIDEO:capture:mplane:buf0: Capture image is outdated. Skipped. Now: 9919136618, vs 9918090816.
device/links.c: RESCALLER:VIDEO:capture:mplane:buf0: Capture image is outdated. Skipped. Now: 9919706627, vs 9918657019.
device/links.c: RESCALLER:VIDEO:capture:mplane:buf0: Capture image is outdated. Skipped. Now: 9920226615, vs 9919176038.
device/links.c: RESCALLER:VIDEO:capture:mplane:buf0: Capture image is outdated. Skipped. Now: 9920786619, vs 9919742239.
device/links.c: RESCALLER:VIDEO:capture:mplane:buf0: Capture image is outdated. Skipped. Now: 9921306608, vs 9920261259.
device/links.c: RESCALLER:VIDEO:capture:mplane:buf0: Capture image is outdated. Skipped. Now: 9921876608, vs 9920827462.
device/links.c: RESCALLER:VIDEO:capture:mplane:buf0: Capture image is outdated. Skipped. Now: 9922396597, vs 9921346480.
[... cut ...]
device/links.c: RESCALLER:VIDEO:capture:mplane:buf0: Capture image is outdated. Skipped. Now: 9926216579, vs 9925168346.
device/links.c: RESCALLER:VIDEO:capture:mplane:buf0: Capture image is outdated. Skipped. Now: 9926736584, vs 9925687364.
device/links.c: RESCALLER:VIDEO:capture:mplane:buf0: Capture image is outdated. Skipped. Now: 9927296567, vs 9926253568.
device/links.c: RESCALLER:VIDEO:capture:mplane:buf0: Capture image is outdated. Skipped. Now: 9927816569, vs 9926772587.
device/links.c: RESCALLER:VIDEO:capture:mplane:buf0: Capture image is outdated. Skipped. Now: 9928386556, vs 9927338788.
^C

suaveolent, 2024-07-24:
@Loevion I found this issue:

https://github.com/ayufan/camera-streamer/issues/95 

After setting the `--camera-snapshot.height=2464` as opposed to `--camera-snapshot.height=1080` it now works for `YUYV` too.

jferguson4, 2024-10-16:
Switching to YUV420 solved same problem for me.  running on RaspBerryPi 3b, Bullseye, with the Raspberry Module 2 8MP cmaera.  I was unable to get this hardfware to workwith Bookworm.