ayufan/camera-streamer #107: Raspberry PI OS Bookworm 64 Bit error during first installation script

phuzzyday, 2023-10-12:
Are there packages which haven't yet caught up to the new OS?

See error posted below

pi@PI4:~ $ if [[ -e /etc/default/raspberrypi-kernel ]]; then
  PACKAGE=camera-streamer-raspi_0.2.6.bullseye_$(dpkg --print-architecture).deb
else
  PACKAGE=camera-streamer-generic_0.2.6.bullseye_$(dpkg --print-architecture).de                                                       b
fi
wget "https://github.com/ayufan/camera-streamer/releases/download/v0.2.6/$PACKAG                                                       E"
sudo apt install "$PWD/$PACKAGE"
--2023-10-12 13:17:03--  https://github.com/ayufan/camera-streamer/releases/down                                                       load/v0.2.6/camera-streamer-generic_0.2.6.bullseye_arm64.deb
Resolving github.com (github.com)... 140.82.114.3
Connecting to github.com (github.com)|140.82.114.3|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://objects.githubusercontent.com/github-production-release-asset-                                                       2e65be/477898191/6f0c9005-c297-4243-82f1-c23fccbfe95b?X-Amz-Algorithm=AWS4-HMAC-                                                       SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20231012%2Fus-east-1%2Fs3%2Faws4_                                                       request&X-Amz-Date=20231012T191704Z&X-Amz-Expires=300&X-Amz-Signature=726e25c522                                                       e28d4dd9cb8bbf98f072ef3774e8c1abcb991dc475142c34c34b1f&X-Amz-SignedHeaders=host&                                                       actor_id=0&key_id=0&repo_id=477898191&response-content-disposition=attachment%3B                                                       %20filename%3Dcamera-streamer-generic_0.2.6.bullseye_arm64.deb&response-content-                                                       type=application%2Foctet-stream [following]
--2023-10-12 13:17:04--  https://objects.githubusercontent.com/github-production                                                       -release-asset-2e65be/477898191/6f0c9005-c297-4243-82f1-c23fccbfe95b?X-Amz-Algor                                                       ithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20231012%2Fus-east                                                       -1%2Fs3%2Faws4_request&X-Amz-Date=20231012T191704Z&X-Amz-Expires=300&X-Amz-Signa                                                       ture=726e25c522e28d4dd9cb8bbf98f072ef3774e8c1abcb991dc475142c34c34b1f&X-Amz-Sign                                                       edHeaders=host&actor_id=0&key_id=0&repo_id=477898191&response-content-dispositio                                                       n=attachment%3B%20filename%3Dcamera-streamer-generic_0.2.6.bullseye_arm64.deb&re                                                       sponse-content-type=application%2Foctet-stream
Resolving objects.githubusercontent.com (objects.githubusercontent.com)... 185.1                                                       99.108.133, 185.199.111.133, 185.199.110.133, ...
Connecting to objects.githubusercontent.com (objects.githubusercontent.com)|185.                                                       199.108.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 10682248 (10M) [application/octet-stream]
Saving to: ‘camera-streamer-generic_0.2.6.bullseye_arm64.deb’

camera-streamer-gen 100%[===================>]  10.19M  10.1MB/s    in 1.0s

2023-10-12 13:17:05 (10.1 MB/s) - ‘camera-streamer-generic_0.2.6.bullseye_arm64.                                                       deb’ saved [10682248/10682248]

Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'camera-streamer-generic' instead of '/home/pi/camera-streamer-g                                                       eneric_0.2.6.bullseye_arm64.deb'
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 camera-streamer-generic : Depends: libavcodec58 (>= 7:4.0)
 libavformat58 : Depends: libavcodec58 (= 8:4.3.6-0+deb11u1+rpt5)
                 Depends: libsrt1.4-gnutls (>= 1.4.2) but it is not installable
E: Unable to correct problems, you have held broken packages.

ayufan, 2023-10-12:
@phuzzyday Not yet, will update it over weekend.

ayufan, 2023-10-17:
Should be fine: https://github.com/ayufan/camera-streamer/releases/tag/v0.2.7