ayufan/camera-streamer #109: device/buffer_queue.c: SNAPSHOT:output:mplane: The dma_buf (CAMERA:capture:buf0) is too long: 460800 vs space=450560

doctorkb, 2023-10-18:
error logged in syslog:
`device/buffer_queue.c: SNAPSHOT:output:mplane: The dma_buf (CAMERA:capture:buf0) is too long: 460800 vs space=450560`

Running on RasPi 4 8GB, using OctoPi version 0.2.5 (41d8dfd)

Trying to use two USB cameras.  This error appears regularly for one camera at a time.

I've tried turning off allow_dma, and changing the number of buffers from 1 to 2 to 3.

This results in periodic "Server Error" on the stream and snapshot URLs.  Reloading a few times usually fixes it.

aderusha, 2023-10-22:
I'm seeing the same behavior here on an rPi3 with USB attached logitech cameras with camera-streamer v0.27.  Also tried changing buffer count without resolution.

opjose, 2024-01-15:
I was able to get this to stop by doing the following. I am running four cameras on a PI4 4GB.
I increased the number of buffers (NBUF) in each of the camera conf files to 10.
I set the output image format to JPEG for both of my Logitech 320s that have hardware mjpeg encoding.
I set the output image format  to YUYV for my Raspi (PiCam) camera.
I set the output image format to YUYV for my USB Borescope camera.

This also fixed my "Server Error" issues with the new camera stack and results in quick connections to each of the cameras.

Note that you will have to check to see what output formats your cameras support and adjust accordingly. 
Mjpeg encoding puts less of a load on the PI and USB.

doctorkb, 2024-02-14:
> I was able to get this to stop by doing the following. I am running four cameras on a PI4 4GB. I increased the number of buffers (NBUF) in each of the camera conf files to 10. I set the output image format to JPEG for both of my Logitech 920s that have hardware mjpeg encoding. I set the output image format to YUYV for my Raspi (PiCam) camera. I set the output image format to YUYV for my USB Borescope camera.
> 
> This also fixed my "Server Error" issues with the new camera stack and results in quick connections to each of the cameras.
> 
> Note that you will have to check to see what output formats your cameras support and adjust accordingly. Mjpeg encoding puts less of a load on the PI and USB.

I upped the NBUF to 10, and switched both cams to JPEG... and now it all works great. Thank you!