ayufan/camera-streamer #112: Understanding/Reproducing the Mentioned Performance

clearRawPath22, 2023-10-30:
I am trying to reproduce the numbers mentioned in https://github.com/ayufan/camera-streamer/blob/main/docs/v4l2-isp-mode.md

I get at best 130ms latency for 120fps on 1280x720, instead of the mentioned 50ms (same camera model, RPi 4 1GB, with fresh last installation of Raspian. Viewing the MJPEG streamed over a WLAN in a Firefox browser in an Linux laptop. Raspi and laptop are next to the router). I know that it depends on the router, monitor, etc. And in my case, the tests were done at relatively dark environment, so the camera had probably larger exposure time.

Now I am confused, since that documentation page and [this](https://github.com/ayufan/camera-streamer/issues/35) thread () have conflicting information:

Thead: "Playing YouTube video, capturing this with camera, and having another tab open with streaming." - I assume this has NO streaming over Wifi involved
Documentation: "... was tested with camera connected to Raspberry PI Zero 2W, streaming MJPEG over WiFi with camera recording https://www.youtube.com/watch?v=e8ZtPSIfWPc from desktop monitor."

How can I understand that?
If the 50ms latency involving streaming over Wifi to another Raspi/PC/Tablet/etc, is is possible to describe the viewing setup (which device, program and/or commands were used to view the stream)?

Thank you,

clearRawPath22, 2023-12-14:
Does anyone know if the 32bits or 64bits Raspian makes any difference on this performance (specifically, latency)?

kbingham, 2023-12-15:
32bits vs 64bits _shouldn_'t make any specific difference to latency.

clearRawPath22, 2023-12-15:
Ok, I just suggested that because the MMAL is not supported on 64bits anymore... maybe one or other library would be using that (but I have very little knowledge about these topics). 

Anyways, about my initial question (repetition of the performance mentioned in this repo), I removed the wireless part (by using Ethernet cable from Raspi and Linux to router, instead of using Wifi) and then the routing part (connecting Raspi and Linux directly through Ethernet) from the latency measurement. The first experiment showed surprisingly almost no improvements at all (well, my Raspi and Linux were anyways next to the router) and the 2nd experiment improved the latency about 5-10ms.

Furthermore, all these experiements were done in a bright environment now. 

I still get clearly above 100ms latency (the best is an average of 110-130ms) under this very favorable conditions, instead of 50ms, using the same setup/commands mentioned in the documentation of the library here... Not sure what I could be doing wrong... My monitor on Linux is 60Hz, so a max of 16.7ms latency could be from there (on average 8.4ms)