ayufan/camera-streamer #118: Compile error against libcamera library

richard9999999999, 2023-11-26:
device/libcamera/buffer_list.cc: In function ‘int libcamera_buffer_list_open(buffer_list_t*)’:
device/libcamera/buffer_list.cc:103:21: error: ‘using element_type = class libcamera::CameraConfiguration’ {aka ‘class libcamera::CameraConfiguration’} has no member named ‘transform’
103 | configurations->transform |= libcamera::Transform::VFlip;
| ^~~~~~~~~
device/libcamera/buffer_list.cc:106:21: error: ‘using element_type = class libcamera::CameraConfiguration’ {aka ‘class libcamera::CameraConfiguration’} has no member named ‘transform’
106 | configurations->transform |= libcamera::Transform::HFlip;
| ^~~~~~~~~
device/libcamera/buffer_list.cc:108:26: error: ‘using element_type = class libcamera::CameraConfiguration’ {aka ‘class libcamera::CameraConfiguration’} has no member named ‘transform’
108 | if (!!(configurations->transform & libcamera::Transform::Transpose)) {
| ^~~~~~~~~
make: *** [Makefile:99: device/libcamera/buffer_list.o] Error 1

ayufan, 2023-11-26:
Yes. I'm aware of this, will fix it later today.

Eduardo-Morales-Alberti, 2023-11-26:
Same here, thank you ayufan!!

ayufan, 2023-11-27:
Resolved: https://github.com/ayufan/camera-streamer/commit/bc231917d811e4a6661fc0b01b3e3750c6babd59.

closed 2023-11-27 by ayufan

richard9999999999, 2023-11-28:
Works, thank you!