ayufan/camera-streamer #120: CSI-2 2 & 4 Lane Support

ETE-Design, 2023-11-30:
Hi.

The Raspberry Pi 5 seems to Support CSI-2 - 4 Lane, and there is already some camera chip that support 4-lane. Wouldn't it give a performance gain to support it? And do you know how to make this part?

kbingham, 2023-11-30:
Yes, the RPi5 supports 4 lane cameras. I have a couple here that I am testing. But this is probably a discussion more appropriate for the raspberry pi forums or kernel repository https://github.com/raspberrypi/linux/ rather than a camera streamer application.

kbingham, 2023-11-30:
In terms of 'performance gain' ... it really depends on what you're trying to achieve. You can either transfer twice the pixels in the same amount of time, or you can use half the clock speeds to transfer the same quantity of data.

ayufan, 2023-11-30:
@ETE-Design Unlikely to happen, since 4k60HZ are unlikely to be hardware encoded by the CPU. I will be testing this yet, but I would not have my hopes high.

kbingham, 2023-12-01:
Indeed, you're going to hit encoder performance limitations long before you hit MIPI Data lane limitations I believe.

kbingham, 2023-12-01:
@ayufan Note that Pi5 does not have a hardware encoder at all. RPi use ffmpeg in their rpicam-apps on pi5.

ayufan, 2023-12-01:
@kbingham Yeah, I corrected my message. I know which makes even h264 1080p@60fps at the edge on pi5 :)

kbingham, 2023-12-01:
https://github.com/raspberrypi/rpicam-apps/blob/main/encoder/libav_encoder.cpp

ayufan, 2023-12-01:
I'm thinking the same, so I will use ffmpeg for that. There's no reason to reinvent a wheel.

kbingham, 2023-12-01:
The awesome part then will mean your camera-streamer will really not be tied to RPi devices ... and it would run on /any/ libcamera supported platform.