ayufan/camera-streamer #121: Failed to enable unit: Unit file /usr/share/camera-streamer/examples/camera-streamer-raspi-v3-12MP.service does not exist.

C8forT, 2023-12-06:
/usr/share/camera-streamer/examples/camera-streamer-raspi-v3-12MP.service appears to be missing in version 0.2.8~bookworm.

kmusser2, 2023-12-12:
FYI: I installed camera-streamer-raspi_0.2.8.bookworm_arm64.deb, and I DO have the referenced file.
kmusser@cm4:/usr/share/camera-streamer/examples $ ls -l
total 36
-rw-r--r-- 1 root    root 1041 May 26  2023 camera-streamer-arducam-16MP.service
-rw-r--r-- 1 root    root 1115 May 26  2023 camera-streamer-arducam-64MP.service
-rw-r--r-- 1 root    root  752 May 26  2023 camera-streamer-generic-usb-cam.service
-rw-r--r-- 1 root    root  837 May 26  2023 camera-streamer-raspi-usb-cam.service
-rw-r--r-- 1 root    root 1098 May 26  2023 camera-streamer-raspi-v2-8MP.service
-rw-r--r-- 1 root    root 1137 May 26  2023 camera-streamer-raspi-v3-12MP.service

2114L3, 2023-12-25:
i have the same issue on a RP Model B (original) 32bit running bookworm 32bit and the examples are missing bar one.
the command `PACKAGE=camera-streamer-$(test -e /etc/default/raspberrypi-kernel && echo raspi || echo generic)_0.2.8.$(. /etc/os-release; echo $VERSION_CODENAME)_$(dpkg --print-architecture).deb`

then checking with `echo $PACKAGE` outputs `camera-streamer-generic_0.2.8.bookworm_armhf.deb`
looking at the releases it should have been  camera-streamer-raspi_0.2.8.bookworm_armhf.deb (note generic to raspi)

/etc/default/raspberrypi-kernel doesnt exist on my installation of 2023-12-11-raspios-bookworm-armhf-lite

boehser-enkel, 2024-07-19:
Same for me. Now i am unable to install the correct package. Is there a cleanup for the generic install?