ayufan/camera-streamer #123: webrtc UDP stream continues even after web-browser closes

kmusser2, 2023-12-12:
VERSION: camera-streamer-raspi_0.2.8.bookworm_arm64.deb

The webrtc video stream works very well -- starts quickly, reliably, and looks good.

However, if you close the browser and open wireshark on the client machine, filtering on UDP, you'll see UDP packets arriving from the camera device (in my case a Raspberry PI cm4).  This UDP streams appears to continue indefinitely until the camera-streamer service is restarted.  (I can't confirm if indeed it runs forever, but I have not seen it stop until the service is stopped.)

opjose, 2024-01-16:
I also see this in TCP, however once the recipient closes the target, e.g. browser, etc. the stream seems to pause.
If the browser, etc. is opened again the stream resumes even if quite some time has occurred.

sarusani, 2024-01-28:
I believe  the problem is that the "webrtc.html" never stops the stream. I think it should include something like:

```
document.onvisibilitychange = () => {
  if (document.visibilityState === "hidden") {
    pc.close();
  }
};
```

To close the stream if the browser/tab is closed.