ayufan/camera-streamer #125: Raspberry  PI 5 with raspicam V3

barthm1, 2023-12-13:
Using this configuration  with latest Rasbian OS 64 Bit - service can not be started 
Using the LCCV software I can use the camera

camera-streamer[40642]: device/camera/camera_output.c: CAMERA: Cannot find encoder to convert from 'YUYV'

[Unit]
Description=camera-streamer web camera for Pi Camera Module 3 12MP on Raspberry PI
After=network.target
# ConditionPathExists=/sys/bus/i2c/drivers/imx708/10-001a/video4linux
ConditionPathExists=/sys/bus/i2c/drivers/imx708/6-001a/video4linux

[Service]
ExecStart=/usr/local/bin/camera-streamer \
  ; -camera-path=/base/soc/i2c0mux/i2c@1/imx708@1a \
  --camera-path=/base/axi/pcie@120000/rp1/i2c@88000/imx708@1a \
  --camera-type=libcamera \
  --camera-format=YUYV \
  --camera-width=1920 --camera-height=1080 \
  --camera-fps=30 \

camera-streamer[40642]: [94:39:45.048625304] [40642]  INFO Camera camera.cpp:1183 configuring streams: (0) 1920x1080-YUYV
Dec 13 11:09:44 raspi601 camera-streamer[40642]: [94:39:45.048745061] [40654]  INFO RPI pisp.cpp:1396 Sensor: /base/axi/pcie@120000/rp1/i2c@88000/imx708@1a - Selected sensor format: 2304x12>
Dec 13 11:09:44 raspi601 camera-streamer[40642]: [94:39:45.049129573] [40642]  INFO Camera camera.cpp:1183 configuring streams: (0) 1920x1080-YUYV (1) 2304x1296-BGGR16_PISP_COMP1
Dec 13 11:09:44 raspi601 camera-streamer[40642]: [94:39:45.049211257] [40654]  INFO RPI pisp.cpp:1396 Sensor: /base/axi/pcie@120000/rp1/i2c@88000/imx708@1a - Selected sensor format: 2304x12>
Dec 13 11:09:44 raspi601 camera-streamer[40642]: device/buffer_list.c: CAMERA:capture: Using: 1920x1080/YUYV, buffers=2, bytesperline=3840, sizeimage=0.0MiB
Dec 13 11:09:44 raspi601 camera-streamer[40642]: device/buffer_list.c: CAMERA:capture: Opened 2 buffers. Memory used: 7.9 MiB
Dec 13 11:09:44 raspi601 camera-streamer[40642]: device/buffer_list.c: CAMERA:capture:1: Using: 2304x1296/BYR2, buffers=2, bytesperline=2304, sizeimage=0.0MiB
Dec 13 11:09:44 raspi601 camera-streamer[40642]: device/buffer_list.c: CAMERA:capture:1: Opened 2 buffers. Memory used: 5.7 MiB
Dec 13 11:09:44 raspi601 camera-streamer[40642]: device/camera/camera_output.c: CAMERA: Cannot find encoder to convert from 'YUYV'
Dec 13 11:09:44 raspi601 systemd[1]: camera-streamer.service: Main process exited, code=exited, status=255/EXCEPTION

DjofPlusPlus, 2023-12-14:
See the comments on this other issue: https://github.com/ayufan/camera-streamer/issues/120

The RPI 5 has no hardware encoding support so software encoding will be needed.

closed 2023-12-16 by barthm1

smurfxx, 2023-12-22:
Sorry if I ask this question but I have zero experience on these things: in the current state of affairs is there a solution to make camera-streamer work with raspicam v3? If so, is there a step-by-step guide that explains it to me?

ayufan, 2023-12-22:
@smurfxx Not yet supported, but I plan to add it very soon.

smurfxx, 2023-12-22:
Ok, thank you very much! I'll follow this github project for future updates!

smurfxx, 2023-12-22:
Another question: does camera-streamer work with older raspicam models or is the problem related to the operating system and/or new hardware of the Pi5?

smurfxx, 2024-02-27:
> @smurfxx I couldn't get my older Raspberry Pi Cam OV5647 working, for what it's worth.

I finished following this project because there is one that makes any webcam (even USB) work without any problem, on customized ports and with many other features.
The project is this: https://github.com/signag/raspi-cam-srv