ayufan/camera-streamer #128: Camera module 3, Raspberry Pi OS Desktop, Raspberry Pi 5

SPYKE357, 2024-01-06:
Good day all

I'm trying to connect a raspi camera module v3 through octoprint.
The os I flashed through the imager is the raspberry pi os 64bitb1.1gb realeased on the 05/12/23. After flashing i used octoprint_deploy to get octoprint up and running.
I followed the manual installation method for camera-streamer but camera still not showing in octoprint. Then tried the packaged install but also no luck. The camera is working as it displays when I have a screen connected to the pi and run libcamera-hello.

Is it possible to get this camera running in octoprint on my device? If it is possible please do help me out with how to get it done. Im bite to this but willing to learn

gatonero, 2024-03-09:
I have the very same problem on a RPi5. On a very fresh installation after `sudo apt update && sudo apt full-upgrade -y`.

 I also set in /boot/firmware/config.txt camera_auto_detect=0 and at the very end after [all] dtoverlay=imx708. Her is my system information [https://paste.debian.net/hidden/de882afe/](url)

DjofPlusPlus, 2024-03-09:
The RP5 isn't currently supported because it doesn't have a hardware video encoder. Eventually there might be software encoding support. You can find more information by searching further back.

gatonero, 2024-03-09:
@DjofPlusPlus 
What do you mean by ' You can find more information by searching further back.' I'm now searching for about 12 hours in two days.