ayufan/camera-streamer #130: Raspberry Pi 5 does not appear to be supported at the moment?

tom-ard, 2024-01-17:
I successfully compiled camera-streamer on raspberry pi 5 and bookworm system.But when I run the program, the following error appears:

[image]

I think this could be related to the fact that the Raspberry Pi 5 doesn't have a hardware encoder.

Currently, libcamera-apps uses software coding when running on Raspberry Pi 5.

When will camera-streamer support Raspberry Pi 5?

DjofPlusPlus, 2024-01-17:
There's no hardware encoder like for previous pi devices so the 5 isn't supported yet. See https://github.com/ayufan/camera-streamer/issues/120

puterboy, 2024-01-22:
So does that mean that camera-streamer does not work at all with the RPi5?
I have both PiCam3 and PiCam3 cameras and would like to get them streaming on my RPi5.
I am stuck on the same error as the OP.

- Is this something that can be fixed now by piping the input/output to a software-based converter?
- If not, is there likely to be a solution in the future?

I am a bit confused since I thought the RPi5 was supposed to be better/easier at video thatn the RPI4 but it seems between the move to libcamera and the lack of a hardware converter that things that "just worked" or were "relatively easy" with the RPI4 no longer work in the world of the RPI5.

Should I just be sticking to RPI4 running old, stable Buster?

barthm1, 2024-01-24:
Have a look at https://github.com/signag/raspi-cam-srv

geerlingguy, 2024-07-30:
picamera2-WebUI is also an alternative that I've been using with Pi 5 lately, it seems easy to get running and provides a lot of easy configuration/easy access to the camera streams: https://github.com/monkeymademe/picamera2-WebUI