ayufan/camera-streamer #131: Unable to view camera through WebRTC stream.

cray12399, 2024-01-19:
My setup is as follows:
Distribution: Debian GNU/Linux 12 (bookworm)
Kernel: Linux 6.1.0-rpi7-rpi-v8 aarch64
Model:  Raspberry Pi 4 Model B Rev 1.5
Browser: Firefox and Chromium

2 Cameras:
[OV5640](https://www.amazon.com/gp/product/B09WQQ3FXP/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1)
[OV2710](https://www.amazon.com/gp/product/B0829HZ3Q7/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1)

I am using this setup with a 3D printer and [Crowsnest](https://github.com/mainsail-crew/crowsnest).

The issue that I am having is that I am trying to stream my cameras via WebRTC, but no matter what I do I cannot view the WebRTC streams. It just gives me a blank screen. I was told by someone on the Mainsail Discord to come here after they were unable to find the issue with my setup. For all intents and purposes, it seems like I have everything set up correctly. I know the cameras work because they run fine with ustreamer mjpeg. 

Here is my Crowsnest log:
[crowsnest (1).log](https://github.com/ayufan/camera-streamer/files/13992848/crowsnest.1.log)

Here is my crowsnest conf:
[crowsnest.conf.txt](https://github.com/ayufan/camera-streamer/files/13992886/crowsnest.conf.txt)

Any idea on what can be going on here?

AlexanderS, 2024-01-25:
You may want to try to raise `gpu_mem` to 128MB. This fixed the issue for me.

cray12399, 2024-01-25:
> You may want to try to raise `gpu_mem` to 128MB. This fixed the issue for me.

Hi Alexander, thank you for your help. Where am I supposed to put that setting?

KennyVB, 2024-01-28:
got same issue, exact
i got 256mb set. rpi 4b 4gb 
and tried camera-always-active=1 
nothing works

AlexanderS, 2024-02-02:
> Hi Alexander, thank you for your help. Where am I supposed to put that setting?

This should be added to `/boot/config.txt`.

And please check if your browser supports WebRTC with H264. You can use for example the following page: https://mozilla.github.io/webrtc-landing/pc_test_no_h264.html

I found the following issues with common browsers:
- In Firefox you have to enable it via `media.webrtc.hw.h264.enabled` in `about:config` (See https://unix.stackexchange.com/a/761511 and https://bugzilla.mozilla.org/show_bug.cgi?id=1717679)
- Chromium in Debian Bullseye does not support h264 because libopenh264 is only available since bookworm.

cray12399, 2024-02-03:
> > Hi Alexander, thank you for your help. Where am I supposed to put that setting?
> 
> This should be added to `/boot/config.txt`.
> 
> And please check if your browser supports WebRTC with H264. You can use for example the following page: https://mozilla.github.io/webrtc-landing/pc_test_no_h264.html
> 
> I found the following issues with common browsers:
> 
>     * In Firefox you have to enable it via `media.webrtc.hw.h264.enabled` in `about:config` (See https://unix.stackexchange.com/a/761511 and https://bugzilla.mozilla.org/show_bug.cgi?id=1717679)
> 
>     * Chromium in Debian Bullseye does not support h264 because libopenh264 is only available since bookworm (See f.e. https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1031352)

I enabled h264 in Firefox and increased my rpi memory to 256mb. I am able to get a picture sometimes showing at this point, but I am getting like 1 frame every 10 seconds or just a blank black screen. Its worse performance than mjpeg with ustreamer. Any suggestions on what I should do next?