ayufan/camera-streamer #135: Raspberry Pi Camera v1.3

Bersaker, 2024-02-14:
Hi. I Have all tested.. but nothing works.  is the raspberry pi v1.3 5M supported?

Riconec, 2024-02-24:
I got it working on bullseye, on bookworm it was working in libcamera but no signal in the camera-streamer

whycao, 2024-05-26:
Same issue.

I'm using zero 2w with a V1 Pi camera, OctoPrint version: 1.10.1.

Currently, the only way is by using http://pizero.local/webcam/control to start streaming.
However, pizero.local/:8080/stream can't reach.
From the browser dev tool I found pizero.local/webcam/stream?_cb=1716623****** is working

I tried on Pi5 fresh install and edit the service file to camera-streamer-raspi-v1-5MP.service
Unlucky not working.

[Unit]
Description=camera-streamer web camera for Pi Camera v1.3 5MP on Raspberry PI
After=network.target
ConditionPathExists=/sys/bus/i2c/drivers/ov5647/6-0036/video4linux

[Service]
ExecStart=/usr/bin/camera-streamer \
  -camera-path=/base/axi/pcie@120000/rp1/i2c@88000/ov5647@36 \
  -camera-type=libcamera \
  -camera-format=YUYV \
  -camera-width=2592 -camera-height=1944 \
  -camera-fps=30 \
  ; use two memory buffers to optimise usage
  -camera-nbufs=2 \
  ; the snapshot is 1438x1080
  -camera-snapshot.height=1080 \
  ; the video/webrtc is 958x720
  -camera-video.height=720 \
  ; the stream is 639x480
  -camera-stream.height=480 \
  ; bump brightness slightly
  -camera-options=brightness=0.1 \
  --http-listen=0.0.0.0 \
  --http-port=8080 \
  -rtsp-port

DynamicUser=yes
SupplementaryGroups=video i2c
Restart=always
RestartSec=10
Nice=10
IOSchedulingClass=idle
IOSchedulingPriority=7
CPUWeight=20
AllowedCPUs=1-2
MemoryMax=250M

[Install]
WantedBy=multi-user.target

kingtat, 2024-07-01:
I also have a v1.3 camera working. RPi B+ v1 debian bullseye. Fortunately, I used the bullseye rpi image. Only because bookworm is not available for the early pi.