ayufan/camera-streamer #137: new libcamera 0.2.0+rpt20240215-1 makes camera-streamer.service failing

rongith, 2024-02-20:
I just upgraded my bookworm system, then libcamera was upgraded to 0.2.0+rpt20240215-1.
camera-streamer.service couldn't start anymore.
$ journalctl -xef -u camera-streamer :
```
The job identifier is 181888 and the job result is done.
Feb 20 16:42:13 3DP-R01-CB systemd[1]: Started camera-streamer.service - camera-streamer web camera for Pi Camera v2.1 8MP on Raspberry PI.
░░ Subject: A start job for unit camera-streamer.service has finished successfully
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░
░░ A start job for unit camera-streamer.service has finished successfully.
░░
░░ The job identifier is 181888.
Feb 20 16:42:14 3DP-R01-CB camera-streamer[52143]: util/http/http.c: ?: HTTP listening on 0.0.0.0:8080.
Feb 20 16:42:14 3DP-R01-CB camera-streamer[52143]: output/rtsp/rtsp.cc: ?: Running RTSP server on '8554'
Feb 20 16:42:14 3DP-R01-CB camera-streamer[52143]: /usr/bin/camera-streamer Version: 0.2.8 (bc23191)
Feb 20 16:42:14 3DP-R01-CB camera-streamer[52143]: [7:06:53.684143159] [52143]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+118-563cd78e
Feb 20 16:42:14 3DP-R01-CB camera-streamer[52143]: terminate called after throwing an instance of 'std::out_of_range'
Feb 20 16:42:14 3DP-R01-CB camera-streamer[52143]:   what():  unordered_map::at
Feb 20 16:42:14 3DP-R01-CB systemd[1]: camera-streamer.service: Main process exited, code=killed, status=6/ABRT
░░ Subject: Unit process exited
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░
░░ An ExecStart= process belonging to unit camera-streamer.service has exited.
░░
░░ The process' exit code is 'killed' and its exit status is 6.
Feb 20 16:42:14 3DP-R01-CB systemd[1]: camera-streamer.service: Failed with result 'signal'.
░░ Subject: Unit failed
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░
░░ The unit camera-streamer.service has entered the 'failed' state with result 'signal'.
Feb 20 16:42:24 3DP-R01-CB systemd[1]: camera-streamer.service: Scheduled restart job, restart counter is at 2440.
░░ Subject: Automatic restarting of a unit has been scheduled
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░
░░ Automatic restarting of the unit camera-streamer.service has been scheduled, as the result for
░░ the configured Restart= setting for the unit.
Feb 20 16:42:24 3DP-R01-CB systemd[1]: Stopped camera-streamer.service - camera-streamer web camera for Pi Camera v2.1 8MP on Raspberry PI.
░░ Subject: A stop job for unit camera-streamer.service has finished
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░
░░ A stop job for unit camera-streamer.service has finished.
```

I then downgraded libcamera to make it run again:
```
wget https://archive.raspberrypi.org/debian/pool/main/libc/libcamera/libcamera-dev_0.1.0+rpt20231122-1_arm64.deb
wget https://archive.raspberrypi.org/debian/pool/main/libc/libcamera/libcamera-ipa_0.1.0+rpt20231122-1_arm64.deb
sudo apt install ./libcamera-dev:arm64=0.1.0+rpt20231122-1 ./libcamera-ipa:arm64=0.1.0+rpt20231122-1
```

I join my $ /usr/share/camera-streamer/dump_cameras.sh
[dump_cameras.sh.txt](https://github.com/ayufan/camera-streamer/files/14351302/dump_cameras.sh.txt)

boecko, 2024-02-21:
I've the same problem with a  rpi4 on the armhf platform.

```$ /usr/share/camera-streamer/dump_cameras.sh > dumcameras.sh.txt``` attached.

[dumpcameras.sh.txt](https://github.com/ayufan/camera-streamer/files/14357245/dumpcameras.sh.txt)

KapJI, 2024-02-22:
This happens for camera-streamer precompiled for libcamera0.1.

When I manually compile it for libcamera0.2, I get this:
```
[1:04:13.626468831] [16703]  INFO Camera camera_manager.cpp:284 libcamera v0.2.0+46-075b54d5
[1:04:13.761606569] [16716]  WARN RPiSdn sdn.cpp:39 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[1:04:13.765678930] [16716]  INFO RPI vc4.cpp:447 Registered camera /base/soc/i2c0mux/i2c@1/imx708@1a to Unicam device /dev/media4 and ISP device /dev/media0
device/libcamera/device.cc: CAMERA: Device path=/base/soc/i2c0mux/i2c@1/imx708@1a opened
[1:04:13.768397560] [16703]  INFO Camera camera.cpp:1183 configuring streams: (0) 1920x1080-YUYV
[1:04:13.768998093] [16716]  INFO RPI vc4.cpp:611 Sensor: /base/soc/i2c0mux/i2c@1/imx708@1a - Selected sensor format: 2304x1296-SBGGR10_1X10 - Selected unicam format: 2304x1296-pBAA
[1:04:13.771658334] [16703]  INFO Camera camera.cpp:1183 configuring streams: (0) 1920x1080-YUYV (1) 2304x1296-SBGGR10_CSI2P
[1:04:13.772143201] [16716]  INFO RPI vc4.cpp:611 Sensor: /base/soc/i2c0mux/i2c@1/imx708@1a - Selected sensor format: 2304x1296-SBGGR10_1X10 - Selected unicam format: 2304x1296-pBAA
[1:04:26.757112936] [16716] ERROR V4L2 v4l2_videodevice.cpp:1697 /dev/video13[26:out]: Failed to queue buffer 0: Invalid argument
[1:04:26.757196232] [16716] ERROR RPISTREAM rpi_stream.cpp:276 Failed to queue buffer for ISP Input
```

boecko, 2024-02-22:
Recompiling with 
libcamera-dev:armhf (0.2.0+rpt20240215-1)
works for me.

tom-ard, 2024-03-01:
I also encountered this problem.

[image]

I found that this problem was solved when I forcefully set "configuration.stride" to 0.

[image]

KapJI, 2024-03-01:
Ok, basically crash is expected when version of `libcamera-ipa` doesn't match `libcamera-dev` with which `camera-streamer` was built. And precompiled version was built with `libcamera-dev 0.1.0`.

And compatibility issue with libcamera0.2 is tracked by #139

ayufan, 2024-03-01:
I will push new compilation on Monday.

DropbearNinja, 2024-05-10:
Hi,
I am also having this issue:

`May 10 03:46:20 duet3 systemd[1]: camera-streamer-raspi-v3-12MP.service: Scheduled restart job, restart counter is at 51.
May 10 03:46:20 duet3 systemd[1]: Stopped camera-streamer-raspi-v3-12MP.service - camera-streamer web camera for Pi Camera Module 3 12MP on Raspberry PI.
May 10 03:46:20 duet3 systemd[1]: Started camera-streamer-raspi-v3-12MP.service - camera-streamer web camera for Pi Camera Module 3 12MP on Raspberry PI.
May 10 03:46:21 duet3 camera-streamer[4127]: util/opts/opts.c: ?: Usage of '-camera-path=/base/soc/i2c0mux/i2c@1/imx708@1a' is deprecated change to '--camera-path=/base/soc>
May 10 03:46:21 duet3 camera-streamer[4127]: util/opts/opts.c: ?: Usage of '-camera-type=libcamera' is deprecated change to '--camera-type=libcamera'.
May 10 03:46:21 duet3 camera-streamer[4127]: util/opts/opts.c: ?: Usage of '-camera-format=YUYV' is deprecated change to '--camera-format=YUYV'.
May 10 03:46:21 duet3 camera-streamer[4127]: util/opts/opts.c: ?: Usage of '-camera-width=2304' is deprecated change to '--camera-width=2304'.
May 10 03:46:21 duet3 camera-streamer[4127]: util/opts/opts.c: ?: Usage of '-camera-height=1296' is deprecated change to '--camera-height=1296'.
May 10 03:46:21 duet3 camera-streamer[4127]: util/opts/opts.c: ?: Usage of '-camera-fps=30' is deprecated change to '--camera-fps=30'.
May 10 03:46:21 duet3 camera-streamer[4127]: util/opts/opts.c: ?: Usage of '-camera-nbufs=2' is deprecated change to '--camera-nbufs=2'.
May 10 03:46:21 duet3 camera-streamer[4127]: util/opts/opts.c: ?: Usage of '-camera-snapshot.height=1080' is deprecated change to '--camera-snapshot.height=1080'.
May 10 03:46:21 duet3 camera-streamer[4127]: util/opts/opts.c: ?: Usage of '-camera-video.height=720' is deprecated change to '--camera-video.height=720'.
May 10 03:46:21 duet3 camera-streamer[4127]: util/opts/opts.c: ?: Usage of '-camera-stream.height=480' is deprecated change to '--camera-stream.height=480'.
May 10 03:46:21 duet3 camera-streamer[4127]: util/opts/opts.c: ?: Usage of '-camera-options=AfMode=2' is deprecated change to '--camera-options=AfMode=2'.
May 10 03:46:21 duet3 camera-streamer[4127]: util/opts/opts.c: ?: Usage of '-camera-options=AfRange=2' is deprecated change to '--camera-options=AfRange=2'.
May 10 03:46:21 duet3 camera-streamer[4127]: util/opts/opts.c: ?: Usage of '-rtsp-port' is deprecated change to '--rtsp-port'.
May 10 03:46:21 duet3 camera-streamer[4127]: util/http/http.c: ?: HTTP listening on 0.0.0.0:8080.
May 10 03:46:21 duet3 camera-streamer[4127]: output/rtsp/rtsp.cc: ?: Running RTSP server on '8554'
May 10 03:46:21 duet3 camera-streamer[4127]: /usr/bin/camera-streamer Version: 0.2.8 (bc23191)
May 10 03:46:21 duet3 camera-streamer[4127]: [0:24:28.027717129] [4127]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+118-563cd78e
May 10 03:46:21 duet3 camera-streamer[4127]: terminate called after throwing an instance of 'std::out_of_range'
May 10 03:46:21 duet3 camera-streamer[4127]:   what():  unordered_map::at
May 10 03:46:21 duet3 systemd[1]: camera-streamer-raspi-v3-12MP.service: Main process exited, code=killed, status=6/ABRT
May 10 03:46:21 duet3 systemd[1]: camera-streamer-raspi-v3-12MP.service: Failed with result 'signal'.`

> I will push new compilation on Monday.
The current version is 0.2.8 which was "released this Nov 28, 2023"

No update yet?