ayufan/camera-streamer #139: The camera-streamer compiled based on libcamera0.2 cannot run normally

tom-ard, 2024-03-01:
When I updated libcamera to version 0.2, the following error occurred when compiling and running camera-streamer again.
[image]
[image]

By printing some key information, I found that this was caused by the RAW stream's configuration.stride equal to 4128.
[image]

You only need to force configuration.stride to be 0 to solve this problem.
[image]
[image]

CmdrAdama, 2024-03-02:
I can confirm that making this change manually and rebuilding it resolved the issue for me as well.

closed 2024-03-08 by ayufan

DropbearNinja, 2024-05-10:
Where do you set configuration.stride = 0?

PR #140 (2024-03-08): Fix bytesperline calculation for libcamera0.2
For libcamera0.2 `configuraiton.stride` is already set to some incorrect value and `configurations->validate()` doesn't update it if it's not 0.

For libcamera0.1 it was set to 0. So set it to 0 manually to make sure `configurations->validate()` computes the correct value.

Before:
```
device/buffer_list.c: CAMERA:capture: Using: 1920x1080/YUYV, buffers=3, bytesperline=3840, sizeimage=0.0MiB
device/buffer_list.c: CAMERA:capture:1: Using: 2304x1296/BG10, buffers=3, bytesperline=5760, sizeimage=0.0MiB
ERROR V4L2 v4l2_videodevice.cpp:1697 /dev/video13 [... cut]