ayufan/camera-streamer #141: Note, selecting 'camera-streamer-generic' instead of '/home/kevin/camera-streamer-generic_0.2.8.bookworm_arm64.deb'

kg117, 2024-03-02:
[camera-streamer.txt](https://github.com/ayufan/camera-streamer/files/14468333/camera-streamer.txt)
My RPi OS is BookWorm V12 installed today.
Does 'selecting 'camera-streamer-generic' instead of' mean that the OS already included this item?
Please see attached text file.

DropbearNinja, 2024-05-10:
I'm getting the same behviour:

`Note, selecting 'camera-streamer-generic' instead of './camera-streamer-generic_0.2.8.bookworm_armhf.deb'`

This means that the examples dir contains a generic example rather than the 6 raspberry pi ones:

`pi@duet3:~ $ ls -al /usr/share/camera-streamer/examples/
total 12
drwxr-xr-x 2 root root 4096 May 10 03:31 .
drwxr-xr-x 3 root root 4096 May 10 03:31 ..
-rw-r--r-- 1 root root  752 May 26  2023 camera-streamer-generic-usb-cam.service
`

DropbearNinja, 2024-05-10:
Ah, it seams that 

`PACKAGE=camera-streamer-$(test -e /etc/default/raspberrypi-kernel && echo raspi || echo generic)_0.2.8.$(. /etc/os-release; echo $VERSION_CODENAME)_$(dpkg --print-architecture).deb`

Selects generic rather than raspi for some reason

If you manually download the appropriate version, it works

i.e. replace $PACKAGE yourself in this line:

`wget "https://github.com/ayufan/camera-streamer/releases/download/v0.2.8/$PACKAGE"`