ayufan/camera-streamer #142: camera-streamer web camera for Pi Camera Module 3 12MP on Raspberry PI was skipped because of an unmet condition check

kg117, 2024-03-03:
[compile04032024.txt](https://github.com/ayufan/camera-streamer/files/14474463/compile04032024.txt)
As shown in the attached text file I have completed a manual install of camera-streamer.
The command: root@raspberrypi:/usr/local/bin# systemctl start camera-streamer-raspi-v3-12MP: Log:
root@raspberrypi:/usr/local/bin# journalctl -xef -u camera-streamer-raspi-v3-12MP
`
Mar 04 09:01:47 raspberrypi systemd[1]: camera-streamer-raspi-v3-12MP.service - camera-streamer web camera for Pi Camera Module 3 12MP on Raspberry PI was skipped because of an unmet condition check (ConditionPathExists=/sys/bus/i2c/drivers/imx708/10-001a/video4linux).
░░ Subject: A start job for unit camera-streamer-raspi-v3-12MP.service has finished successfully
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ A start job for unit camera-streamer-raspi-v3-12MP.service has finished successfully.
░░ 
░░ The job identifier is 2190.
`
I am a beginner for Raspberry Pi. Your help would be much appreciated. I have looked at: /sys/bus/i2c/drivers/imx477/
While installing RPi 4B OS yesterday I don't remember configuring anything for I2C?
Kevin.

kg117, 2024-03-04:
[imx477 04032024.txt](https://github.com/ayufan/camera-streamer/files/14475467/imx477.04032024.txt)
The attached file shows where imx477 appears in the rpicam-apps Bug Report. Yet the log above shows: 
ConditionPathExists=/sys/bus/i2c/drivers/imx708/10-001a/video4linux
which states imx708
My system only has: /sys/bus/i2c/drivers/imx477/ which is correct.
Please help. Kevin

kg117, 2024-03-04:
[compile04032024A.txt](https://github.com/ayufan/camera-streamer/files/14477849/compile04032024A.txt)

Changed the name of camera-streamer-raspi-v3-12MP.service to camera-streamer,service and changed three references in camera-streamer,service file from imx708 to imx477 as I am using a HQ Camera. I was confused by the use of "V3" which does not appear in:
https://www.raspberrypi.com/documentation/computers/camera_software.html
V1 camera (OV5647)	dtoverlay=ov5647
V2 camera (IMX219)	dtoverlay=imx219
HQ camera (IMX477)	dtoverlay=imx477
GS camera (IMX296)	dtoverlay=imx296
Camera Module 3 (IMX708)dtoverlay=imx708
To call the last item "V3" is confusing. Sometimes it is called "CM3"?
I made the error of assuming HQ Camera was V3.
All my text files show everything output to the terminal.
Is my camera-streamer now complete?
Kevin.

kg117, 2024-03-06:
http://192.168.0.15:8080/ please view files attached.
[first page 06032024.txt](https://github.com/ayufan/camera-streamer/files/14503707/first.page.06032024.txt)
[status page 06032024.txt](https://github.com/ayufan/camera-streamer/files/14503708/status.page.06032024.txt)
http://192.168.0.15:8080/snapshot
On windows 10 in Firefox "Server error" only item displayed.
Any suggestions please.
Kevin.

kg117, 2024-05-18:
Arducam camera UC-A65 Rev B 12MP IMX708 model B0310 120°(H) wide-angle M12 Lens ribbon UC-376 on Raspberry Pi 4B
Camera-streamer works 100%
Many thanks

Dephrilibrium, 2025-07-27:
> Camera Module 3 (IMX708)dtoverlay=imx708
> To call the last item "V3" is confusing. Sometimes it is called "CM3"?
> I made the error of assuming HQ Camera was V3.
> All my text files show everything output to the terminal.
> Is my camera-streamer now complete?

That's true. HQ-Cam is often also named v3-module. I also felt into the same rabbit hole. 
However, i was successfully changing the service to open the camera. But it tells me now that it does not find a YUYV-converter
```journalctl -u camera-streamer-rapi-HQCam-12MP.service``` yields:
```bash
Jul 27 14:34:11 raspi5 systemd[1]: Started camera-streamer-raspi-HQCam-12MP.service - camera-streamer web camera for Pi High Quality Camera Module 12MP on Raspberry PI.
Jul 27 14:34:11 raspi5 camera-streamer[39116]: util/http/http.c: ?: HTTP listening on 0.0.0.0:8080.
Jul 27 14:34:11 raspi5 camera-streamer[39116]: /usr/local/bin/camera-streamer Version: v0.3.0-1-g400e036 (400e036)
Jul 27 14:34:11 raspi5 camera-streamer[39116]: [4:18:10.752401050] [39116]  INFO Camera camera_manager.cpp:326 libcamera v0.5.1+100-e53bdf1f
Jul 27 14:34:11 raspi5 camera-streamer[39116]: [4:18:10.761599021] [39129]  INFO RPI pisp.cpp:720 libpisp version v1.2.1 981977ff21f3 29-04-2025 (14:13:50)
Jul 27 14:34:11 raspi5 camera-streamer[39116]: [4:18:10.776284899] [39129]  INFO RPI pisp.cpp:1179 Registered camera /base/axi/pcie@1000120000/rp1/i2c@88000/imx477@1a to CFE device /dev/media3 and ISP device /dev/media0 using Pi>
Jul 27 14:34:11 raspi5 camera-streamer[39116]: device/libcamera/device.cc: CAMERA: Device path=/base/axi/pcie@1000120000/rp1/i2c@88000/imx477@1a opened
Jul 27 14:34:11 raspi5 camera-streamer[39116]: [4:18:10.776724328] [39116]  INFO Camera camera.cpp:1205 configuring streams: (0) 1920x1080-YUYV/sYCC
Jul 27 14:34:11 raspi5 camera-streamer[39116]: [4:18:10.776829921] [39129]  INFO RPI pisp.cpp:1483 Sensor: /base/axi/pcie@1000120000/rp1/i2c@88000/imx477@1a - Selected sensor format: 2028x1080-SBGGR12_1X12 - Selected CFE format:>
Jul 27 14:34:11 raspi5 camera-streamer[39116]: [4:18:10.777222276] [39116]  INFO Camera camera.cpp:1205 configuring streams: (0) 1920x1080-YUYV/sYCC (1) 2028x1080-BGGR_PISP_COMP1/RAW
Jul 27 14:34:11 raspi5 camera-streamer[39116]: [4:18:10.777290665] [39129]  INFO RPI pisp.cpp:1483 Sensor: /base/axi/pcie@1000120000/rp1/i2c@88000/imx477@1a - Selected sensor format: 2028x1080-SBGGR12_1X12 - Selected CFE format:>
Jul 27 14:34:11 raspi5 camera-streamer[39116]: device/buffer_list.c: CAMERA:capture: Using: 1920x1080/YUYV, buffers=2, bytesperline=3840, sizeimage=0.0MiB
Jul 27 14:34:11 raspi5 camera-streamer[39116]: device/buffer_list.c: CAMERA:capture: Opened 2 buffers. Memory used: 7.9 MiB
Jul 27 14:34:11 raspi5 camera-streamer[39116]: device/buffer_list.c: CAMERA:capture:1: Using: 2028x1080/BYR2, buffers=2, bytesperline=2048, sizeimage=0.0MiB
Jul 27 14:34:11 raspi5 camera-streamer[39116]: device/buffer_list.c: CAMERA:capture:1: Opened 2 buffers. Memory used: 4.2 MiB
# Error is here
>> Jul 27 14:34:11 raspi5 camera-streamer[39116]: device/camera/camera_output.c: CAMERA: Cannot find encoder to convert from 'YUYV'
>> Jul 27 14:34:11 raspi5 systemd[1]: camera-streamer-raspi-HQCam-12MP.service: Main process exited, code=exited, status=255/EXCEPTION
# ------------------
Jul 27 14:34:11 raspi5 systemd[1]: camera-streamer-raspi-HQCam-12MP.service: Failed with result 'exit-code'.
Jul 27 14:34:21 raspi5 systemd[1]: camera-streamer-raspi-HQCam-12MP.service: Scheduled restart job, restart counter is at 1.
Jul 27 14:34:21 raspi5 systemd[1]: Stopped camera-streamer-raspi-HQCam-12MP.service - camera-streamer web camera for Pi High Quality Camera Module 12MP on Raspberry PI.
```

In addition, here is my ```camera-streamer-raspi-HQCam-12MP.service``` file:
```
;
; Official Raspberry Pi Camera Module 3 12MP camera based on the Sony IMX708
; https://www.raspberrypi.com/products/camera-module-3/
;
[Unit]
Description=camera-streamer web camera for Pi High Quality Camera Module 12MP on Raspberry PI
After=network.target
ConditionPathExists=/sys/bus/i2c/drivers/imx477/10-001a/video4linux

[Service]
ExecStart=/usr/local/bin/camera-streamer \
  --camera-path=/base/axi/pcie@1000120000/rp1/i2c@88000/imx477@1a \
  --camera-type=libcamera \
  --camera-format=YUYV \
  --camera-width=2304 --camera-height=1296 \
  --camera-fps=30 \
  ; use two memory buffers to optimise usage
  --camera-nbufs=2 \
  ; the snapshot is 1920x1080
  --camera-snapshot.height=1080 \
  ; the video/webrtc is 1280x720
  --camera-video.height=720 \
  ; the stream is 853x480
  --camera-stream.height=480 \
  ; enable continuous autofocus
  --camera-options=AfMode=2 \
  --camera-options=AfRange=2 \
  --http-listen=0.0.0.0 \
  --http-port=8080 \
  --rtsp-port

DynamicUser=yes
SupplementaryGroups=video i2c
Restart=always
RestartSec=10
Nice=10
IOSchedulingClass=idle
IOSchedulingPriority=7
CPUWeight=20
AllowedCPUs=1-2
MemoryMax=250M

[Install]
WantedBy=multi-user.target
```

May someone has some input to solve this.

Best regards,
Deph