ayufan/camera-streamer #145: Add Horizontal/Vertical flipping and 90 degree rotation.

jdwhite, 2024-03-16:
Please add the ability to flip images/video horizontally/vertically and rotate images/video 90 degree clockwise/counterclockwise.

Currently this can be done within OctoPrint with a plugin such as MultiCam, but only works within the OctoPrint UI. Anything streamed directly via camera-streamer gets the default orientation.

jdwhite, 2024-03-16:
I'm aware of the `--camera-hflip` and `--camera-vflip` options but my camera (Logitech c920) doesn't support this. I gather that supporting my requested functionality would require post-processing by something such as `ffmpeg`.

Svanto1963, 2024-05-02:
Cant get --camera-hflip and --camera-vflip to work on my Raspberry pi + Raspi cam v3.0 PLZ. help.

crycode-de, 2024-08-07:
For the Raspberry Pi Camera (tested with v2.1) `--camera-hflip` and `--camera-vflip` are not working, but you can add the parameters `--camera-options=horizontalflip=1` and `--camera-options=verticalflip=1` which will do the trick.

By calling `camera-streamer --camera-list_options` you can query all available `--camera-options` and `--camera-isp.options` for your camera.

boehser-enkel, 2024-09-02:
> --camera-options=horizontalflip=1

shows:
```
device/v4l2/device_options.c: CAMERA: Configuring option 'Horizontal Flip' (00980914) = 0                                                                                                                          
device/v4l2/device_options.c: CAMERA: ioctl(ret=-1, errno=16): Can't set option Horizontal Flip                                                                                                                    
device/v4l2/device_options.c: CAMERA: Configuring option 'Vertical Flip' (00980915) = 0                                                                                                                            
device/v4l2/device_options.c: CAMERA: ioctl(ret=-1, errno=16): Can't set option Vertical Flip  
```                                                                                                         

           

> --camera-options=horizontalflip=1

Does nothing

Pi4 + Pi Cam V3

jerryfudd, 2025-04-24:
trying to find out the answer to this also