ayufan/camera-streamer #148: Logitech c920 not working with camera-streamer

fabiandj64, 2024-05-04:
I tried almost everything I could find online to get it to work but with no success.
Adding log and conf

[crowsnest.log](https://github.com/ayufan/camera-streamer/files/15210960/crowsnest.log)
[crowsnest.conf](https://github.com/ayufan/camera-streamer/files/15210961/crowsnest.txt)

fabiandj64, 2024-05-04:
All always get an "Can't set format" error

fabiandj64, 2024-05-05:
i changed a few entries in the camer-streamer.service file and now it is working.

closed 2024-05-05 by fabiandj64

elgatobavaria, 2025-01-22:
@fabiandj64 Would be nice if you could share your changes with the community. I would like to use the camera, too. Thank you

fabiandj64, 2025-01-23:
i dont even know if the changes to the camera-streamer.service made a difference; i tried to look up the service and saw, that they changed to having different services; so i guess the service is/was not the problem. i include my crowsnest.conf file; with this config it works for me. 
To get the absolute path of the camera you could use `ls /dev/v4l/` like you would get the serial of the mcu. (someone mentioned to use the absolute path instead of the short one)

[crowsnest.txt](https://github.com/user-attachments/files/18518151/crowsnest.txt)