ayufan/camera-streamer #149: Is RPi5 + Pi Camera v3 supported? (CAMERA: Cannot find encoder to convert from 'YUYV')

puterboy, 2024-06-05:
I tried running the following from the command line based off of `
```
/usr/bin/camera-streamer -camera-path=/base/axi/pcie@120000/rp1/i2c@88000/imx708@1a --camera-type=libcamera --camera-format=YUYV --camera-width=2304 --camera-height=1296 --camera-fps=30 --camera-nbufs=2 --camera-snapshot.height=1080 --camera-video.height=720 --camera-stream.height=480 --camera-options=AfMode=2 --camera-options=AfRange=2 --http-listen=0.0.0.0 --http-port=8080 --rtsp-port
```
I get back:
```
/usr/bin/camera-streamer Version: 0.2.8 (bc23191)
util/http/http.c: ?: HTTP listening on 0.0.0.0:8080.
output/rtsp/rtsp.cc: ?: Running RTSP server on '8554'
[63:20:06.896738749] [758671]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+118-563cd78e
[63:20:06.910870584] [758685]  INFO RPI pisp.cpp:653 libpisp version v1.0.5 999da5acb4f4 17-04-2024 (14:29:29)
[63:20:06.923921737] [758685]  INFO RPI pisp.cpp:1112 Registered camera /base/axi/pcie@120000/rp1/i2c@88000/imx708@1a to CFE device /dev/media0 and ISP device /dev/media1 using PiSP variant BCM2712_C0
device/libcamera/device.cc: CAMERA: Device path=/base/axi/pcie@120000/rp1/i2c@88000/imx708@1a opened
[63:20:06.924327421] [758671]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format Y16
[63:20:06.924355106] [758671]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format RGB6
[63:20:06.924363050] [758671]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format BGR6
[63:20:06.924394087] [758671]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format PC1M
[63:20:06.924610660] [758671]  INFO Camera camera.cpp:1183 configuring streams: (0) 1920x1080-YUYV
[63:20:06.924756771] [758685]  INFO RPI pisp.cpp:1396 Sensor: /base/axi/pcie@120000/rp1/i2c@88000/imx708@1a - Selected sensor format: 2304x1296-SBGGR10_1X10 - Selected CFE format: 2304x1296-PC1B
[63:20:06.925191547] [758671]  INFO Camera camera.cpp:1183 configuring streams: (0) 1920x1080-YUYV (1) 2304x1296-BGGR16_PISP_COMP1
[63:20:06.925302695] [758685]  INFO RPI pisp.cpp:1396 Sensor: /base/axi/pcie@120000/rp1/i2c@88000/imx708@1a - Selected sensor format: 2304x1296-SBGGR10_1X10 - Selected CFE format: 2304x1296-PC1B
device/buffer_list.c: CAMERA:capture: Using: 1920x1080/YUYV, buffers=2, bytesperline=3840, sizeimage=0.0MiB
device/buffer_list.c: CAMERA:capture: Opened 2 buffers. Memory used: 7.9 MiB
device/buffer_list.c: CAMERA:capture:1: Using: 2304x1296/BYR2, buffers=2, bytesperline=2304, sizeimage=0.0MiB
device/buffer_list.c: CAMERA:capture:1: Opened 2 buffers. Memory used: 5.7 MiB
device/camera/camera_output.c: CAMERA: Cannot find encoder to convert from 'YUYV'
```
I also tried  setting `--camera-format=`  to `MJPEG`, `YUV420` and `H264` to no avail.

So is there any way to get a PiCamera v3 to work on an RPi5 using `camera-streamer`?

-

puterboy, 2024-06-05:
I just tried `--camera-format=RGB` and I got a little further but a different error message:
```
/usr/bin/camera-streamer Version: 0.2.8 (bc23191)
util/http/http.c: ?: HTTP listening on 0.0.0.0:8080.
output/rtsp/rtsp.cc: ?: Running RTSP server on '8554'
[64:11:58.060617861] [763391]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+118-563cd78e
[64:11:58.075024644] [763405]  INFO RPI pisp.cpp:653 libpisp version v1.0.5 999da5acb4f4 17-04-2024 (14:29:29)
[64:11:58.088469097] [763405]  INFO RPI pisp.cpp:1112 Registered camera /base/axi/pcie@120000/rp1/i2c@88000/imx708@1a to CFE device /dev/media0 and ISP device /dev/media1 using PiSP variant BCM2712_C0
device/libcamera/device.cc: CAMERA: Device path=/base/axi/pcie@120000/rp1/i2c@88000/imx708@1a opened
[64:11:58.088893040] [763391]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format Y16
[64:11:58.088923225] [763391]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format RGB6
[64:11:58.088931003] [763391]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format BGR6
[64:11:58.088940428] [763391]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format PC1M
[64:11:58.089232779] [763391]  INFO Camera camera.cpp:1183 configuring streams: (0) 1920x1080-RGB888
[64:11:58.089386871] [763405]  INFO RPI pisp.cpp:1396 Sensor: /base/axi/pcie@120000/rp1/i2c@88000/imx708@1a - Selected sensor format: 2304x1296-SBGGR10_1X10 - Selected CFE format: 2304x1296-PC1B
[64:11:58.089821666] [763391]  INFO Camera camera.cpp:1183 configuring streams: (0) 1920x1080-RGB888 (1) 2304x1296-BGGR16_PISP_COMP1
[64:11:58.089904110] [763405]  INFO RPI pisp.cpp:1396 Sensor: /base/axi/pcie@120000/rp1/i2c@88000/imx708@1a - Selected sensor format: 2304x1296-SBGGR10_1X10 - Selected CFE format: 2304x1296-PC1B
device/buffer_list.c: CAMERA:capture: Using: 1920x1080/RGB3, buffers=2, bytesperline=5760, sizeimage=0.0MiB
device/buffer_list.c: CAMERA:capture: Opened 2 buffers. Memory used: 11.9 MiB
device/buffer_list.c: CAMERA:capture:1: Using: 2304x1296/BYR2, buffers=2, bytesperline=2304, sizeimage=0.0MiB
device/buffer_list.c: CAMERA:capture:1: Opened 2 buffers. Memory used: 5.7 MiB
device/camera/camera_output.c: CAMERA: Cannot find source for 'SNAPSHOT' for one of the formats 'JPEG, MJPG'.
```

Any thoughts?

DjofPlusPlus, 2024-06-13:
I'll reply since nobody else has yet... The RPI5 isn't supported yet because it doesn't have the hardware encoded that previous devices had. It will require significant changes to integrate software encoding.

puterboy, 2024-06-14:
That's too bad as other streamers support RPI5 since they allow you to use libcamera commands plus ffmpeg to generate the stream. This seems like a real limitation of camera-streamer relative to other options.

krokosik, 2024-07-12:
@puterboy can you recommend some different streamers? I'm stuck on a Pi5 :/

ETE-Design, 2024-07-12:
@puterboy You know of any "Light" Streamer like this one with as low as possible latensy who offer ISP Streaming for RPI HQ Camera? Cause haven't found Antony yet :-( But would like to try others who is up-to-date - Think this one is "Dead"