ayufan/camera-streamer #151: Failed to register/ no available cam (Pi4 + Pi Cam V3)

boehser-enkel, 2024-07-19:
Hi,
fresh install of Pi OS.

journalctl -xef -u camera-streamer-raspi-v3-12MP.service

> Jul 19 10:40:50 pi2 camera-streamer[2025]: [0:09:10.590338769] [2025]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+118-563cd78e                                                                            
> Jul 19 10:40:50 pi2 camera-streamer[2025]: [0:09:10.724705816] [2038] ERROR RPI pipeline_base.cpp:802 Failed to load a suitable IPA library                                                                        
> Jul 19 10:40:50 pi2 camera-streamer[2025]: [0:09:10.724764075] [2038] ERROR RPI vc4.cpp:216 Failed to register camera imx708: -22                                                                                  
> Jul 19 10:40:50 pi2 camera-streamer[2025]: device/libcamera/device.cc: CAMERA: No available cameras                                                                                                                
> Jul 19 10:40:50 pi2 camera-streamer[2025]: device/libcamera/device.cc: CAMERA: Camera `/base/soc/i2c0mux/i2c@1/imx708@1a` was not found.                                                                           
> Jul 19 10:40:50 pi2 camera-streamer[2025]: device/device.c: CAMERA: Can't open device: /base/soc/i2c0mux/i2c@1/imx708@1a                                                                                           
> Jul 19 10:40:50 pi2 systemd[1]: camera-streamer-raspi-v3-12MP.service: Main process exited, code=exited, status=255/EXCEPTION

> pi@pi2:~ $ libcamera-hello --list-cameras
> Available cameras                                                                                                                                                                                                  
> -----------------                                                                                                                                                                                                  
> 0 : imx708 [4608x2592 10-bit RGGB] (/base/soc/i2c0mux/i2c@1/imx708@1a)                                                                                                                                             
>     Modes: 'SRGGB10_CSI2P' : 1536x864 [120.13 fps - (768, 432)/3072x1728 crop]                                                                                                                                     
>                              2304x1296 [56.03 fps - (0, 0)/4608x2592 crop]                                                                                                                                         
>                              4608x2592 [14.35 fps - (0, 0)/4608x2592 crop]          

libcamera-hello also works

What can i do?

greetings

zergeborg, 2024-08-02:
I have the exact same problem. I see the exact same error in the journal. Running Bookworm 64 bit on a Raspi 4B model, with Camera Module 3 (wide) attached.

gyscos, 2024-08-22:
You might need to recompile `camera-streamer`. If running from `crowsnest` there is a `build.sh` script to help with that.
I had the same issue you had, and recompiling camera-streamer solved it for me.

boehser-enkel, 2024-09-02:
> You might need to recompile `camera-streamer`. If running from `crowsnest` there is a `build.sh` script to help with that. I had the same issue you had, and recompiling camera-streamer solved it for me.

At least that worked yes. Thank you